roleoroleo/yi-hack-MStar #1: release 0.0.1 RTSP server

grigoroff, 2019-09-20:
hi,
many many thanks to you for this great FW !! my 1080p Home cam 6FU with 4.5.0.0B stock FW is flashed and running great ! 
I have active RTSP low-res for testing but when i try to connect with VLC or jeedom(domotic), no video appear.
On SSH i stop an start rRTSPserver and all seems to be good on 554 port.
See attach if i'm doing some thing wrong.
thanks again for huge work 
[image]
[image]
[image]

roleoroleo, 2019-09-20:
Could you please verify if it's work, with cloud enabled?
You can start manually rRTSPServer with an env var to select resolution:
`RRTSP_RES=1 rRTSPServer &`
0: high res
1: low res
2: both

grigoroff, 2019-09-20:
you right !!! if I enable cloud it works !
any idea why ?

roleoroleo, 2019-09-20:
I don't know exactly, but if cloud application doesn't start the circular buffer with h264 frames is empty.
To solve this problem system.sh starts cloud application, waits for circular buffer filling and then kills cloud. Something is wrong in this procedure
You could try to increment wait time. From
`while [ "$IDX" -eq "0000" ] && [ $N -lt 50 ]; do`
to
`while [ "$IDX" -eq "0000" ] && [ $N -lt 100 ]; do`

roleoroleo, 2019-09-21:
I think I solved. The while loop was incorrect.
Please, try release 0.1.1.

grigoroff, 2019-09-21:
Cool I will see it when I come back! Thanks for your effort

JoseFerre, 2019-09-23:
Great job @roleoroleo :)
Camera is working fine but rtsp need a active external network connection to start working for first time when restart it, once the stream is up and working I can disable the external network connection without problems.
I will investigate it.
Thank you!!

roleoroleo, 2019-09-23:
Probably cloud application doesn't start reading if it can't connect to the server.
I will investigate too.

closed 2019-09-28 by roleoroleo