roleoroleo/yi-hack-MStar #2: Where I find home_y203c and sys_y203c files?

pidiman, 2019-09-28:
Sorry, but where I can find these files ?

"Save both files (home_y203c and sys_y203c)" 

Maibe I am blind, but I dont see these files.. 

Thanks :)

marcotuna, 2019-09-28:
Hello @pidiman 
You need to download them here:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases/

The current latest release is here: https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases/download/0.1.2/yi-hack-6FUS_4.5.0_0.1.2.zip

steef84, 2019-09-28:
lol @pidiman, im stuck at the point **_Download and install the SDK for MStar platform: the file name is "MStar MSC3XX SDK.zip" (Google is your friend)._** I found the zip but don't know what to do with it :)

roleoroleo, 2019-09-28:
You need SDK only if you want to build your own custom firmware.
If you want instead to load the firmware image, download the zip file from the release section and follow "Getting Started" instruction.

steef84, 2019-09-28:
Thanks @roleoroleo! I already got your latest release installed on 2 cams, but want to try your latest commits

roleoroleo, 2019-09-29:
You need to install the SDK on /opt/yi
But the problem is that you have to copy the original firmware on stock_firmware.
You need this file and I think is not so simple to find it.

steef84, 2019-10-03:
So I can’t compile a latest version with your latest commits without my own original firmware? The Ill just wait for another release

roleoroleo, 2019-10-04:
The updates available from the xiaoyi site are only partial file systems and from those it is not possible to reconstruct a complete fw.
I unsoldered the flash and dumped it to solve the problem.

gitjim, 2019-10-15:
Applied the y203c, initially can see the rtsp stream. but failed to stream after a reboot. now the camera auto reset everytime reboot and cannt pair with app (Yi home). although accessible to SSH, http://[IP]:8080 and rtsp server, high resolution and low resolution enabled, but still cannt stream to vlc. Portquery detected the 554 port is listen toT CP only. Please help!
[image]

roleoroleo, 2019-10-15:
Which version did you install?
Does the camera restart continuously?

gitjim, 2019-10-15:
firmware: 0.1.2
base ver: 4.5.0.0B_201909041616

sorry for confuse you, its auto reset to factory mode everytimes I shutdown/power off. its able to connect WIFI, but no able to pairing to Yi home(tested android and windows). no restart continuously

roleoroleo, 2019-10-15:
Please, install version 0.1.3.
A reset to factory is normal when a new firmware is installed.
After upgrade try to delete home and sys files from SD card.

gitjim, 2019-10-15:
Thanks, the video stream back. however still not able to pair with Yi home app. hence unable to disable the indicate light (blue light). also Yi water mark appear....

looking forward to new version

[image]

roleoroleo, 2019-10-15:
The switch "Disable Cloud" is on or off? If you wanto to use Yi app switch off it.
You can disable blue led from Yi app when you are able to use it.
Regarding Yi watermark, sorry but I didn't work on this feature.
With this firmware you can't disable it.

roleoroleo, 2019-10-15:
I will try to add this feature into the next version.

roleoroleo, 2019-10-31:
To get home and sys file, read issue #10 .

closed 2019-10-31 by roleoroleo