roleoroleo/yi-hack-MStar #4: 4.6.0.0A firmware on camera will not flash

skylarhays, 2019-10-02:
Trying to get this to flash to my 6fus camera but its not flashing. the firmware version on my cameras are 4.6.0.a_201903141924 im assuming the reason its not flashing is because its an older version. Is there a way to downgrade my firmware on my cameras to 4.5? This is the firmware that shipped on the cameras i have never allowed them to update. any solution would be awesome I was pretty excited to try this firmware out only to be let down again....

roleoroleo, 2019-10-02:
As specified in the readme:
"Currently this project supports only the following camera:
Yi 1080p Home 6FUS with firmware 4.5.0.0A or 4.5.0.0B"

I think that 4.5.0 and 4.6.0 refer to different hardware versions.
This explains why updated versions of both fw continue to be released.
In any case, I do not have a 4.6.0 version so I cannot check it.

skylarhays, 2019-10-02:
I can open the camera and get pictures of the hardware if you would like.

skylarhays, 2019-10-09:
only reason i bought the cams were for yihack4 I feel like the description could be clearer that you cant get the cams anymore that the hack supports.... is there any hope for 6fus with 4.6.0 firmware?

roleoroleo, 2019-10-10:
I don't have a 4.6.0 6FUS so I'm not working on it.

roleoroleo, 2019-10-26:
See this issue:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10

closed 2019-10-26 by roleoroleo