roleoroleo/yi-hack-MStar #6: RTSP user

jjdiazgarcia, 2019-10-03:
Hello,
I am trying to connect to camera using RTSP but I am not able to view stream. I have been trying to configure it using Homeassistant aswell as Synology Surveillance but I can't get it working.
Which user and pass do I need to use?
Thanks

jjdiazgarcia, 2019-10-03:
I have been noticed I don't need a user to be able to watch rtsp stream. I wasn't be able to watch RTSP because I need to enable cloud. I have the same problem as @grigoroff
@roleoroleo Have you been able to figure out why it is happenning?

roleoroleo, 2019-10-03:
This problem was resolved on version 0.1.1.
What version are you using?

jjdiazgarcia, 2019-10-04:
I am using 0.1.2 release.

I have been doing some tests and I have figured out that the problem was my HA running in my RPi. RPi resources are running out. I will plan to move my HA to a computer.
Anyway, I am getting some errors when I try to use ffmpeg from my computer and record video to a file.

I will open another issue to try to figure out what is the problem and improve next release.

I post the code I am using in HA to get it working to help people that want to integrate camera with HA although video gets freeze sometime due to my RPi resources aswell as some errors I will show in a new issue.

`stream:
ffmpeg:
  ffmpeg_bin: /usr/bin/ffmpeg

camera:
  - platform: ffmpeg
    name: Yi Home
    input: -rtsp_transport tcp -i rtsp://192.168.1.141/ch0_1.h264`

Thanks for all.

closed 2019-10-04 by jjdiazgarcia