roleoroleo/yi-hack-MStar #7: Error recording video from camera

jjdiazgarcia, 2019-10-04:
Hello,

I am trying to record the video of my camera using ffmpeg tool installed in my laptop but I am getting some errors.

Below error is for SD quality:

```
ffmpeg -rtsp_transport tcp -i rtsp://192.168.1.141/ch0_1.h264 test.mp4
ffmpeg version N-95216-ge6625ca41f-tessus  https://evermeet.cx/ffmpeg/  Copyright (c) 2000-2019 the FFmpeg developers
  built with Apple clang version 11.0.0 (clang-1100.0.33.8)
  configuration: --cc=/usr/bin/clang --prefix=/opt/ffmpeg --extra-version=tessus --enable-avisynth --enable-fontconfig --enable-gpl --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libfreetype --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libmysofa --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenh264 --enable-libopenjpeg --enable-libopus --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvmaf --enable-libvo-amrwbenc --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-version3 --pkg-config-flags=--static --disable-ffplay
  libavutil      56. 35.100 / 56. 35.100
  libavcodec     58. 59.101 / 58. 59.101
  libavformat    58. 33.100 / 58. 33.100
  libavdevice    58.  9.100 / 58.  9.100
  libavfilter     7. 61.100 /  7. 61.100
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
[h264 @ 0x7fb8f3801800] SEI type 128 size 1912 truncated at 912
[h264 @ 0x7fb8f3801800] SEI type 128 size 1912 truncated at 896
[h264 @ 0x7fb8f3801800] concealing 746 DC, 746 AC, 746 MV errors in P frame
[h264 @ 0x7fb8f3801800] SEI type 41 size 1888 truncated at 816
[h264 @ 0x7fb8f3801800] SEI type 41 size 1888 truncated at 799
[h264 @ 0x7fb8f3801800] concealing 860 DC, 860 AC, 860 MV errors in P frame
[h264 @ 0x7fb8f3801800] SEI type 176 size 1000 truncated at 528
[h264 @ 0x7fb8f3801800] SEI type 176 size 1000 truncated at 509
[h264 @ 0x7fb8f3801800] left block unavailable for requested intra mode
[h264 @ 0x7fb8f3801800] error while decoding MB 0 1, bytestream 6172
[h264 @ 0x7fb8f3801800] concealing 920 DC, 920 AC, 920 MV errors in P frame
[h264 @ 0x7fb8f3801800] SEI type 76 size 1312 truncated at 664
[h264 @ 0x7fb8f3801800] SEI type 76 size 1312 truncated at 647
[h264 @ 0x7fb8f3801800] concealing 797 DC, 797 AC, 797 MV errors in P frame
[h264 @ 0x7fb8f3801800] SEI type 88 size 1464 truncated at 1208
[h264 @ 0x7fb8f3801800] SEI type 88 size 1464 truncated at 1192
[h264 @ 0x7fb8f3801800] concealing 519 DC, 519 AC, 519 MV errors in P frame
[h264 @ 0x7fb8f3801800] concealing 786 DC, 786 AC, 786 MV errors in P frame
[h264 @ 0x7fb8f3801800] Invalid NAL unit 1, skipping.
[h264 @ 0x7fb8f3801800] concealing 518 DC, 518 AC, 518 MV errors in P frame
[h264 @ 0x7fb8f3801800] concealing 824 DC, 824 AC, 824 MV errors in P frame
[h264 @ 0x7fb8f3801800] concealing 671 DC, 671 AC, 671 MV errors in P frame
[h264 @ 0x7fb8f3801800] concealing 617 DC, 617 AC, 617 MV errors in P frame
[h264 @ 0x7fb8f3801800] SEI type 110 size 1536 truncated at 520
Input #0, rtsp, from 'rtsp://192.168.1.141/ch0_1.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_1.h264
  Duration: N/A, start: 8.683767, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, unknown/unknown/bt470bg, progressive), 640x360, 29.25 tbr, 90k tbn, 180k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[h264 @ 0x7fb8f2807e00] SEI type 128 size 1912 truncated at 896
[h264 @ 0x7fb8f2808400] concealing 746 DC, 746 AC, 746 MV errors in P frame
[h264 @ 0x7fb8f280bc00] SEI type 41 size 1888 truncated at 799
[h264 @ 0x7fb8f280c200] concealing 860 DC, 860 AC, 860 MV errors in P frame
[h264 @ 0x7fb8f280ce00] SEI type 176 size 1000 truncated at 509
[h264 @ 0x7fb8f280d400] left block unavailable for requested intra mode
[h264 @ 0x7fb8f280d400] error while decoding MB 0 1, bytestream 6172
[h264 @ 0x7fb8f280d400] concealing 920 DC, 920 AC, 920 MV errors in P frame
[libx264 @ 0x7fb8f2803c00] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[libx264 @ 0x7fb8f2803c00] profile Progressive High, level 3.0, 4:2:0, 8-bit
[libx264 @ 0x7fb8f2803c00] 264 - core 157 r2969 d4099dd - H.264/MPEG-4 AVC codec - Copyleft 2003-2019 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=11 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=crf mbtree=1 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'test.mp4':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_1.h264
    encoder         : Lavf58.33.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuvj420p(pc), 640x360, q=-1--1, 29.25 fps, 14976 tbn, 29.25 tbc
    Metadata:
[... 922 lines cut ...]
^CError writing trailer of test.mp4: Immediate exit requested
frame=  868 fps= 18 q=-1.0 Lsize=    3584kB time=00:01:02.60 bitrate= 469.0kbits/s dup=649 drop=0 speed=1.27x
video:3812kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
[libx264 @ 0x7fcb77845000] frame I:12    Avg QP:12.87  size: 66211
[libx264 @ 0x7fcb77845000] frame P:271   Avg QP:13.79  size:  9603
[libx264 @ 0x7fcb77845000] frame B:585   Avg QP:18.76  size:   866
[libx264 @ 0x7fcb77845000] consecutive B-frames:  5.3% 13.4%  3.5% 77.9%
[libx264 @ 0x7fcb77845000] mb I  I16..4: 22.5% 59.2% 18.3%
[libx264 @ 0x7fcb77845000] mb P  I16..4:  3.9%  6.6%  1.0%  P16..4: 10.6%  2.4%  1.3%  0.0%  0.0%    skip:74.2%
[libx264 @ 0x7fcb77845000] mb B  I16..4:  0.1%  0.2%  0.0%  B16..8:  7.5%  0.2%  0.0%  direct: 0.1%  skip:91.9%  L0:35.7% L1:63.2% BI: 1.1%
[libx264 @ 0x7fcb77845000] 8x8 transform intra:58.0% inter:66.0%
[libx264 @ 0x7fcb77845000] coded y,uvDC,uvAC intra: 39.6% 0.2% 0.1% inter: 1.7% 0.0% 0.0%
[libx264 @ 0x7fcb77845000] i16 v,h,dc,p: 24% 25% 22% 29%
[libx264 @ 0x7fcb77845000] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 30% 25% 32%  2%  2%  2%  4%  1%  2%
[libx264 @ 0x7fcb77845000] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 34% 28% 17%  3%  7%  3%  6%  2%  2%
[libx264 @ 0x7fcb77845000] i8c dc,h,v,p: 100%  0%  0%  0%
[libx264 @ 0x7fcb77845000] Weighted P-Frames: Y:4.1% UV:0.0%
[libx264 @ 0x7fcb77845000] ref P L0: 82.3%  3.7% 10.0%  3.6%  0.4%
[libx264 @ 0x7fcb77845000] ref B L0: 72.8% 25.5%  1.6%
[libx264 @ 0x7fcb77845000] ref B L1: 96.2%  3.8%
[libx264 @ 0x7fcb77845000] kb/s:539.62
Exiting normally, received signal 2.
```

I am 100% available to test anything you need or to code anything you need

roleoroleo, 2019-10-04:
Yes I know, this is the problem I described in the readme.
I can't detect correctly p-frames for low and high res.
If you want to help me, make a copy of the file /dev/fshare_frame_buf and analyze it.
It's a set of h264 frames.

You will find a sequence like this for IDR frames:
00 00 00 01 67 ... - SPS
00 00 00 01 68 ... - PPS
00 00 00 01 06 ... - SEI (???)
00 00 00 01 65 ... - IDR
This sequence is simple to detect beacuse the SPS contains the resolution.

Between two IDR frames you will find a lot of sequence:
00 00 00 01 06 ... - SEI (???)
00 00 00 01 41 ... - P-SLICE
and I don't know how to assign this to low res stream or to high res stream.

Ideas?

jjdiazgarcia, 2019-10-05:
Hi @roleoroleo I have been trying to understand how h264 works but I am not able to do it without any knowledge (I have never worked with it so I can't help you with this problem).
Anyway, I have been looking for h264 decoder to ffmpeg and I have found this github project. Maybe it helps you to try to figure out how to read h264 frames. [https://github.com/shengbinmeng/ffmpeg-h264-dec](url)

roleoroleo, 2019-10-18:
Please try last version.

jjdiazgarcia, 2019-10-18:
@roleoroleo I have just updated firmware but camera seems that it is not working properly at all.
If I enable rtsp high quality, I can't watch video using VLC player (rtsp://CAMERA_IP/ch0_0.h264). However, if I don't enable high quality, I can watch video using VLC player (rtsp://CAMERA_IP/ch0_1.h264). Although video is working in low res, every 2-3 seconds video blink and bright/quality changes. Furthermore, video gets freezed some times (not so much but maybe two or three times it happened).
Do you need any configuration or dump?

jjdiazgarcia, 2019-10-18:
@roleoroleo I am trying to watch video right now and it doesn't work (high quality nor low quality)

jjdiazgarcia, 2019-10-18:
@roleoroleo after some time powered on, I can't connect to camera web page. If i try to reach web page port, I get host is down.
```
telnet 192.168.X.XXX 8080
Trying 192.168.X.XXX...
telnet: connect to address 192.168.X.XXX: Host is down
telnet: Unable to connect to remote host
```

roleoroleo, 2019-10-18:
Very strange.
Which version are you using?
Please post your config.

jjdiazgarcia, 2019-10-20:
I have installed version 0.1.4.
Here you are my configuration:
'''
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_HIGH=no
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
'''
I have noticed that if cloud is disabled, camera doesn't work properly. However, if I enable cloud, it works properly.

jjdiazgarcia, 2019-10-20:
@roleoroleo I have reset camera and it seems to be working properly. Here you are my configuration right now:
```
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_HIGH=no
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
```
Anyway, I have noticed that camera is detecting motion when there is no motion. I have just formatted sd card and videos size are more than 50MB
```
/home/yi-hack/etc # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.8M     28.0K  99% /
tmpfs                    29.7M      1.8M     27.9M   6% /dev
/dev/block/mtdblock3     12.2M     10.2M      1.9M  84% /home
tmpfs                    32.0M     76.0K     31.9M   0% /tmp
/dev/block/mmcblk0p1      7.3G     53.7M      7.2G   1% /tmp/sd
```
Xiaomi app also reports me motion when there are no motion. Maybe video processing is introducing some noise and camera detects it as movement?

roleoroleo, 2019-10-21:
The video processing added in this hack is "read-only".
It access the kernel memory buffer without blocking.
The only problem introduced could be the excessive memory occupation of the processes, especially during the RTSP streaming.
The h264grabber process, if there are no RTSP clients running, is locked in a write so it does not take up CPU, but when using a RTSP client, it can take up a lot of resources.

Anyway, I tried your config and it doesn't give me any problems
Could you verify if the motion detection problem happens even when you're not using RTSP?
(enabled but no clients running).

jjdiazgarcia, 2019-10-23:
I have been testing during all the day with the rtsp client off. There isn't any video inside record folder. I have disabled cloud and powered off rtsp client to test and no video recorded.
Here you are my configuration which was recording videos so frequently (about 8GB per day)
```
/tmp/sd/record # du -h .
311.6M	./2019Y10M21D17H
497.2M	./2019Y10M21D18H
504.3M	./2019Y10M21D19H
322.6M	./2019Y10M21D20H
352.5M	./2019Y10M21D21H
497.3M	./2019Y10M21D22H
496.2M	./2019Y10M21D23H
496.9M	./2019Y10M22D00H
497.5M	./2019Y10M22D01H
497.1M	./2019Y10M22D02H
497.5M	./2019Y10M22D03H
495.0M	./2019Y10M22D04H
342.5M	./2019Y10M22D05H
222.7M	./2019Y10M22D06H
57.3M	./2019Y10M22D07H
5.9G
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_HIGH=no
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
```
And this is my current configuration which is not recording any video
```
/tmp/sd/record # du -h .
8.0K	.
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_HIGH=no
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
```

jjdiazgarcia, 2019-10-24:
@roleoroleo I have been all the day yesterday monitoring camera and it didn't record anything (although there was a rtsp client connected to it). Here is my configuration for this testing. record folder date is 22nd October because folder was created that day but camera hasn't record anything.
```
/home/yi-hack # cd /tmp/sd/
/tmp/sd # ls -l
total 16
drwx------    2 root     0             8192 Oct 21 18:59 log
drwx------    2 root     0             8192 Oct 22 07:33 record
/tmp/sd # cd
/home/yi-hack # cd etc/
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_HIGH=yes
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
```
I have just enabled cloud and camera started to record. It seems that cloud option must be enabled to record motion. Here you are my config
```
/home/yi-hack # cd /tmp/sd/
/tmp/sd # ls -l
total 16
drwx------    2 root     0             8192 Oct 21 18:59 log
drwx------    3 root     0             8192 Oct 24 06:27 record
/tmp/sd # cd record/
/tmp/sd/record # ls
2019Y10M24D06H  tmp.mp4.tmp
/tmp/sd/record # cd
/home/yi-hack # cd etc/
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_HIGH=yes
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
```

roleoroleo, 2019-10-24:
So now it's working properly?
Even with a rtsp client running?
In any case I'm trying to modify some processes to decrease the CPU load.
Keep me updated.

jjdiazgarcia, 2019-10-24:
@roleoroleo I think it is not working properly because I have to enable cloud to record videos when motion is detected (although motion is detected even if there is no movement so camera is recording a lot of videos and sd card space is running out every 20-24 hours)
Fantastic. Camera CPU is high right now.

roleoroleo, 2019-10-25:
I partially understood the problem.
I need to restore the old mode: cloud application must be started otherwise recording and motion detection do not start.

jjdiazgarcia, 2019-10-28:
@roleoroleo Ok, I understand the issue related to motion detection and record.
I have updated my camera to version 0.1.6 and everything seems to work fine except motion detection (I know why it doesn't work. You have told me why in the previous comment) and problem with motion detection and record when cloud is enabled. 
I don't know why but camera is still detecting motion when there is no motion so camera is recording and sd space is running out every day (camera records about 8GB of videos every day).
Below is the output from today disk usage
```
/tmp/sd/record # du -h .
83.6M	./2019Y10M28D10H
480.4M	./2019Y10M28D00H
474.7M	./2019Y10M28D01H
479.9M	./2019Y10M28D02H
479.3M	./2019Y10M28D03H
476.0M	./2019Y10M28D04H
344.2M	./2019Y10M28D05H
107.4M	./2019Y10M28D07H
189.8M	./2019Y10M28D09H
320.3M	./2019Y10M28D06H
396.0M	./2019Y10M28D08H
3.7G	.
```
You can see there is a lot of videos during the night and all of us are sleeping (neither we have any pets at home so there is no motion)

jjdiazgarcia, 2019-10-30:
@roleoroleo I have been some days trying to find a pattern of recording and I think I have found it.
It looks like camera detects motion based of image quality. The darker it is, the more movement it detects.
```
/tmp/sd/record # du -h .
99.6M	./2019Y10M30D07H
289.4M	./2019Y10M29D09H
323.7M	./2019Y10M29D10H
62.0M	./2019Y10M29D11H
31.0M	./2019Y10M29D16H
263.1M	./2019Y10M29D17H
408.1M	./2019Y10M29D18H
485.6M	./2019Y10M29D19H
359.2M	./2019Y10M29D20H
480.5M	./2019Y10M30D00H
480.4M	./2019Y10M30D01H
481.5M	./2019Y10M30D02H
478.0M	./2019Y10M30D03H
480.7M	./2019Y10M30D04H
262.5M	./2019Y10M30D06H
381.7M	./2019Y10M30D05H
420.1M	./2019Y10M29D21H
516.7M	./2019Y10M29D22H
482.3M	./2019Y10M29D23H
6.6G	.
```

roleoroleo, 2019-11-07:
I also noticed some motion detection events during the night but I can't explain the reason.
Are you sure that with the original firmware this problem doesn't happen?

roleoroleo, 2019-11-08:
I tested the cam withiout hack and I can confirm that I have a lot of false motion detection.
The event is really detected because I can see it on the system log.
May be an error added to 4.5.0.0C_201910080934 (not present in the previous 4.5.0.0A_201902261502)?

jjdiazgarcia, 2019-11-08:
Hi @roleoroleo Sorry for my delayed answer.
I am not able to test with any other versions because I don't have any other camera neither any other firmware version. Do you have stock 4.5.0.0A_201902261502 firmware to test it?

roleoroleo, 2019-11-10:
Yes, I have.
I can try,

jjdiazgarcia, 2019-11-11:
Nice. Thanks so much @roleoroleo

roleoroleo, 2019-11-12:
I reinstalled the original firmware 4.5.0.0A and unfortunately I have the same problems.
There are a lot of motion detection events even if there are apparently no movements.
Therefore it is not a problem due to the hacking procedure.
I will try to decrease the sensitivity of the detection algorithm.

roleoroleo, 2019-11-12:
Decreasing the sensitivity it works better.

jjdiazgarcia, 2019-11-13:
Perfect.
Anyway, I am not sure if I want to configure camera with less sensivity. Is it possible to add a field in webpage to adjust it as we want? I suggest this to avoid a fix sensivity.

roleoroleo, 2019-11-13:
Now, I'm using low sensitivity and it works very well. No false detections and no undetected events.
Add a field to the configuration web page is complicated. We should reverse the binary configuration file where settings are stored.

jjdiazgarcia, 2019-11-13:
How have you changed sensivity? If you have to re-run a binary with a new parameter, this new field could store value in a file and, when camera boot, read this file to start this binary file.
Is it possible?

roleoroleo, 2019-11-13:
I changed it from YI app.

jjdiazgarcia, 2019-11-13:
Oh, ok.
My last doubt releated to this issue. What happen with low sensivity? Will camera detect movement far away? For example 7-8 meters?

roleoroleo, 2019-11-13:
Sniffing the communication between processes I realized that it is possible to send some settings using IPC messages. And sensitivity seems to be one of them.
So maybe I can add a setting on the web interface.
Regarding the detection distance, I suggest you try some tests.

jjdiazgarcia, 2019-11-13:
That sounds fantastic. Then, I will wait for a new release to be able to configure sensitivity using IPC message before close this issue. Thanks.

roleoroleo, 2019-11-13:
Please try version 0.1.9.

jjdiazgarcia, 2019-11-14:
I have just updated camera to version 0.1.9 and it is not detecting motion right now.
I will test it during some days and I will let you know.
Thanks so much

jjdiazgarcia, 2019-11-17:
@roleoroleo now, motion detection is working better. I close this issue. Thanks so much for all your help and contribution to this project.

closed 2019-11-17 by jjdiazgarcia