roleoroleo/yi-hack-MStar #8: 4.5.0.0C firmware

ivangurin, 2019-10-10:
It doesn't work on 4.5.0.0C firmware. Could you fix it? Thanks.

roleoroleo, 2019-10-11:
Did you try it?
Analyzing  4.5.0.0C update there is no reason that it should not work.

ivangurin, 2019-10-11:
Yes, I did try it. After copy files on flash card, inset flash card to a camera and turn on the camera nothing was happened. The camera hasn't been upgraded to your firmware.  And I don't know why. I am assuming that cause is new official firmware version.

roleoroleo, 2019-10-12:
It is possible that yi changed the loading procedure.
But unfortunately I can't fix it without having the same fw version.
If you are able to connect a serial cable to the pcb we could analyze the bootlog togheter.

ivangurin, 2019-10-12:
I'd like to help you, but I don't have serial cable, don't know what is pcb and don't know how to connect cable to camera.

Christian-Krupa, 2019-10-16:
@roleoroleo i successfully installed your firmware over an 4.5.0.0C_201910... original firmware. rtsp, ssh and webinterface seems to work as expected

roleoroleo, 2019-10-17:
Good news!
Did you follow a different procedure than the one described in the README?
I would like to understand why @ivangurin didn't succeed.

richmidwinter, 2019-10-18:
Works for me on 4.5.0.0C_201910080934... at least until I reboot, then I have to go through the pairing process from scratch.

roleoroleo, 2019-10-18:
And after the pairing it works?

ivangurin, 2019-10-19:
I still can't upgrade my camera even with new version of firmware. I everything do by instruction. I have 2Gb with fat32, copy files to root, insert card to camera and turn it on. But nothing happens. Camera starts quickly as usual without upgrade. 
On the back of my camera there is qr code with number 6FUSY21KKR. Is firmware suitable for my camera?

richmidwinter, 2019-10-19:
For me the pairing "fails" as far as the app is concerned but the camera's on the network, services running, I can telnet in, receive the rtsp stream, etc.

And then I reboot and I get the irritating waiting to connect voice message every few seconds and I have to go through the process again.

roleoroleo, 2019-10-19:
Try to reset the camera and restart the process from scratch.
But before the reset, enable the cloud from the web interface.

ivangurin, 2019-10-19:
Finally I installed your firmware on my camera. Something was wrong with my sd card. I formatted the sd card from the camera app. After that the firmware was installed successfully. Thank you.

closed 2019-10-19 by ivangurin