roleoroleo/yi-hack-MStar #9: image snapshot service not working on yi-hack-6FUS_4.5.0_0.1.4

Christian-Krupa, 2019-10-16:
i am unable to get an image snapshot from the camera:
`$ curl -vv http://xxx.xxx.xxx.xxx:8080/cgi-bin/snapshot.sh?res=low
*   Trying xxx.xxx.xxx.xxx...
...
> GET /cgi-bin/snapshot.sh?res=low HTTP/1.1
...
> User-Agent: curl/7.64.1
> Accept: */*
>
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Content-type: image/jpeg
<
* Closing connection 0`

also the buffer2jpg command does not seem to work:
`/home/yi-hack # buffer2jpg --output=/tmp/sd/test.jpg --res=high
/home/yi-hack # ls /tmp/sd/test.jpg
ls: /tmp/sd/test.jpg: No such file or directory`

`/home/yi-hack # buffer2jpg --output=stdout --res=high
/home/yi-hack #
`
any idea whats wrong?

roleoroleo, 2019-10-17:
RTSP is working properly?
framefinder daemon is running?
Could you dump the content of /tmp/iframe.idx?
`hexdump /tmp/iframe.idx`

Christian-Krupa, 2019-10-17:
* rtsp://IP/ch0_0.h264 - works fine
* ONVIF - works fine (even faster than rtsp)
* framefinder seems to run, are there any debug logs?:
```
/home/yi-hack # ps ax|grep framefinder
  597 root      0:00 framefinder
 2840 root      0:00 grep framefinder
```
* iframe.idx is not available:
```
/home/yi-hack # ls -al /tmp/iframe.idx
ls: /tmp/iframe.idx: No such file or directory
```
```
/home/yi-hack # find / -name iframe.idx
/home/yi-hack #
```

roleoroleo, 2019-10-17:
Could you dump the first 100 bytes of /dev/fshare_frame_buf file?

Christian-Krupa, 2019-10-17:
```
/home/yi-hack # hexdump /dev/fshare_frame_buf
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
0000020 0200 0000 0000 0000 0001 0000 0402 0000
0000030 0000 0000 0000 0000 0000 0000 0000 0000
*
01b40e0
```

roleoroleo, 2019-10-17:
Ok, the buffer is empty.
Last request.
Could you post your config?

Christian-Krupa, 2019-10-17:
is this enough?
```
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=yes
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=no
MQTT=yes
RTSP=yes
RTSP_HIGH=yes
ONVIF=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
```

```
/home/yi-hack/etc # cat mqttv4.conf
###############################################################################
#                       Configuration file for mqttv4                         #
###############################################################################

# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=192.168.0.xxx
MQTT_PORT=1883

MQTT_CLIENT_ID=yi-cam

MQTT_USER=xxxxxxxxxxx
MQTT_PASSWORD=xxxxxxxxxxx

MQTT_PREFIX=yicam

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_MOTION=motion

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN=1
```

roleoroleo, 2019-10-17:
Your configuration is the same as mine, I don't understand why it doesn't work.
The problem is that the circular buffer does not fill up.
If you look at the file /home/yi-hack/script/system.sh you can see that there is a trick to start filling the buffer:
```
if [[ $(get_config DISABLE_CLOUD) == "no" ]] ; then
    (
        cd /home/app
        sleep 2
        ./mp4record &
        ./cloud &
        ./p2p_tnp &
        ./oss &
        ./watch_process &
    )
else
    (
        cd /home/app
        sleep 2
        ./mp4record &
        # Trick to start circular buffer filling
        ./cloud &
        IDX=`hexdump -n 16 /dev/fshare_frame_buf | awk 'NR==1{print $8}'`
        N=0
        while [ "$IDX" -eq "0000" ] && [ $N -lt 50 ]; do
            IDX=`hexdump -n 16 /dev/fshare_frame_buf | awk 'NR==1{print $8}'`
            N=$(($N+1))
            sleep 0.2
        done
        killall cloud
        if [[ $(get_config REC_WITHOUT_CLOUD) == "no" ]] ; then
            killall mp4record
        fi
    )
fi
```

If cloud application doesn't start, the buffer is empty.
So I start it, I wait for buffer filling and I kill it.
Something is not working in this sequence.

Christian-Krupa, 2019-10-17:
i tried the example from the script:
```
/home/app # ./cloud &
/home/app # IDX=`hexdump -n 16 /dev/fshare_frame_buf | awk 'NR==1{print $8}'`
/home/app # N=0
/home/app #  while [ "$IDX" -eq "0000" ] && [ $N -lt 50 ]; do
> IDX=`hexdump -n 16 /dev/fshare_frame_buf | awk 'NR==1{print $8}'`
> N=$(($N+1))
> sleep 1
> done
/home/app # echo $IDX
0000
```

after that i started the cloud tool and killed it after 26 seconds:
```
/home/app # date;./cloud
Thu Oct 10 19:13:10 UTC 2019
^C
/home/app # date
Thu Oct 10 19:13:36 UTC 2019
```

there seems to be no change.

```
/home/app # hexdump /dev/fshare_frame_buf
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
0000020 0200 0000 0000 0000 0001 0000 0402 0000
0000030 0000 0000 0000 0000 0000 0000 0000 0000
*
01b40e0
```

```
curl -v http://xxx:8080/cgi-bin/snapshot.sh
Do 17 Okt 2019 22:28:40 CEST
...
> GET /cgi-bin/snapshot.sh HTTP/1.1
...
> User-Agent: curl/7.64.1
> Accept: */*
>
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Content-type: image/jpeg
<
* Closing connection 0
```

```
/home/app # buffer2jpg --output=stdout --res=high
/home/app #
```
at the same time rtsp streaming works as expected.

Christian-Krupa, 2019-10-17:
i got the problem: i configured my firewall that the camera is NOT able to connect to the internet. 

i thought that the "cloud" app only initialize the video buffer. but it seems that the cloud app only do so when an internet connection is available.

with an connection to the "cloud" server i get this output:
````
/home/app # ./cloud
trans_json_ex key = "code", result = 20000, len = 5, size = 16
trans_json_ex key = "time", result = 1571345132617, len = 13, size = 32
trans_json_ex key = "code", result = 20000, len = 5, size = 16
trans_json_ex key = "css_flag", result = 0, len = 1, size = 16
trans_json_ex key = "css_mode", result = 1, len = 1, size = 16
trans_json_ex key = "appParam", result = , len = 0, size = 512
trans_json_ex key = "code", result = 20000, len = 5, size = 16
trans_json_ex key = "data", result = "css_flag":0,"css_mode":1,"appParam":"","tz_offset":7200000,"timezone":"GMT+02:00","language":"en-US", len = 101, size = 1024
trans_json_ex key = "tz_offset", result = 7200000, len = 7, size = 16
trans_json_ex key = "css_flag", result = 0, len = 1, size = 16
trans_json_ex key = "css_mode", result = 1, len = 1, size = 16
trans_json_ex key = "appParam", result = , len = 0, size = 512
cmd(rm /tmp/sd/log_first_login.tar.gz;rm /tmp/sd/first_login -r;mkdir /tmp/sd/first_login;ps > /tmp/sd/first_login/ps1.txt;cp /home/homever /tmp/sd/first_login;lsmod > /tmp/sd/first_login/lsmod.txt;ls -laSR /home > /tmp/sd/first_login/file.txt;cp /tmp/wpa_supplicant.conf  /tmp/sd/first_login;cp /tmp/log.txt /tmp/sd/first_login;cp /tmp/log_oss.txt /tmp/sd/first_login;df /home > /tmp/sd/first_login/df.txt;df / >> /tmp/sd/first_login/df.txt;tar -cf /tmp/sd/log_first_login.tar.gz /tmp/sd/first_login;mkdir /tmp/sd/log; mv /tmp/sd/log_first_login.tar.gz /tmp/sd/log;rm /tmp/sd/first_login -r &)
rm: can't remove '/tmp/sd/log_first_login.tar.gz': No such file or directory
rm: can't remove '/tmp/sd/first_login': No such file or directory
tar: removing leading '/' from member names
mkdir: can't create directory '/tmp/sd/log': File exists
trans_json_ex key = "code", result = xxxxx, len = 5, size = 16
trans_json_ex key = "DID", result = xxxxx, len = 20, size = 32
trans_json_ex key = "License", result = xxxxx:, len = 7, size = 32
trans_json_ex key = "InitString", result = xxxxx, len = 82, size = 128
^C
`````

when i now kill "cloud" i am able to take screenshots via the weburl: .....:8080/cgi-bin/snapshot.sh

roleoroleo, 2019-10-18:
Now I remember...
Another user posted the same issue.
I can solve the problem using s different image buffer but I don't know if there is enough cpu.
I will try...

roleoroleo, 2019-10-20:
Try release 0.1.5.

Christian-Krupa, 2019-10-20:
i enabled the firewall for the cam again. now it seems to run as expected. without an internet connection i am able to create snapshots via curl, thank you!

```
$ curl http://192.xxx.xxx.xxx:8080/cgi-bin/snapshot.sh > test.jpg
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  234k    0  234k    0     0  53023      0 --:--:--  0:00:04 --:--:-- 53023
```

closed 2019-10-20 by Christian-Krupa