roleoroleo/yi-hack-MStar #10: Is anyone going to work on 4.6.0

skylarhays, 2019-10-21:
this is now listed as the alternative hack for the mstar cameras....   is there any work being done on ALL the 6fus cams??? a lot of us are waiting on this.

roleoroleo, 2019-10-21:
I don’t have a 4.6.0 so i’m not working on it.
I don’t know the difference between 4.5.0 and 4.6.0 but I think that 4.6.0 is not only an upgrade.
Probably the platforms are a little bit different.
If someone is confident with a welder and linux we can try together.

skylarhays, 2019-10-21:
i can solder and I am ok with linux... i would gladly disassemble one of mine to try to get the ball rolling. what is needed a usb to serial ttl adapter? I have one if that is what is needed.

roleoroleo, 2019-10-22:
Yes, for the 1st step is enough.

1. Open the cam
2. Find the 3 uart pads (normally labeled rx, tx and gnd)
3. Connect the ttl to rs232/usb adapter
4. Open putty with logging enabled
5. Switch on the cam
6. Wait for loading to complete
7. Check if we are lucky and a shell is available
8. Post the log (eventually cleaned up from your data, if present)

With the log we can investigate about the difference between 4.5.0 and 4.6.0.
And we could find the name of the upgrade file on SD.

skylarhays, 2019-10-22:
I will try to do this this weekend.

skylarhays, 2019-10-26:
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.10.25 21:41:44 =~=~=~=~=~=~=~=~=~=~=~=

IPL gd156225
D-01.

HW Reset
64MB

BIST0_0001-OK

offset:00010000

size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4

MXP found at 0x00020000

  decomp_size=0x000414d4

-----------------------U-Boot 2015.01 (Feb 26 2019 - 10:53:16)-----------------------

Version: I3g138a6f9
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one.bin 
    one.bin Not exist(fatsize err) 

+++++++++++++++++++   check one_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one_h201c 
    one_h201c Not exist(fatsize err) 

+++++++++++++++++++   check uboot_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 uboot_h201c 
    uboot_h201c Not exist(fatsize err) 

+++++++++++++++++++   check kernel_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 kernel_h201c 
    kernel_h201c Not exist(fatsize err) 

+++++++++++++++++++   check sys_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 sys_h201c 
    sys_h201c Not exist(fatsize err) 

+++++++++++++++++++   check home_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 home_h201c 
[... 266 lines cut ...]
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
MSYS: DMEM request: [S0:VENCDMOUT]:0x00006400
MSB2@v1.1-01:r&d analysis.
MSYS: DMEM request: [VSPL-I0P1B0]:0x0005A000
MSYS: DMEM request: [VSPL-I0P1B1]:0x0005A000
MSYS: DMEM request: [VSPL-I0P1B2]:0x0005A000
MSYS: DMEM request: [VENC-48]:0x00025800
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
MSYS: DMEM request: [S1:VENCDMOUT]:0x00005600
MSYS: DMEM request: [S1:VENCDMP0]:0x00056400
MSYS: DMEM request: [S1:VENCDMP1]:0x00056400
MSYS: DMEM request: [VSPL-I0P3B0]:0x00016800
MSYS: DMEM request: [VSPL-I0P3B1]:0x00016800
MSYS: DMEM request: [VSPL-I0P3B2]:0x00016800

[PID_LIST] pid_list_init ok, [ ver=Feb 26 2019, 14:28:36 ] 
usbcore: deregistering interface driver rtl8188fu
==20150512==> hub_port_init 1 #0
Plug in USB Port1
usb 2-1: reset high-speed USB device number 2 using soc:Mstar-ehci-1
usbcore: registered new interface driver rtl8188fu
cfg80211: Calling CRDA to update world regulatory domain
MSYS: DMEM request: [VENC-49]:0x00007800

skylarhays, 2019-10-26:
there is a shell too!

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.10.25 21:48:09 =~=~=~=~=~=~=~=~=~=~=~=

U-Boot 2015.01 (Feb 26 2019 - 10:53:16)
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) 4.8.3 20140401 (prerelease)
GNU ld (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) 2.24.0.20140311 Linaro 2014.03
MStar # 

U-Boot 2015.01 (Feb 26 2019 - 10:53:16)
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) 4.8.3 20140401 (prerelease)
GNU ld (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) 2.24.0.20140311 Linaro 2014.03
MStar # help
?       - alias for 'help'
base    - print or set address offset
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
cmp     - memory compare
cp      - memory copy
crc32   - checksum calculation
dbg     - set debug message level. Default level is INFO
dcache  - enable or disable data cache
debug   - Disable uart rx via PAD_DDCA to use debug tool
dhcp    - boot image via network using DHCP/TFTP protocol
dstar   - script via SD/MMC
eeprom  - EEPROM sub-system
env     - environment handling commands
estar   - script via network
estart  - EMAC start
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatread - FAT fatread with FSTART
fatsize - determine a file's size
go      - start application at address 'addr'
gpio    - Config gpio port
help    - print command description/usage
i2c     - I2C sub-system
icache  - enable or disable instruction cache
initDbgLevel- Initial varaible 'dbgLevel' 
loop    - infinite loop on address range
macaddr - setup EMAC MAC addr
md      - memory display
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mssdmmc - Mstar SD/MMC IP Verification System
mstar   - script via TFTP
mw      - memory write (fill)
mxp     - MXP function for Mstar MXP partition
net_upgrade- do net update from the specified file that is in tftpserver

nm      - memory modify (constant address)
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
reset   - Perform RESET of the CPU
riu     - riu  - riu command

run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
sdupgrade- do SD card auto upgrade - one.bin

setenv  - set environment variables
sf      - SPI flash sub-system
sfbin   - for uploading sf image to a server(via network using TFTP protocol)
srcfg   - sensor pin and mclk configuration.
tftpboot- boot image via network using TFTP protocol
version - print monitor, compiler and linker version
MStar # version

U-Boot 2015.01 (Feb 26 2019 - 10:53:16)
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) 4.8.3 20140401 (prerelease)
GNU ld (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) 2.24.0.20140311 Linaro 2014.03
MStar #

skylarhays, 2019-10-26:
let me know what steps to take next.

roleoroleo, 2019-10-26:
It's very very similar.

- Same flash size
- Same toolchain
- Same kernel version
- Same partition schema
- Different file suffix: h201c instead of y203c

Now we should extract sys (rootfs) and home partition from uboot. With these files we are able to reconstruct new files to load from SD.
Try to follow this guide as a suggestion:
https://reverseengineering.stackexchange.com/questions/6300/extracting-a-firmware-image-via-u-boot
But there are a lot of examples if you google "dump spi flash from uboot".
The problem is that we don't have an ethernet port and the list of available commands is short.

It should be something like that:
sf probe to init spi bus
sf read to copy spi content to memory
md to dump memory (this command is very slow!!!)

We miss the SPI NOR offset, could you post a printenv from uboot?

Sorry but I can't help you using my camera because I broke uart pads and I can no longer reconnect to it.

skylarhays, 2019-10-26:
sure i will do printenv now. and i will start trying to dump the uboot, do i need a sd card inserted for this?

skylarhays, 2019-10-26:
MStar # printenv
baudrate=115200
bootcmd=sf probe 0;sf read 0x21000000 0x50000 0x00210000;bootm 0x21000000
bootdelay=0
cpu_part_start=143b0000
filesize=0
home_h201c_crc32=0xa64f6eb9
home_h307_crc32=0xaae53d57
kernel_h307_crc32=0x3ab2362d
one.bin_crc32=0x119a2697
sf_kernel_size=180000
sf_kernel_start=50000
sf_part_size=c30000
sf_part_start=3b0000
stderr=serial
stdin=serial
stdout=serial
sys_h307_crc32=0x8c04e1c9
uboot_h307_crc32=0x80065c93

Environment size: 463/4092 bytes

skylarhays, 2019-10-26:
sf read and sf probe both work

MStar # sf probe
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB

roleoroleo, 2019-10-26:
No you have to capture bytes from the screen...
Yes, it's very boring.
Enable logging and dump the memory.
I think you'll have to do it in "little" pieces (probably 65536 is enough).

Reading your bootlog the command should be:
```
sf probe
sf read 0x21000000 0x1d0000 0x1e0000
md 0x21000000 65536
md 0x21010000 65536
md 0x21020000 65536
....
md 0x211d0000 65536
```
for rootfs and
```
sf probe
sf read 0x21000000 0x3b0000 0xc30000
md 0x21000000 65536
md 0x21010000 65536
md 0x21020000 65536
....
md 0x21c30000 65536
```
for home partition.

roleoroleo, 2019-10-26:
Or you could write a little python program that does this for you.

skylarhays, 2019-10-26:
i am doing it manually now and saving the logs bootlog and rootfs. What next?

roleoroleo, 2019-10-26:
Send me the logs.
rootfs and home partitions do not contain personal data.
I will recreate partitions and start the hack.

skylarhays, 2019-10-26:
awesome! how do i directly e-mail or msg you?

roleoroleo, 2019-10-26:
You can use a dropbox/googledrive or other cloud system to share your files.

skylarhays, 2019-10-26:
https://drive.google.com/open?id=1bH6RbMOJXzbvz5BMOHiosRVdMP7ytIZe

The bootlog worked without issue. the rootfs reboots before it completes on several of the commands. do i need to dump it in shorter segments? the bootlog is in the google drive folder and i will put the rootfs there when i get it dumped correctly.

roleoroleo, 2019-10-26:
Unfortunately there are some problems in the file ...
For example:
```
21000d90: 01fcf7e8 81556c2d 73fe9a42 698be625    ....-lU.B..s%..i
21000da0: 56151443 2b660e23 baef5ad4 480bf420    C..V#.f+.Z.. ..H
21000db0: d4794f79 d4e0fc3c ff6878ff 11cf27b9    yOy.<....xh..'..
21000dc0: 81a8abfa d4f37de6 2eedb40505c 121efd00 ffffff32    .N.F\.......2...
21001050: e0011985 0000002b 7d266ee6 00000001    ....+....n&}....
21001060: 00000009 0000000b 5c7c9e90 00000403    ..........|\....
21001070: cd46c044 564ee2de ff62696c e0021985    D.F...NVlib.....
```
or 
```
210019f0: 0000a1ff 03e903e9 00000007 5c7c9e90    ..............|\
21001a00: 5c7c9e90 5c7c9e90 00000000 00000007    ..|\..|\........
21001a10: 00000007 00000000 186b259f 636a09d3    .........%k...jc
21001a20: 79737562 ff786f62 e0011985 0000002a    busyb
21001ca0: 7ce0e2f1 9a7f06c6 247e0112 01ccb49a    ...|......~$....
21001cb0: 30635d52 420d3325 9abcd2b2 fdc6d838    R]c0%3.B....8...
21001cc0: 9b025929 5da7ea84 5d50e153 25372188    )Y.....]S.P].!7%
```

roleoroleo, 2019-10-26:
Probably the serial port without flow is too slow.

skylarhays, 2019-10-26:
im going to try again on my desktop now. was doing it on the couch in the living room....

roleoroleo, 2019-10-26:
There are no particular settings.
If it doesn't work we need to write a little python program that uses very short blocks (about 256 or 512 bytes).

roleoroleo, 2019-10-26:
https://github.com/IonAgorria/u-dump/blob/master/main.py
https://github.com/gmbnomis/uboot-mdb-dump

skylarhays, 2019-10-26:
sure. do I run it on the camera?

roleoroleo, 2019-10-26:
No, on your pc.
You have to use putty to stop the boot.
Then, close putty and run the script.
Probably you need python 3 installed.

skylarhays, 2019-10-26:
Ok I will do this when I get home anything specific I plug in?

skylarhays, 2019-10-27:
py main.py
  File "main.py", line 7
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax

This is what im getting when i try to run it. i have installed python 3.8.0

roleoroleo, 2019-10-27:
With python 3.6 it works.

```
Python 3.6.0 (v3.6.0:41df79263a11, Dec 23 2016, 08:06:12) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
=============== RESTART: C:\Users\user\Desktop\main.py ===============
usage: main.py [-h] [--step STEP] [--size SIZE] [--timeout TIMEOUT]
               [--previous PREVIOUS] [--ignore-log] [--debug] [--reset]
               port baud start end
main.py: error: the following arguments are required: port, baud, start, end
>>> 
```

skylarhays, 2019-10-27:
py main.py com4 115200 0x21000000 0x211d0000

This is the syntax im trying and im getting

C:\mstar>py main.py com4 115200 0x21000000 0x211d0000
Traceback (most recent call last):
  File "main.py", line 264, in <module>
    main()
  File "main.py", line 244, in main
    log = open(name + ".log", "w")
OSError: [Errno 22] Invalid argument: '2019-10-27T04:16:23 0x21000000 0x211d0000.log'

I had to install pyserial as well... i think i just have the syntax wrong.

skylarhays, 2019-10-27:
I also tried this 

C:\mstar>py main.py --debug com4 115200 0x1d0000 0x1e0000
Debug mode enabled
Traceback (most recent call last):
  File "main.py", line 264, in <module>
    main()
  File "main.py", line 244, in main
    log = open(name + ".log", "w")
OSError: [Errno 22] Invalid argument: '2019-10-27T04:30:52 0x1d0000 0x1e0000.log'

roleoroleo, 2019-10-27:
Ok, try to change line 240 from
`name = datetime.fromtimestamp(time()).strftime("%Y-%m-%dT%H:%M:%S") + " " + hex(opts.start) + " " + hex(opts.end)`
to
`name = datetime.fromtimestamp(time()).strftime("%Y-%m-%dT%H%M%S") + "_" + hex(opts.start) + "_" + hex(opts.end)`

skylarhays, 2019-10-27:
changed that line out and now i get this

C:\mstar>py main.py com4 115200 0x1d0000 0x1e0000
Traceback (most recent call last):
  File "main.py", line 264, in <module>
    main()
  File "main.py", line 248, in main
    raise e
  File "main.py", line 246, in main
    data = dump(serial, opts, log)
  File "main.py", line 155, in dump
    addr, line_data, text = parse_line(line, opts.size)
  File "main.py", line 16, in parse_line
    raise Exception("Line size %s doesn't match %s: %s" % (len(line), size, visible(line)))
Exception: Line size 21 doesn't match 67: 001d0000:data abort

C:\mstar>py main.py com4 115200 0x21000000 0x211d0000
Traceback (most recent call last):
  File "main.py", line 264, in <module>
    main()
  File "main.py", line 248, in main
    raise e
  File "main.py", line 246, in main
    data = dump(serial, opts, log)
  File "main.py", line 136, in dump
    assert len(chunk) > 0
AssertionError

roleoroleo, 2019-10-27:
Ok.
I need to test the script.

roleoroleo, 2019-10-27:
Try this:

```
#!/usr/bin/env python3
__author__ = 'Ion Agorria'
import argparse
from datetime import datetime
from time import time
from serial import Serial

def visible(text):
    text.replace("\n", "\\n")
    text.replace("\r", "\\r")
    return text

def parse_line(line, size):
    # Break down each part
    if len(line) != size:
        raise Exception("Line size %s doesn't match %s: %s" % (len(line), size, visible(line)))
    i = 0
    addr = line[i:i + 8]
    i += 8
    assert line[i:i + 2] == ": "
    i += 2
    hex_data = line[i:i + 8]
    i += 8
    assert line[i:i + 1] == " "
    i += 1
    hex_data += line[i:i + 8]
    i += 8
    assert line[i:i + 1] == " "
    i += 1
    hex_data += line[i:i + 8]
    i += 8
    assert line[i:i + 1] == " "
    i += 1
    hex_data += line[i:i + 8]
    i += 8
    assert line[i:i + 4] == "    "
    i += 4
    text = line[i:i + 16]
    i += 16
    assert line[i:i + 2] == "\r\n"

    #Convert hex string into actual data
    data = []
    for i in range(0, len(hex_data), 2):
        piece = hex_data[i:i + 2]
        data.append(int(piece, 16))

    #For redundancy check if data matches text, but only for printable characters (non dot)
#    for i, piece in enumerate(data):
#        if text[i] != ".":
#            assert chr(piece) == text[i]

    return int(addr, 16), data, text

def write(serial, opts, data):
    if opts.debug:
        print("Debug write: " + visible(data))
    data = bytes(data, "ascii")
    serial.write(data)

def dump(serial, opts, log):
    data = []
    finish = False
    start_addr = opts.start
    last_addr = start_addr - 0x10
    do_reset = False

[... 175 lines cut ...]
        data = dump(serial, opts, log)
    except Exception as e:
        raise e
    finally:
        serial.close()
        log.close()

    #Write to file
    file = open(name + ".img", "wb")
    try:
        for line in data:
            file.write(line)
    except Exception as e:
        raise e
    finally:
        file.close()

main()
```

with timeout=1 and step=64
Pay attention: the 2nd parameters is not the length but the end address. So the arguments will be:
0x1d0000 0x3b0000
and
0x3b0000 0xfe0000

skylarhays, 2019-10-27:
ok. much more progress this time but something is not quite right... it takes about 3 seconds before it errors out and the camera reboots.

C:\mstar>py here.py --timeout 1 --step 64 com4 115200 0x1d0000 0x3b0000
Traceback (most recent call last):
  File "here.py", line 264, in <module>
    main()
  File "here.py", line 248, in main
    raise e
  File "here.py", line 246, in main
    data = dump(serial, opts, log)
  File "here.py", line 155, in dump
    addr, line_data, text = parse_line(line, opts.size)
  File "here.py", line 16, in parse_line
    raise Exception("Line size %s doesn't match %s: %s" % (len(line), size, visible(line)))
Exception: Line size 21 doesn't match 67: 001d0000:data abort

roleoroleo, 2019-10-27:
I don't understand.
Could you post the output of the md command executed manually with putty?
Or could you enable debug?

roleoroleo, 2019-10-27:
Sorry, I didn't read correctly your message.
You have to execute the command with the base address where you copied the data with sf read command.
So if the sf read is:
sf read 0x21000000 0x1d0000 0x1e0000
the python program will be:
py here.py --timeout 1 --step 64 com4 115200 0x21000000 0x211e0000

If the sf read is:
sf read 0x21000000 0x3b0000 0xc30000
the python program will be:
py here.py --timeout 1 --step 64 com4 115200 0x21000000 0x21c30000

skylarhays, 2019-10-27:
It appears to be running successfully now. I will share files as soon as it is complete.

skylarhays, 2019-10-28:
Ok The first one dumped without any issues at all. the second one almost completed. I put both here https://drive.google.com/drive/folders/1bH6RbMOJXzbvz5BMOHiosRVdMP7ytIZe    If i need to re run for the rootfs I will. 

This is what happened on the 2nd part

0x2121DFD0  55 2B B7 38 B2 54 8E D5 67 C1 13 D8 3A AF 88 DE |8.+U..T....g...:|
0x2121DFE0  BC AB F1 9D 4A E9 34 B0 DE 2F 0D 72 0E 18 72 54 |.....4.Jr./.Tr..|
Traceback (most recent call last):
  File "here.py", line 264, in <module>
    main()
  File "here.py", line 248, in main
    raise e
  File "here.py", line 246, in main
    data = dump(serial, opts, log)
  File "here.py", line 155, in dump
    addr, line_data, text = parse_line(line, opts.size)
  File "here.py", line 16, in parse_line
    raise Exception("Line size %s doesn't match %s: %s" % (len(line), size, visible(line)))
Exception: Line size 39 doesn't match 67: 274ddd94 3465d98d    .*.=.'....MW..e4

skylarhays, 2019-10-28:
I started the 2nd file again at the point it stopped 21378fe0 and have it running to 0x21c30000. I will put the 3rd file that completes the rootfs dump as soon as it completes.

roleoroleo, 2019-10-28:
Ok, the rootfs is pretty much the same.
Try this:
https://drive.google.com/open?id=1vwbxTLpOpw23qJQEv10QXcwhLPZu7a8B
You shuold have telnet enabled.

skylarhays, 2019-10-28:
so use this sys_h201c with what other file?

roleoroleo, 2019-10-28:
Only sys_h201c.
At the moment the home partition is not modified.

skylarhays, 2019-10-28:
so just put this on a sd card and put it in camera and power up? im not
certain how to enable telnet?

skylarhays, 2019-10-28:
The rest of the rootfs is still dumping. I started where the file stops in
2.log. will you need that? I will stick it in the google share folder as
soon as it completes... it looks like there are a lot of FFs near the end.

roleoroleo, 2019-10-28:
Put the file on sd and power up.
Then use putty to connect to the cam using telnet (not ssh).
This operation is dangerous and the chamber may no longer start.
I did my best but I can't assure you that it works.

Yes I need the 2nd dump because contains the home partition: the most important part of the system.

skylarhays, 2019-10-28:
ok the last part of the rootfs is in the google drive folder. I will try to flash the camera later tonight when  I get home.

roleoroleo, 2019-10-28:
I prepared home partition also.
Let me know if sys will load correctly.
After this, i will send the full hack.

skylarhays, 2019-10-28:
Ok will do. Hopefully my cam doesnt get bricked. It will be later this
evening before i can try. I will post an update tonight.

skylarhays, 2019-10-28:
It successfully flashed the sys_h201c i watched it over serial with putty. what ip do i connect to with telnet?

skylarhays, 2019-10-28:
telnet is working!   i have a login prompt!

roleoroleo, 2019-10-29:
Ok. Now try these:
https://drive.google.com/open?id=17pGpE0GxQ2OIT6vK5m5uHDKR8aqwTZ2W
https://drive.google.com/open?id=1AFOQb_ZXIzAohbf6RjdhToTlvad8Z_wZ

roleoroleo, 2019-10-29:
Before trying the new files, make a backup copy of the entire flash.
Insert a SD card and execute dd:
```
dd if=/dev/mtd0 of=/tmp/sd/mtd0.bin
dd if=/dev/mtd1 of=/tmp/sd/mtd1.bin
dd if=/dev/mtd2 of=/tmp/sd/mtd2.bin
dd if=/dev/mtd3 of=/tmp/sd/mtd3.bin
dd if=/dev/mtd4 of=/tmp/sd/mtd4.bin
dd if=/dev/mtd5 of=/tmp/sd/mtd5.bin
```

skylarhays, 2019-10-29:
dd doesnt seem to work i get this

MStar # dd if=/dev/mtd0 of=/tmp/sd/mtd0.bin
Unknown command 'dd' - try 'help'

I have 3 of these cameras I will get a dump from one of them before
flashing. I went ahead and flashed. IT WORKS! telnet,ssh, and web access.
Thanks so much for putting in the effort of making this happen on a camera
you don't own. There will be quite a few people happy this is going to
finally be available. If you know what im doing wrong with DD let me know
so i can get a full backup of one of these cameras before i flash my other
2. Thanks again!

roleoroleo, 2019-10-29:
You have to run dd on a linux shell and not on uboot shell.

1. Load the 1st sys_h201c that I sent you (only sys, not home)
2. Login with telnet
3. Execute dd

But you must use another camera to have a "real" backup.

bart11064, 2019-10-29:
Nice work guys ! I was Following the conversation crossing finger that it works.
Thanks to both of you for your work and time!
Do you think this hack will work on a 6FUS yi dome 1080p with 4.6.0 firmware, or is it reserved for yi home only ?
Thanks again guys !

roleoroleo, 2019-10-29:
@skylarhays 
When you have time I would like to know if the main services are working correclty.
Specially rtsp, snapshot and mqtt.

@bart11064 
I don't know.
It depends on whether the platform is the same and if someone provides a flash dump.

bart11064, 2019-10-29:
I could try that this evening.
I have this interface at home : http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=8&pcid=41
Do you think it will work ?

skylarhays, 2019-10-29:
This actually is a black yi dome security 1080p i did the hack on. 4.6.0.

@roleoroleo I telneted into the cam and it gave the same error, i must have still been on uboot. I will try to test everything out tonight after work. Thanks again for the work! this is amazing to finally have use of these cams and be able to take them out of the cloud. I don't see the ONVIF stuff anywhere?

bart11064, 2019-10-29:
I'd be happy to try it out myself if needed 😀

[27 more replies]

closed 2019-10-30 by skylarhays