roleoroleo/yi-hack-MStar #11: Login & Port

Montaraz1, 2019-10-23:
Hi,
id like to see camera from outside.
Can i add username and password for access (even RTSP) and change RTSP and HTTP port?

roleoroleo, 2019-10-24:
At the moment yo can't set RTSP authentication and RTSP port.
You can change manually the tcp port of HTTP server and ONVIF server editing
/home/yi-hack/script/system.sh
file.
I think it will not so complicated to add it in the future.

Montaraz1, 2019-10-24:
ok thanks

roleoroleo, 2019-10-25:
Try the 0.1.6.

Montaraz1, 2019-10-27:
thanks man.
you added RTSP port too.

I'm not expert of github.. how can i do a small donation to your project?

Montaraz1, 2019-10-27:
I have a problem :(
I think password's field cant manage special characters like @!^... or maybe i mistake setting password...i dont know.

The problem is that i have resetted the cam and now it connect to wifi only via QR code.
Every time i reconnect the camera i need to use QR code in YiCamera App to connect to my Wifi.

How can i fix? i cant find a stock firmware too.

thanks

roleoroleo, 2019-10-28:
About special chars, I will check it.
For your problem, you don't need a stock firmware, because the difference between stock and hack is very little.
Try to restore the "normal" behavior of the stock firmware: enable cloud and disable all unneeded services (except httpd and ssh that must be enabled).
Then restarts and check if it works.

roleoroleo, 2019-10-29:
What's BFUS? Home? Dome?

GarretBlueMango, 2019-10-29:
@roleoroleo I see that @Montaraz1 gave a confirmation that your hack also works on the BFUS variant of the Yi Dome Camera 1080P. I'm going to try it and will report on the results

Montaraz1, 2019-10-29:
> What's BFUS? Home? Dome?

1080p home camera as 6FUS

closed 2019-11-02 by Montaraz1