roleoroleo/yi-hack-MStar #13: Motion detection stops working after a while

edofullin, 2019-10-28:
I think the title doesn't really need explanation. 
Is mqtt motion detection supposed to work? 
For me it stops working after 1-2 days of uptime. 

I am using the version 0.1.6 of the firmware on a 6FUS 1080p Yi home camera.
I don't have physical access to the camera right now but I do have SSH access if I need to run commands, I would be very happy to contribute if I could!

Thanks for porting the firware to the 6FUS

roleoroleo, 2019-10-28:
I copied the mqtt feature from TheCrypt0 yi-hack-v4 and I don't know how it works.
I use it with hass and, for my experience, it's working correctly.
Please, post your config and, if possible, the output of top command.
Do you have an active rtsp client (always active)?

edofullin, 2019-10-29:
I'm using it with hass too.
Here is my config:
```
stream:
ffmpeg:
camera:
  - platform: ffmpeg
    name: yi-camera
    input: -rtsp_transport tcp -i rtsp://10.0.0.216/ch0_1.h264
```
So, yes I guess it is always active.
Are there better ways to do that? A picture every 30s-1m would be fine as well.

And here is the output of top:
```
Mem: 58796K used, 2028K free, 2056K shrd, 720K buff, 11792K cached
CPU: 22.0% usr 14.0% sys  0.0% nic 63.9% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 1.64 1.49 1.56 1/100 2361
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  148     1 root     S    47024 76.9   0 33.3 ./rmm
  384     1 root     S     6540 10.7   0  0.9 rRTSPServer
 2330     1 root     S    11676 19.1   0  0.3 ./p2p_tnp
 1340  1334 root     R     2064  3.3   0  0.3 top
    7     2 root     SW       0  0.0   0  0.3 [rcu_preempt]
 1308   337 root     S     2012  3.2   0  0.1 dropbear -R
    3     2 root     SW       0  0.0   0  0.1 [ksoftirqd/0]
  292     2 root     SW       0  0.0   0  0.1 [RTW_CMD_THREAD]
 2324     1 root     S     9220 15.0   0  0.0 ./cloud
  242     1 root     S     7612 12.4   0  0.0 ./mp4record
  245     1 root     S     6680 10.9   0  0.0 ./oss
  139     1 root     S     4036  6.6   0  0.0 ./dispatch
  346     1 root     S     3576  5.8   0  0.0 mqttv4
  272     1 root     S     2344  3.8   0  0.0 telnetd
 2248     1 root     S     2256  3.6   0  0.0 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplic
  378     1 root     S     2200  3.6   0  0.0 ntpd -p pool.ntp.org
 1334  1308 root     S     2060  3.3   0  0.0 -sh
  267     1 root     S     2056  3.3   0  0.0 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
 2276     1 root     S     2056  3.3   0  0.0 udhcpc -i wlan0 -b -s /home/app/script/default.script
  337     1 root     S     1728  2.8   0  0.0 dropbear -R
  299     1 root     S     1660  2.7   0  0.0 pure-ftpd (SERVER)
  138     1 root     S     1516  2.4   0  0.0 ./log_server
  247     1 root     S     1504  2.4   0  0.0 ./watch_process
  383     1 root     S     1440  2.3   0  0.0 h264grabber low
  444     1 root     S     1372  2.2   0  0.0 ./arp_test
   39     1 root     S      408  0.6   0  0.0 /bin/ueventd
    1     0 root     S      384  0.6   0  0.0 /init
   74     2 root     SW       0  0.0   0  0.0 [mmcqd/0]
  239     2 root     SW       0  0.0   0  0.0 [kworker/u2:3]
   17     2 root     SW       0  0.0   0  0.0 [kswapd0]
   32     2 root     SW       0  0.0   0  0.0 [VIPDazaTask]
   40     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd3]
   10     2 root     SW       0  0.0   0  0.0 [watchdog/0]
   16     2 root     SW       0  0.0   0  0.0 [kworker/0:1]
   41     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd2]
  740  2324 root     Z        0  0.0   0  0.0 [cloudAPI]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
    5     2 root     SW<      0  0.0   0  0.0 [kworker/0:0H]
    6     2 root     SW       0  0.0   0  0.0 [kworker/u2:0]
    8     2 root     SW       0  0.0   0  0.0 [rcu_sched]
    9     2 root     SW       0  0.0   0  0.0 [rcu_bh]
   11     2 root     SW<      0  0.0   0  0.0 [khelper]
   12     2 root     SW<      0  0.0   0  0.0 [writeback]
   13     2 root     SW<      0  0.0   0  0.0 [crypto]
   14     2 root     SW<      0  0.0   0  0.0 [bioset]
```

The MQTT detection feature works flawlessly for 5 or 6 hours after reboot so I guess the MQTT config on the camera is set correctly.

Thanks for your help

roleoroleo, 2019-10-29:
The hass cfg is ok. The rtsp on hass isn't always active but only when you open stream in lovelace.
And indeed there isn't a memory problem: rmm is a camera native application and it is normal that it requires a lot of memory. But h264grabber is low and rRTSPServer is low.
I will investigate.

roleoroleo, 2019-11-01:
I'm sorry but I can't replicate your problem.
My cam has been online for 5 days and motion detection is working properly.
I can not understand.

edofullin, 2019-11-01:
I'll investigate further then. 
I'm using a 6FUS camera with the latest firmware (0.1.7)

Is there a way to log the values of motion detection on a file? That is to check whether is problem is with motion detection itself or mqtt.

Also, is there a way to adjust the detection "tolerance", does the setting from the official app still work?

Thank you for your effort

roleoroleo, 2019-11-01:
If you kill mqttv4 and restart it, you can see the log on console.
In the hack part, there aren't settings to adjust motion detection.
But the official app still work. The mqttv4 daemon simply intercepts the detection made by the native yi application.

edofullin, 2019-11-01:
Thank yo, I'll let you know in a few days if I find something

edofullin, 2019-11-10:
Running top now it seems like mqttv4 is not running for some reasons. 
Is it supposed to be running all time?

```
/home/yi-hack # ps
PID   USER     TIME  COMMAND
    1 root      0:02 /init
    2 root      0:00 [kthreadd]
    3 root      0:03 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root      0:17 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:00 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [kworker/0:1]
   17 root      0:02 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:00 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   74 root      0:05 [mmcqd/0]
   85 root      0:00 [cryptodev_queue]
   96 root      0:00 [spi0]
  121 root      0:00 [kworker/0:2]
  128 root      0:00 [cfg80211]
  138 root      0:00 ./log_server
  139 root      0:13 ./dispatch
  148 root     23:46 ./rmm
  225 root      1:52 ./mp4record
  228 root      0:01 ./oss
  229 root      0:00 ./watch_process
  236 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  261 root      0:00 pure-ftpd (SERVER)
  284 root      0:04 [RTW_CMD_THREAD]
  291 root      0:02 [kworker/u2:3]
  297 root      0:00 dropbear -R
  340 root      0:03 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
  361 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script
  366 root      0:00 ntpd -p pool.ntp.org
  373 root      0:09 h264grabber -r high
  374 root      0:47 rRTSPServer
  375 root      0:00 dropbear -R
  379 root      0:00 onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Home 1080p --manufacturer Yi --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S 
  383 root      0:00 -sh
  429 root      0:01 ./cloud
  433 root      0:15 ./p2p_tnp
  436 root      0:00 ./arp_test
  881 root      0:00 ps
```
Also, how does mqtt intecept data from the detection process? Is there a sort of dbus?
I tried to look in the code but I cannot find where that happens.

Sorry for disturbing you but i really want to solve this. I use the camera as my home alarm

Thnak you

roleoroleo, 2019-11-10:
Also on my cam I noticed that mqttv4 is not running.
I'm trying to collect logs to solve the problem.
But is not simple because the code isn't mine.

edofullin, 2019-11-10:
Oh, at least I'm not the only one. 
If you need help or testing feel free to ask me.

roleoroleo, 2019-12-06:
Please try 0.2.2.
I patched mqttv4.

edofullin, 2019-12-06:
I just installed, I'll let you know

edofullin, 2019-12-11:
It has been working flawlessly for a few days now. 

Good job and thank you very much for your help

closed 2019-12-11 by edofullin