roleoroleo/yi-hack-MStar #15: Restore original firmware

feiticeir0, 2019-10-30:
I know this is not an issue, but I was wondering if it is possible to restore the original firmware of the camera. 

Thank you

roleoroleo, 2019-10-30:
Only if you have a backup.
The file you can download from xiao website is not a full firmware.

feiticeir0, 2019-10-30:
I do have the original firmware, from here:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/23321022068

roleoroleo, 2019-10-30:
The link is not valid.
But if you have the files, put them on SD and switch on.

feiticeir0, 2019-10-30:
I did, but no luck. Doesn't matter. 
I've upgraded with your files to the latest version. My only issue was that the stream was a bit slow... Don't know why... I'm integrating it with home assistant - how does the mqtt motion sensor works ? it sends a mqtt packet when it detects motion ?

roleoroleo, 2019-10-30:
Mqtt is a feature copied from TheCrypt0 yi-hack-v4 and is based on original feature.
The program intercepts a ipc communication between native processes and sends a mqtt packet.

closed 2019-11-03 by roleoroleo