roleoroleo/yi-hack-MStar #17: WIFI not reconnecting on reboot

reptil1990, 2019-10-30:
The Wifi doesn`t seem to reconnect to the same hotspot after reboot.
Can that be fixed by entering the information in a file via SSH?

feiticeir0, 2019-10-30:
Have the same problem with the latest version... After a reboot, camera just talks "Waiting to connect"

reptil1990, 2019-10-30:
Yes same for me!

feiticeir0, 2019-10-30:
I'm having now the same issue not only with the latest version, but also 0.1.6

feiticeir0, 2019-10-30:
and 0.1.5... It does not save wifi configurations...

roleoroleo, 2019-10-31:
Camera model?
Try to enable cloud and re-execute pairing.

reptil1990, 2019-10-31:
> Camera model?
> Try to enable cloud and re-execute pairing.

Will try later today!

feiticeir0, 2019-10-31:
> Camera model?
> Try to enable cloud and re-execute pairing.

Will try later.
Yi Home 1080p

22MarioZ, 2019-11-02:
Same here. After reboot, camera is waiting for connect. With Yi app I can "successfully" connect to WiFi, but the camera is never shown on network. Also paring with the App is not successful. I have upgrade logs if needed.

22MarioZ, 2019-11-02:
I had same problem, not connection to WiFi and loosing the connection after reboot. After couple of retries the camera connected to network and I was able to open web interface. I enable cloud services so that Yi app and camera paired. After that the camera didn't lost WiFi anymore.

roleoroleo, 2019-11-02:
In the next release I will change the cloud default setting.
So you should be able to repeat pairing.

stevekanter, 2019-11-03:
I solved this kinda annoyingly but:

1. Factory reset
2. Flashed hacked firmware
3. Connected to wifi from app
4. Went to configuration and disabled Disable Cloud
5. Rebooted
6. Re-connected to wifi and paired fully
7. Re-enabled Disable Cloud

GuitarSkater, 2019-11-04:
> I solved this kinda annoyingly but:
> 
> 1. Factory reset
> 2. Flashed hacked firmware
> 3. Connected to wifi from app
> 4. Went to configuration and disabled Disable Cloud
> 5. Rebooted
> 6. Re-connected to wifi and paired fully
> 7. Re-enabled Disable Cloud

I tried this about a dozen times, it just times out on the pairing. Any other ideas?

stevekanter, 2019-11-05:
> I tried this about a dozen times, it just times out on the pairing. Any other ideas?

Does it successfully read the QR code but just fails to pair? I never experienced THAT. Albeit what I did experience was a hell-of-a-time getting the camera to read the QR code off of my phone. I ended up just sending them to my computer and then the camera read them off the computer no problem. I did this exact set of steps with my 4 different cameras, never a single problem.

GuitarSkater, 2019-11-05:
Yep, it gets all the way to the successful wifi, I can see it on the net, access the config via browser, it just refuses to pair with the app and times out and reboots. 

I went in and made sure cloud wasn't disabled to no avail. I think I'll just have to wait for the next release

roleoroleo, 2019-11-05:
Please try 0.1.8.

GuitarSkater, 2019-11-06:
> Please try 0.1.8.

Can Confirm, Back in action! I also noticed on 1.7 it would periodically reboot, so far that hasnt happened.

Thanks for the great support!!!

closed 2019-11-27 by roleoroleo

LouiS22, 2019-12-01:
I still notice this with 0.2 and with a Yi 1080 camera.