roleoroleo/yi-hack-MStar #18: BFUS black 1080p Dome Security Cam

skylarhays, 2019-10-30:
I will get started on this tonight. do you advise to  use the python program or try with putty first? I will post the bootlog and the shell if available as soon asap.

skylarhays, 2019-10-30:
bootlog

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.10.30 18:54:15 =~=~=~=~=~=~=~=~=~=~=~=
 àü à ü à 
IPL gd156225
D-01.

HW Reset
64MB

BIST0_0001-OK

[SDMMC] NO SD!
[NOR]
offset:00010000

size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4

MXP found at 0x00020000

[SDMMC] NO SD!
  decomp_size=0x000414d4

-----------------------U-Boot 2015.01 (Feb 26 2019 - 10:53:16)-----------------------

Version: I3g138a6f9
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one.bin 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    one.bin Not exist(fatsize err) 

+++++++++++++++++++   check one_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    one_h201c Not exist(fatsize err) 

+++++++++++++++++++   check uboot_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 uboot_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    uboot_h201c Not exist(fatsize err) 

+++++++++++++++++++   check kernel_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
[... 271 lines cut ...]
MSYS: DMEM request: [VSPL-I0P2B1]:0x0005A000
MSYS: DMEM request: [VSPL-I0P2B2]:0x0005A000
MSB2@v1.1-01:r&d analysis.
MSYS: DMEM request: [MS-00]:0x00357000
MSYS: DMEM request: [MS-01]:0x00357000
MSYS: DMEM request: [MS-02]:0x00357000
MSYS: DMEM request: [VENC-32]:0x000FF000
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
MSYS: DMEM request: [S0:VENCDMOUT]:0x00006400
MSB2@v1.1-01:r&d analysis.
MSYS: DMEM request: [VSPL-I0P1B0]:0x0005A000
MSYS: DMEM request: [VSPL-I0P1B1]:0x0005A000
MSYS: DMEM request: [VSPL-I0P1B2]:0x0005A000
MSYS: DMEM request: [VENC-48]:0x00025800
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
MSYS: DMEM request: [S1:VENCDMOUT]:0x00005600
MSYS: DMEM request: [S1:VENCDMP0]:0x00056400
MSYS: DMEM request: [S1:VENCDMP1]:0x00056400

[PID_LIST] pid_list_init ok, [ ver=Feb 26 2019, 14:28:36 ] 
MSYS: DMEM request: [VSPL-I0P3B0]:0x00016800
MSYS: DMEM request: [VSPL-I0P3B1]:0x00016800
MSYS: DMEM request: [VSPL-I0P3B2]:0x00016800

skylarhays, 2019-10-30:
shell with printenv

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.10.30 18:55:49 =~=~=~=~=~=~=~=~=~=~=~=
 àü à ü à 
IPL gd156225
D-01.

HW Reset
64MB

BIST0_0001-OK

[SDMMC] NO SD!
[NOR]
offset:00010000

size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4

MXP found at 0x00020000

[SDMMC] NO SD!
  decomp_size=0x000414d4

-----------------------U-Boot 2015.01 (Feb 26 2019 - 10:53:16)-----------------------

Version: I3g138a6f9
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one.bin 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    one.bin Not exist(fatsize err) 

+++++++++++++++++++   check one_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    one_h201c Not exist(fatsize err) 

+++++++++++++++++++   check uboot_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 uboot_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    uboot_h201c Not exist(fatsize err) 

+++++++++++++++++++   check kernel_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 kernel_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    kernel_h201c Not exist(fatsize err) 

+++++++++++++++++++   check sys_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 sys_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    sys_h201c Not exist(fatsize err) 

+++++++++++++++++++   check home_h201c    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 home_h201c 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    home_h201c Not exist(fatsize err) 
MStar # 
MStar # 
MStar # 
MStar # 
MStar # 
MStar # printenv
baudrate=115200
bootcmd=sf probe 0;sf read 0x21000000 0x50000 0x00210000;bootm 0x21000000
bootdelay=0
cpu_part_start=143b0000
filesize=0
home_h201c_crc32=0xa64f6eb9
home_h307_crc32=0xaae53d57
kernel_h307_crc32=0x3ab2362d
one.bin_crc32=0x119a2697
sf_kernel_size=180000
sf_kernel_start=50000
sf_part_size=c30000
sf_part_start=3b0000
stderr=serial
stdin=serial
stdout=serial
sys_h307_crc32=0x8c04e1c9
uboot_h307_crc32=0x80065c93

Environment size: 463/4092 bytes
MStar #

roleoroleo, 2019-10-31:
The file name is h201c, the same as 6FUS.
Probably the other hack works.
If you want to be sure, dump the partitions with python.

skylarhays, 2019-11-01:
sorry been busy. I will dump rootfs tonight... likely will take until saturday though to complete.

roleoroleo, 2019-11-01:
Dump only the rootfs partition.
We will dump home partition later using dd.

```
sf read 0x21000000 0x1d0000 0x1e0000
py here.py --timeout 1 --step 64 com4 115200 0x21000000 0x211e0000
```

skylarhays, 2019-11-02:
Ok. Got the dump.

https://drive.google.com/open?id=1zBBO3lA7hSyS7IUAvKHixLlY8A0Ge87R

roleoroleo, 2019-11-02:
You dumped the wrong partition: home.
I need the rootfs first.
I think you executed:
```
sf read 0x21000000 0x3b0000 0xc30000
py here.py --timeout 1 --step 64 com4 115200 0x21000000 0x21c30000
```
and not
```
sf read 0x21000000 0x1d0000 0x1e0000
py here.py --timeout 1 --step 64 com4 115200 0x21000000 0x211e0000
```

skylarhays, 2019-11-02:
oops. dumping now.

skylarhays, 2019-11-02:
https://drive.google.com/open?id=1Vs-cKutmf5km6NI9EUmpBrDPWdiU9aeG

roleoroleo, 2019-11-03:
The dump is perfectly identical to the 6FUS 4.6.0.
You can use h201c files.

skylarhays, 2019-11-03:
awesome! Thanks! what is the DD command to dump this again from Telnet? i would like to get a full backup since this is the same as the others.

roleoroleo, 2019-11-03:
Copy all partitions to sd with the following commands:
```
dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin
dd if=/dev/mtd/mtd1 of=/tmp/sd/mtd1.bin
dd if=/dev/mtd/mtd2 of=/tmp/sd/mtd2.bin
dd if=/dev/mtd/mtd3 of=/tmp/sd/mtd3.bin
dd if=/dev/mtd/mtd4 of=/tmp/sd/mtd4.bin
dd if=/dev/mtd/mtd5 of=/tmp/sd/mtd5.bin
```
If you apply hack, mtd2 and mtd3 will be overwritten.
For a full backup use rootfs and home dumped with python before hack.
If you want I cam resend to you in binary form.
mtd2 = rootfs
mtd3 = home

Let me know if hack works without problems.

skylarhays, 2019-11-04:
So I should dump with python over serial? will the dd backup not work?

roleoroleo, 2019-11-04:
No, it's not necessary.
It's a dog chasing its own tail...

You can dump partitions with dd with a telnet connection.
But you can connect with telnet only after applying the update.
And if you apply the update, mtd2 and mtd3 will be overwritten, so when you dump them with dd, you don't dump the original partition but the hacked partition.

You should dump:
```
dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin
dd if=/dev/mtd/mtd1 of=/tmp/sd/mtd1.bin
dd if=/dev/mtd/mtd4 of=/tmp/sd/mtd4.bin
dd if=/dev/mtd/mtd5 of=/tmp/sd/mtd5.bin
```
and save the log you sent me previously instead of dd dump.
The format is different but it does not matter.

skylarhays, 2019-11-04:
Ok great. Thanks again! I will let you know the results when I update.

skylarhays, 2019-11-18:
Just wanted to give an update. I finally had time to do this. Everything is working great on BFUS.

roleoroleo, 2019-11-18:
Ok, thank you.

closed 2019-11-19 by roleoroleo

woniuzfb, 2024-12-30:
Anyone got a backup mtd4.bin ?