roleoroleo/yi-hack-MStar #23: 4FUS

enrysan0, 2019-11-04:
Hi,

Do you plan to port this hack to 4FUS (FW:4.2.0.0H_201909041620)? 
It is the Amazon version claimed as YI Camera 1080p version 3 (in Europe).
Here is an image of the camera disassembled:
https://media.discordapp.net/attachments/531894543759441935/635156208332046336/Yi_Home_1080p_v.3_4FUS.jpg?width=1082&height=596

Thanks

roleoroleo, 2019-11-04:
I don't know this camera.
If you want, I could check if it's the same platform.
But you have to dump a log as described here:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10

enrysan0, 2019-11-04:
Thanks for the answer roleoroleo. 
I don't have a "rs232/usb adapter" at the moment. However as soon as i have a bit of time I'll try to dump the log.

enrysan0, 2019-11-09:
Here is the dump:

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.11.09 10:35:21 =~=~=~=~=~=~=~=~=~=~=~=
 Ä    ‡¸ ‡ ¸ ‡ 
IPL gd156225
D-01.

HW Reset
64MB

BIST0_0001-OK

[SDMMC] NO SD!
[NOR]
offset:00010000

size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4

MXP found at 0x00020000

[SDMMC] NO SD!
  decomp_size=0x00041314

-----------------------U-Boot 2015.01 (Apr 28 2018 - 17:08:00)-----------------------

Version: I3ge2accce
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
MXIC REMS: 0xC2,0x17
Flash is detected (0x0509, 0xC2, 0x20, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
MXIC REMS: 0xC2,0x17
Flash is detected (0x0509, 0xC2, 0x20, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one.bin 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    one.bin Not exist(fatsize err) 

+++++++++++++++++++   check one_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one_y25 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    one_y25 Not exist(fatsize err) 

+++++++++++++++++++   check uboot_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 uboot_y25 
_[sdmmc_0] Card Detect Fail! 
** Bad device mmc 0 **
    uboot_y25 Not exist(fatsize err) 
[... 679 lines cut ...]
MT7601_ChipSwitchChannel: SwitchChannel#8(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#9(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#10(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#11(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#12(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#13(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#1(RF=15, 1T)
80211> cfg80211_scan_done
#####check WPS IE
ConnInfo.bWpsConnection ===> Not WPS IE
80211> Connect bssid e0:3f:49:25:6f:f8
MT7601_ChipSwitchChannel: SwitchChannel#9(RF=15, 1T)
PeerBeaconAtJoinAction(): HT-CtrlChannel=9, CentralChannel=>9
PeerBeaconAtJoinAction(): Set CentralChannel=9
MT7601_ChipSwitchChannel: SwitchChannel#9(RF=15, 1T)
random: wpa_supplicant urandom read with 58 bits of entropy available
Rcv Wcid(1) AddBAReq
Start Seq = 00000002
RTMP_TimerListAdd: add timer obj c4b1dafc!
RTMP_TimerListAdd: add timer obj c4b1badc!
RTMP_TimerListAdd: add timer obj c4b1bb1c!
RTMP_TimerListAdd: add timer obj c4b1bb5c!
hue, cpld_ioctl() cmd(6) invalid !!! 
hue, cpld_ioctl() cmd(6) invalid !!! 
```

enrysan0, 2019-11-09:
Here is another dump with a fat32 SdCard inserted:

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.11.09 11:00:39 =~=~=~=~=~=~=~=~=~=~=~=
  ‡‡¸Ä‡é‡‡‡é ‡ 
IPL gd156225
D-01.

HW Reset
64MB

BIST0_0001-OK

offset:00010000

size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4

MXP found at 0x00020000

  decomp_size=0x00041314

-----------------------U-Boot 2015.01 (Apr 28 2018 - 17:08:00)-----------------------

Version: I3ge2accce
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
MXIC REMS: 0xC2,0x17
Flash is detected (0x0509, 0xC2, 0x20, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
MXIC REMS: 0xC2,0x17
Flash is detected (0x0509, 0xC2, 0x20, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one.bin 
    one.bin Not exist(fatsize err) 

+++++++++++++++++++   check one_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one_y25 
    one_y25 Not exist(fatsize err) 

+++++++++++++++++++   check uboot_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 uboot_y25 
    uboot_y25 Not exist(fatsize err) 

+++++++++++++++++++   check kernel_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 kernel_y25 
    kernel_y25 Not exist(fatsize err) 

+++++++++++++++++++   check sys_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 sys_y25 
[... 666 lines cut ...]
MT7601_ChipSwitchChannel: SwitchChannel#6(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#7(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#8(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#9(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#10(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#11(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#12(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#13(RF=15, 1T)
MT7601_ChipSwitchChannel: SwitchChannel#1(RF=15, 1T)
80211> cfg80211_scan_done
#####check WPS IE
ConnInfo.bWpsConnection ===> Not WPS IE
80211> Connect bssid e0:3f:49:25:6f:f8
MT7601_ChipSwitchChannel: SwitchChannel#9(RF=15, 1T)
PeerBeaconAtJoinAction(): HT-CtrlChannel=9, CentralChannel=>9
PeerBeaconAtJoinAction(): Set CentralChannel=9
MT7601_ChipSwitchChannel: SwitchChannel#9(RF=15, 1T)
random: wpa_supplicant urandom read with 68 bits of entropy available
RTMP_TimerListAdd: add timer obj c4b1dadc!
Rcv Wcid(1) AddBAReq
Start Seq = 00000004
RTMP_TimerListAdd: add timer obj c4b1fafc!
RTMP_TimerListAdd: add timer obj c4b1db1c!
RTMP_TimerListAdd: add timer obj c4b1db5c!
```

roleoroleo, 2019-11-10:
The system is similar but the not enough.
You can't use the hack currently released.

Now you should dump the rootfs and the home partitions.

rado0x54, 2019-11-24:
Hey @enrysan0 and @roleoroleo,

I'll use this thread before opening a new one because my camera firmware is exactly the same as mentioned in this thread. More specifically:

1. I own a Yi Home Camera v3 (1080p) bought in the US
2. It's labelled as 9FUS
3. Current Firmware is 4.2.0.0F_201905241332 (but want's to upgrade to 4.2.0.0H_201909041620)
4. The current release (9FUS 4.5.0 is not working for me)

I'd like to support this thread and dump log output + partitions, but for the love of me I'm not finding any pins/pads labelled rx/tx/gnd (on either side). @enrysan0 I also don't see any on the image you provided. Would you be able to give me a hint on how you connected your serial adapter?

Thanks!

camonice, 2019-11-25:
> The system is similar but not enough.
> You can't use the hack currently released.
> 
> Now you should dump the rootfs and the home partitions.

I am following up this topic as I also have a 4FCN camera. No solution for it as far as I am aware of.

Regards, Camonice

Rocket200, 2019-11-28:
4FUS will be support soon?

roleoroleo, 2019-11-29:
Probably this one:
https://drive.google.com/open?id=1WwBl0n8LZpLjYdB6ZKpmh81_BBngR7LE
works with 4FUS 4.2.0 but I can't guarantee it.
It would be better if you dumped a backup before trying it. If you can.

camonice, 2019-11-29:
> Probably this one:
> https://drive.google.com/open?id=1WwBl0n8LZpLjYdB6ZKpmh81_BBngR7LE
> works with 4FUS 4.2.0 but I can't guarantee it.
> It would be better if you dumped a backup before trying it. If you can.

I have tried this on my 4FCN camera. It does not flash. Maybe because my firmware is newer? I cannot confirm this easily as I can not pair the camera with my phone...outside China. What about changing the file names to y203c?

roleoroleo, 2019-11-29:
I will send you a 4FCN version tomorrow.
Don't rename the files.

camonice, 2019-11-29:
> I will send you a 4FCN version tomorrow.
> Don't rename the files.

@roleoroleo Excellent. Thank you!

roleoroleo, 2019-11-30:
Try this for 4FCN:
https://drive.google.com/open?id=1YtGf1XkmTZbcl_hbOlM_xbwElNQdiG-h

roleoroleo, 2019-11-30:
If it works (both 4FUS and 4FCN) i will make a new release.

enrysan0, 2019-11-30:
Hi roleoroleo, thanks for the help. 
I don't know how to dump the firmware. If you can suggest me a guide or how to I'll try to dump it. Are there my personal information (wifi password, Yi account and password) in the dump?

roleoroleo, 2019-11-30:
See here:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10

For personal info it depends...
Remove /etc/back.bin before the dump if you want to be sure.

camonice, 2019-11-30:
> Try this for 4FCN:
> https://drive.google.com/open?id=1YtGf1XkmTZbcl_hbOlM_xbwElNQdiG-h

Hi roleoroleo, my 4FCN camera still does not flash. I think it might be due to the firmware version. It was bought recently.

roleoroleo, 2019-11-30:
What is the name of the file that u-boot looks for?

camonice, 2019-11-30:
> What is the name of the file that u-boot looks for?
> y25?
> y203c?
> ...

I don’t know how to check this. I searched and found that the other user used the name of y23...

roleoroleo, 2019-11-30:
y23 at the moment is not supported.
To support it I need a dump.

camonice, 2019-11-30:
> y23 at the moment is not supported.
> To support it I need a dump.

How can I make a dump for you without breaking the camera? 

Also I found myself and other people who use Yi-Hack-V4 need to reboot frequently. My outdoor camera often turns off or the proxychain stops working without a reboot:(

Your version seems to have no such issues!

roleoroleo, 2019-11-30:
My version is only for MStar platforn.
If you are using v4, your camera is not a MStar.

camonice, 2019-11-30:
> My version is only for MStar platforn.
> If you are using v4, your camera is not a MStar.

I have one outdoor 6CCN, a few Yi home 1080p - 6FCN, 6FUS, 4FCN. So I am a test bed for different firmwares:)

Rocket200, 2019-12-01:
> Probably this one:
> https://drive.google.com/open?id=1WwBl0n8LZpLjYdB6ZKpmh81_BBngR7LE
> works with 4FUS 4.2.0 but I can't guarantee it.
> It would be better if you dumped a backup before trying it. If you can.

I can not make a backup sorry.
I not try to flash it now ,I won't be sure it work .
but i check the Firmware on my 4FUS.
My 4FUS have Firmware 4.5.0.0C_201910080934 (I not make a Firmware update)

roleoroleo, 2019-12-01:
> I can not make a backup sorry.
> I not try to flash it now ,I won't be sure it work .
> but i check the Firmware on my 4FUS.
> My 4FUS have Firmware 4.5.0.0C_201910080934 (I not make a Firmware update)
> 
> maybe i can dump a log from it
> https://www.youtube.com/watch?v=fBjKTfWN4TI

You can dump a log and make a backup with the same method.
Your 4FUS should be a y203, based on firmware version.

Rocket200, 2019-12-01:
the Firmware from gdrive work with it?
or i need to wait for a update?
i can not dump (Not find USB-TTL)

roleoroleo, 2019-12-01:
The y203c works.

Rocket200, 2019-12-01:
I not have a backup from my Cam
i open the cam and i not find PINS for USB-TTL
[image]
what Firmware I can try to Flash? (on the picture u see y203c)

roleoroleo, 2019-12-01:
All the files with y203c suffix should work.
Try 0.2.2.

Rocket200, 2019-12-01:
Thank you !  it works
I try the 0.2.2 from gdrive here and it works
can I check or try something with the 4FUS - FW 4.5 - yi-hack 0.2.2 ?

I have a second 4FUS I flash the same 0.2.2 or one from the 0.2 releases ?

roleoroleo, 2019-12-01:
Use 0.2.2.

Rocket200, 2019-12-01:
can I use a new 0.2.2 for my BFUS Cams? 
I like the new settings in 0.2.2 (bevor Reboot now Maintenance)

roleoroleo, 2019-12-01:
Yes you can use it.
Wait for my official release.

camonice, 2019-12-01:
> Yes you can use it.
> Wait for my official release.

I open the back of the camera. Can @roleoroleo or anyone please tell if the chip is the right one? Thanks! 

I didn’t check the front of the board though..

[image]

roleoroleo, 2019-12-01:
I think you can use 0 2 2 but I never tried it on your model.
Wait for official release.

borodiliz, 2019-12-05:
Tested v0.2.2 on a 4FUS Firmware 4.5.0.0C_201910080934. It works like a charm!

roleoroleo, 2019-12-05:
home or dome?

borodiliz, 2019-12-05:
> home or dome?
Product Name: Yi 1080p Home Camera
Model: YYS.2016
QC Pass date: 2019.11
Recently purchased at amazon.es

roleoroleo, 2019-12-05:
Did you use y203c files?

borodiliz, 2019-12-05:
> Did you use y203c files?

Yes, y203c files extracted from 0.2.2 release

simox83971, 2019-12-06:
I tryed on  Yi Home Camera 3 (4FUS), but it does not work...
Is there something wrong or not supported yet?

roleoroleo, 2019-12-06:
Which firmware version do you have ?

enrysan0, 2019-12-06:
Same problem here. I tried to use "y203c_0.2.2" version but didn't work: the camera simply booted as normal.
My firmware is 4.2.0.0H_201909041620.

(@roleoroleo regarding the dump of the firmware i didn't understand the information in the link so i gave up)

roleoroleo, 2019-12-06:
I think you have to use y25_0.2.2.
Regarding the dump I could help you if you want.

enrysan0, 2019-12-06:
> Probably this one:
> https://drive.google.com/open?id=1WwBl0n8LZpLjYdB6ZKpmh81_BBngR7LE
> works with 4FUS 4.2.0 but I can't guarantee it.
> It would be better if you dumped a backup before trying it. If you can.

I tried this but not lucky.. The Cam is cycling trhought solid led yellow to blinking blue led to solid yellow.
Is there any procedure to follow before I totally brick the cam? Thanks

enrysan0, 2019-12-06:
This is the log i found in the SD Card:

```
[
][1/1/0:6:44:934]: dispatch.c(main-5243) hw_type(2)
[
][1/1/0:6:47:490]: rmm.c(msg_proc-4523) pid[193][
][1/1/0:6:47:578]: dispatch.c(main-5303) open_ptz fail[
][1/1/0:6:47:712]: dispatch.c(get_config-1223) got sn(XXXXXXXXXXXXXXXXXXXXXX)[
][1/1/0:6:47:713]: dispatch.c(get_config-1224) got pwd(XXXXXXXXXXXXXXXXXXXXXX)[
][1/1/0:6:47:713]: dispatch.c(get_config-1225) got ssid(XXXXXXXXXXXXXXXXXXXXXX)[
][1/1/0:6:47:713]: dispatch.c(get_config-1226) got tnp_init_string(MMFBJPLDIEEPKPHOOIFEPNEHHDCJFMGOHJENKLIHIJBKDABIPMIIAONIPBCLBOKOPIKPDKPFNJDPAIDKBE)[
][1/1/0:6:47:748]: dispatch.c(choose_server-650) in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, log_server = http://log.us.xiaoyi.com, sname = familymonitor-y25, dlproto = mius
[
][1/1/0:6:47:749]: dispatch.c(choose_server-650) in choose_server, region_id = 16, api_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sname = familymonitor-y25, dlproto = mieu
[
][1/1/0:6:47:758]: dispatch.c(main-5385) init ok, cost time(2822 ms)[
][1/1/0:6:48:528]: rmm.c(init_font-972) font:/home/base/iso8859-1-24x48.bin
[
][1/1/0:6:48:590]: rmm.c(init_font-984) filesize=36864
[
][1/1/0:6:48:590]: rmm.c(init_font-1004) init font /home/base/iso8859-1-24x48.bin ptr size(4)[
][1/1/0:6:48:591]: rmm.c(init_font-972) font:/home/base/iso8859-1-8x16.bin
[
][1/1/0:6:48:591]: rmm.c(init_font-984) filesize=4096
[
][1/1/0:6:48:591]: rmm.c(init_font-1004) init font /home/base/iso8859-1-8x16.bin ptr size(4)[
][1/1/0:6:48:799]: dispatch.c(do_monitor_wifi-1668) wifi disconnected, now reconnect wifi
[
][1/1/0:6:49:609]: rmm.c(rmm_init_isp-1480) pSensorID =8757 S C 2 2
[
][1/1/0:6:51:117]: rmm.c(send_start_recorde_msg-3957) msg snd success[
][1/1/0:6:51:118]: rmm.c(motion_proc-5155) send start recorde detect 
[
][1/1/0:6:51:517]: rmm.c(send_start_recorde_msg-3957) msg snd success[
][1/1/0:6:51:517]: rmm.c(motion_proc-5155) send start recorde detect 
[
][1/1/0:6:51:951]: cloud.c(sys_init-5517) open share mem ok
[
][1/1/0:6:52:204]: oss.c(main-2957) oss stream = ./oss_lapse, venc chn = 0x2000[
][1/1/0:6:52:317]: oss.c(main-2976) fshare_open ok[
][1/1/0:6:52:317]: oss.c(main-2957) oss stream = ./oss_fast, venc chn = 0x1000[
][1/1/0:6:52:473]: oss.c(main-2976) fshare_open ok[
][1/1/0:6:52:474]: oss.c(main-2957) oss stream = ./oss, venc chn = 0x400[
][1/1/0:6:52:649]: oss.c(main-2976) fshare_open ok[
][1/1/0:6:52:703]: p2p_tnp.c(main-6653) fshare_open ok[
][1/1/0:6:52:806]: p2p_tnp.c(p2p_set_tnp_init_status-2112) p2p_set_tnp_init_status 1 send_msg ok!
[
][1/1/0:6:52:911]: watch_process.c(get_watch_info-116) check_interval=10
[
][1/1/0:6:53:104]: watch_process.c(get_watch_info-134) process=dispatch;cmd=cd /home/app;./dispatch &;
[
][1/1/0:6:53:104]: watch_process.c(get_watch_info-134) process=cloud;cmd=cd /home/app;./cloud &;
[
][1/1/0:6:53:104]: watch_process.c(get_watch_info-134) process=rmm;cmd=reboot;
[
][1/1/0:6:53:104]: watch_process.c(get_watch_info-134) process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[
][1/1/0:6:53:104]: watch_process.c(get_watch_info-134) process=mp4record;cmd=cd /home/app;./mp4record &;
[
][1/1/0:6:53:104]: watch_process.c(get_watch_info-134) process=arp_test;cmd=cd /home/app;./arp_test &;
[
][1/1/0:6:53:105]: watch_process.c(get_watch_info-134) process=oss;cmd=cd /home/app;./oss &;
[
][1/1/0:6:53:105]: watch_process.c(get_watch_info-134) process=oss_fast;cmd=cd /home/app;./oss_fast &;
[
][1/1/0:6:53:105]: watch_process.c(get_watch_info-134) process=oss_lapse;cmd=cd /home/app;./oss_lapse &;
[
][12/1/15:56:48:97]: dispatch.c(p_worker-3292) DISPATCH_SET_DEFAULT_TIME 1575215808[
][12/1/15:56:48:194]: mp4record.c(main-1198) init_finish(1), start_with_reset(0)[
][12/1/15:56:48:264]: mp4record.c(main-1200) mp4 mode start init_finish[
][12/1/15:56:48:265]: mp4record.c(main-1206) fshare_open ok[
][12/1/15:56:53:420]: rmm.c(send_start_recorde_msg-3957) msg snd success[
][12/1/15:56:53:531]: rmm.c(motion_proc-5155) send start recorde detect 
[
][12/1/15:56:53:774]: rmm.c(send_start_recorde_msg-3957) msg snd success[
][12/1/15:56:53:774]: rmm.c(motion_proc-5155) send start recorde detect 
[
][12/1/15:56:54:400]: dispatch.c(do_monitor_wifi-1610) wpa may crashed, now reset
[
][12/1/15:56:54:514]: dispatch.c(do_monitor_wifi-1615) no usb dev

```

roleoroleo, 2019-12-06:
The log doesn't show nothing strange.

We have to restore the original fw.
If you have another cam of the same model we can dump it.
Or we could find another user.

enrysan0, 2019-12-06:
so is this normal?

```
[12/1/15:56:54:400]: dispatch.c(do_monitor_wifi-1610) wpa may crashed, now reset
[
][12/1/15:56:54:514]: dispatch.c(do_monitor_wifi-1615) no usb dev

```

Unfortunally I own only one YI Home v3 4FUS. I hope some other user could help me..

roleoroleo, 2019-12-06:
Try to re-pair the app.

enrysan0, 2019-12-07:
I tried a reset using the small pin-hole behind the camera. Put the camera side to the router, followed the normal procedure with scanning of the qr code. At the end the camera always say "waiting for connect" as it could see/connect to the wifi. 

The log i found in the SDCard card contain a lot of crashes Cloud/p2p/wpa:

```
[
][1/1/0:0:9:685]: dispatch.c(main-5243) hw_type(2)
[
][1/1/0:0:12:236]: rmm.c(msg_proc-4523) pid[193][
][1/1/0:0:12:331]: dispatch.c(main-5303) open_ptz fail[
][1/1/0:0:12:518]: dispatch.c(get_config-1223) got sn(XXXXXX_REMOVED_XXXXXXXX)[
][1/1/0:0:12:518]: dispatch.c(get_config-1224) got pwd()[
][1/1/0:0:12:518]: dispatch.c(get_config-1225) got ssid()[
][1/1/0:0:12:518]: dispatch.c(get_config-1226) got tnp_init_string()[
][1/1/0:0:12:518]: dispatch.c(get_config-1234) reset because no wifi config[
][1/1/0:0:12:518]: dispatch.c(send_saveconfig_msg-426) send_saveconfig_msg msg snd success[
][1/1/0:0:12:518]: dispatch.c(get_config-1281) get_config save conf
[
][1/1/0:0:12:518]: dispatch.c(choose_server-650) in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, log_server = http://log.us.xiaoyi.com, sname = familymonitor-y25, dlproto = mius
[
][1/1/0:0:12:518]: dispatch.c(choose_server-650) in choose_server, region_id = 16, api_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sname = familymonitor-y25, dlproto = mieu
[
][1/1/0:0:12:528]: dispatch.c(main-5385) init ok, cost time(2840 ms)[
][1/1/0:0:12:528]: dispatch.c(save_config-1024) got pwd()()[
][1/1/0:0:12:529]: dispatch.c(save_config-1025) got ssid()()[
][1/1/0:0:12:529]: dispatch.c(save_config-1026) config save finish 1[
][1/1/0:0:12:829]: dispatch.c(p_worker-3204) save_config[
][1/1/0:0:13:274]: rmm.c(init_font-972) font:/home/base/iso8859-1-24x48.bin
[
][1/1/0:0:13:360]: rmm.c(init_font-984) filesize=36864
[
][1/1/0:0:13:360]: rmm.c(init_font-1004) init font /home/base/iso8859-1-24x48.bin ptr size(4)[
][1/1/0:0:13:360]: rmm.c(init_font-972) font:/home/base/iso8859-1-8x16.bin
[
][1/1/0:0:13:360]: rmm.c(init_font-984) filesize=4096
[
][1/1/0:0:13:360]: rmm.c(init_font-1004) init font /home/base/iso8859-1-8x16.bin ptr size(4)[
][1/19/6:41:36:97]: dispatch.c(p_worker-3292) DISPATCH_SET_DEFAULT_TIME 1453185696[
][1/19/6:41:36:790]: rmm.c(rmm_init_isp-1480) pSensorID =8757 S C 2 2
[
][1/19/6:41:38:773]: cloud.c(sys_init-5517) open share mem ok
[
][1/19/6:41:39:3]: oss.c(main-2957) oss stream = ./oss_fast, venc chn = 0x1000[
][1/19/6:41:39:138]: oss.c(main-2976) fshare_open ok[
][1/19/6:41:39:138]: oss.c(main-2957) oss stream = ./oss, venc chn = 0x400[
][1/19/6:41:39:255]: oss.c(main-2957) oss stream = ./oss_lapse, venc chn = 0x2000[
][1/19/6:41:39:371]: oss.c(main-2976) fshare_open ok[
][1/19/6:41:39:371]: oss.c(main-2976) fshare_open ok[
][1/19/6:41:39:456]: p2p_tnp.c(main-6653) fshare_open ok[
][1/19/6:41:39:540]: p2p_tnp.c(p2p_set_tnp_init_status-2112) p2p_set_tnp_init_status 1 send_msg ok!
[
][1/19/6:41:39:818]: watch_process.c(get_watch_info-116) check_interval=10
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=dispatch;cmd=cd /home/app;./dispatch &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=cloud;cmd=cd /home/app;./cloud &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=rmm;cmd=reboot;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=mp4record;cmd=cd /home/app;./mp4record &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=arp_test;cmd=cd /home/app;./arp_test &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=oss;cmd=cd /home/app;./oss &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=oss_fast;cmd=cd /home/app;./oss_fast &;
[
][1/19/6:41:39:903]: watch_process.c(get_watch_info-134) process=oss_lapse;cmd=cd /home/app;./oss_lapse &;
[... 17 lines cut ...]
[
][1/19/6:41:52:432]: rmm.c(trans_info-513) trans_json result=EU52eipjsY34NAev
[
][1/19/6:41:52:432]: rmm.c(zbar_proc-4430) rmm got /tmp/got_wpa ssid(XXXXXX_REMOVED_XXXXXXXX) pwd(XXXXXX_REMOVED_XXXXXXXX) bind(XXXXXX_REMOVED_XXXXXXXX)
[
][1/19/6:41:52:432]: dispatch.c(do_mq_process-5058) invalid msg 0x1005
[
][1/19/6:41:52:509]: rmm.c(send_scanok_msg-646) msg snd success[
][1/19/6:41:52:562]: dispatch.c(choose_server-650) in choose_server, region_id = 16, api_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sname = familymonitor-y25, dlproto = mieu
[
][1/19/6:41:52:627]: rmm.c(send_wifi_conf_msg-676) msg snd success[
][1/19/6:41:52:713]: dispatch.c(do_mq_process-3413) got pwd(XXXXXX_REMOVED_XXXXXXXX)[
][1/19/6:41:52:844]: dispatch.c(send_connectting_msg-138) msg snd success[
][1/19/6:41:52:845]: dispatch.c(do_mq_process-5058) invalid msg 0x1004
[
][1/19/6:41:52:845]: dispatch.c(do_monitor_wifi-1668) wifi disconnected, now reconnect wifi
[
][1/19/6:41:54:694]: rmm.c(msg_proc-4648) got RMM_SPEAK_SCAN_OK[
][1/19/6:41:56:656]: rmm.c(msg_proc-4583) got RMM_SPEAK_CONNECTTING[
][1/19/6:41:59:839]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:41:59:892]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:42:1:852]: dispatch.c(do_monitor_wifi-1610) wpa may crashed, now reset
[
][1/19/6:42:1:944]: dispatch.c(do_monitor_wifi-1615) no usb dev
```

enrysan0, 2019-12-07:
I also don't understand why if my camera is a 4FUS (y203c) it only updates if the file is named y25 (that is for 9FUS)!!
So I tried also to use "y203c_0.2.2.tgz" renaming it to y25. The camera flashed well, but during the process when it try to connect to wifi it crashes and resets itself.

borodiliz, 2019-12-07:
@enrysan0  Probably not related but regarding the resets, I had problems with two YI Outdoor Camera. The problem was with the included power supply (1.0A). I experienced random reboots when trying to pair to wifi, using the speaker, or using IR LED's. Problem was solved using a different power supply (1.5A)

roleoroleo, 2019-12-07:
If your camera accepts y25 use y25_0.2.2.tgz.

enrysan0, 2019-12-07:
> If your camera accepts y25 use y25_0.2.2.tgz.

Ok, Thanks. However it seems I have a y25 4FUS:
[image]

enrysan0, 2019-12-07:
> @enrysan0 Probably not related but regarding the resets, I had problems with two YI Outdoor Camera. The problem was with the included power supply (1.0A). I experienced random reboots when trying to [...]

So now:
1. I'm using a 5v 3A power supply
2. I downloaded "y25_0.2.2.tgz"
3. put the firmware in the FAT32 formatted SDCard
4. flashed the firmare in the camera (long yellow led during the flash)
5. Resetted the camera using the pin-hole
6. Followed the steps on the app with the QR code.
After I let see the QRCode to the camera the yellow led turn to blinking blue and the the camera reset completely (i hear the noise of the speaker like when you power off the camera). After that the camera tells again "waiting for connect" waiting for the QRCode. If i let see the QRCode the camera again reset itself.

enrysan0, 2019-12-09:
Hi, I contacted YI support, they gave me a firmware image called "home_y25m". But after I put the file in the FAT32 formatted SD Card and power on the camera it stay all the time with the yellow led like is flashing. But after half an hour it still has the yellow led. Is there any way to force the firmware restore?

roleoroleo, 2019-12-09:
You need also the rootfs partition.
Ask technical support for the "sys_y25m" file as well.

camonice, 2019-12-09:
> You need also the rootfs partition.
> Ask technical support for the "sys_y25m" file as well.

I could not flash firmware using your released y25 and y203c files. May I change your released file names to y25m and have a go?

roleoroleo, 2019-12-10:
Could you share home_y25m you received from the support?
I think it's a format that you cannot use to restore a cam. It's only an upgrade.

enrysan0, 2019-12-10:
> Could you share home_y25m you received from the support?
> I think it's a format that you cannot use to restore a cam. It's only an upgrade.

Yes sure. Where I can send you the file? (as I don't know what kind of copyright follow the YI firmware and I prefer to not post it on github)

[97 more replies]

closed 2020-04-19 by roleoroleo