roleoroleo/yi-hack-MStar #25: BFUS Yi Home Camera

dnorman12345, 2019-11-08:
What needs done to a BFUS Yi Home 1080p to dump the contents to see if your firmware will work? It looks like the BFUS Dome camera works, but I am looking for the Home 1080p camera with the same letters.

dnorman12345, 2019-11-08:
I just bit the bullet and applied it. It seems to work fine with the other Yi Home firmware. This was probably not the best way to test it, but it seems like it all turned out fine.

electricJP, 2019-11-08:
I have one too. I can try to help getting the stock firmware. I have a FTDI if that helps.

dnorman12345, 2019-11-08:
> I have one too. I can try to help getting the stock firmware. I have a FTDI if that helps.

I used the version for the 6FUS and it seems to work fine.

roleoroleo, 2019-11-08:
See this issue:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10

The 1st info I need is the fw version.
Then a log of the boot procedure.
Probably the hack is the same as the 6FUS version.

electricJP, 2019-11-08:
> > I have one too. I can try to help getting the stock firmware. I have a FTDI if that helps.
> 
> I used the version for the 6FUS and it seems to work fine.

My point is to have a working copy of the stock firmware to be able to unbrick if needed.

electricJP, 2019-11-08:
I'm not sure where to find the uart connections:
[https://1drv.ms/u/s!Ar6TWe1DRNrxoc8QI3NyLNF-xql62w?e=YmWdLm](url)

Are they on the other side?

GuitarSkater, 2019-11-08:
I skimmed through the files for the dome and home. For the most part they are the same with the exception of the PTZ settings on the dome. 

I loaded the BFUS Dome firmware on the dome and home and haven't seen any issues. I could only imagine that if you loaded a Home version FW on the dome, you would have faults.

chesterfield82, 2019-11-15:
I took the 9FUS Version to flash the BFUS 1080p home, works pretty well

roleoroleo, 2019-11-15:
Could you please tell me what firmware version you had?

chesterfield82, 2019-11-15:
Version was 4.5.0, i believe version 1 (i think of had a february Date), after First start it asked to Upgrade to 4.5.0 Version 3 i think. I declined and flashed your FW for 9FUS. So ist can be considered compatible. Features are working AS expected (as far i could Test), still have to Test rtsp reliability.

closed 2020-02-22 by roleoroleo