roleoroleo/yi-hack-MStar #30: unable to set timezone

lvdfrnt, 2019-11-18:
anyone know how to set timezone with this hack? the command below works on hack v4 but unable to get it to "stick" on my 6FUS cameras. I telnet, run command, send reboot command and it does not reboot. Even if i reboot through WEBUI, timezone does not set.

sorokin-e, 2019-12-27:
Confirm. This way to set TZ doesn't work - https://github.com/TheCrypt0/yi-hack-v4/wiki/Setting-the-Timezone

roleoroleo, 2019-12-27:
The uclibc compiled by Yi doesn't support timezone.
And I think it isn't a good idea to rebuild all uclibc only for timezones.
But I could verify another way.

roleoroleo, 2019-12-27:
But what's the purpose to set TZ?
Which is the process you want to speak local time?

sorokin-e, 2019-12-29:
It's should be useful but not for the processes. Correct TZ helps to fix directory naming in Motion events because currently, it's UTC only. I think TZ also affects using with alternative systems like HA, but I'm not sure )
Thanks in advance.

roleoroleo, 2020-01-09:
Try 0.2.5.

sorokin-e, 2020-01-12:
It works. Thanks a lot.

closed 2020-01-17 by roleoroleo