roleoroleo/yi-hack-MStar #33: Support for 6FCN cameras

camonice, 2019-11-25:
Hello,

I manage to flash this firmware on a Chinese version 6FCN camera, but with a few issues. The main one is the geo-lock which was overcome by Yi-Hack-V4, but such feature is not carried on since 6FUS cameras has no geo-lock. Is there any plan to include proxy-chain function in future versions for 6FCN cameras?

Many thanks,
Camonice

roleoroleo, 2019-11-26:
I think it's possible to port the feature from yi-hack-v4 sources.
Are you able to make a PR?

camonice, 2019-11-26:
> I think it's possible to port the feature from yi-hack-v4 sources.
> Are you able to make a PR?

Hi roleoroleo,

I am not sure if I have made a PR successfully. What I did was to find the files from Yi-hack-v4 sources which refer to proxy-chain and copy them to the camera, then change related directory. But it doesn't work yet. The only way I can bypass the geo-lock is by turning off the Yi cloud.

Other issues related to 6FCN cameras are that the wifi setting is not saved when restart, and I don't seem to be able to save videos in SD card nor detect motion yet. I don't seem to be able to control status light etc, but I can log into the camera from a browser and capture snapshots.

Thanks,
Camonice.

roleoroleo, 2019-11-26:
Do you remember the version of the original firmware?

camonice, 2019-11-26:
> Do you remember the version of the original firmware?
> Is it home or dome?

I bought it recently from China, and it was manufactured in 2019. I never manage to pair it with my phone due to geo-lock so I don't know how to check the firmware version. Maybe I can check its package? If you teach me how to do it without the Yi app, I do have a second 6FCN camera with store firmware. It is a home 1080p camera.

Many thanks,
Camonice

roleoroleo, 2019-11-26:
Without the app, the only way is using a serial adapter connected to the uart pads.
It would be useful if you could dump the bootlog and file systems as described here:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10

roleoroleo, 2019-11-26:
Please try this for proxychain:
https://drive.google.com/open?id=14Ikeb2THH2hXFBTpzCuZG_zCtY6tdRVv

camonice, 2019-11-26:
> Without the app, the only way is using a serial adapter connected to the uart pads.
> It would be useful if you could dump the bootlog and file systems as described here:
> #10

Hi roleoroleo,

I can confirm that the new firmware you sent works for 6FCN cameras! I can pair it with the Yi app and save the wifi info now. I look forward to its release.

Thank you so much.
Camonice

roleoroleo, 2019-11-27:
Let me know if proxychains-ng works.

camonice, 2019-11-27:
> Let me know if proxychains-ng works.

Yes, it does work perfectly. 6FCN cameras can be added to the list.

Many thanks!

roleoroleo, 2019-11-27:
Did you try to change serial number overwriting mtd4?
Maybe you could change 4FCN to 4FUS.
What happen?
I know this is not correct, but just out of curiosity...

closed 2020-01-05 by roleoroleo

linsong837, 2020-02-16:
@camonice Were you able to pair camera with the Yi app? I have the same model - 6FCN. However, I can't pair it.

ShikyC, 2020-02-27:
Hi, can anybody here give me a hint how to get it working? I have a brand new 6FCN model out of box and I can't get it connected while I'm here in the U.S.
I copied the home_y203c and sys_203c file to the root folder and connected the camera but nothing happened, the camera just booted as usual and repeated saying "waiting to connect..."

linsong837, 2020-02-28:
Here is what I did.
Have a formatted sd card ready
Copy those two files into sd card.
Plug sd card into camera.
Power on camera
Open Yi home app, select to add a new camera.
Follow the instruction to generate qr code for wifi.
Scan qr code to connect WiFi.
Pairing camera with app.
Now, it's failed at pairing step.

linsong837, 2020-04-11:
@camonice @ShikyC 
Were any of you able to pair 6FCN with Yi Home app?
I never paired successfully.

ShikyC, 2020-04-11:
@linsong837 Neither do I. We can't pair it before we get the firmware flashed. I'm still looking for a workaround.