roleoroleo/yi-hack-MStar #34: [Support Request] 9FUS 4.2.0

rado0x54, 2019-11-26:
Hey @roleoroleo,

thanks already for the great project and the support of the MStar platform. As mentioned in the "4FUS" topic I own a Yi Home 1080p Camera v3 "9FUS", bought in the US. Attached you'll find the screen output during boot.

**Camera**
[image]
**Bootlog:**
[screenlog_boot.txt](https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/files/3889845/screenlog_boot.txt)

The installed version is 4.2.0.0F_201905241332 (but want's to upgrade to 4.2.0.0H_201909041620).

I'm a litte confused on how to take the next steps, because my shell does not look like `MStar #`, and I don't have the commands listed in the related [thread](https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10#issuecomment-546623449).

Any Pointers on how to proceed or is my model too different? How similar is it to the [4FUS issue](https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/23)?

Thanks!

rado0x54, 2019-11-26:
Ok, just out of curiosity, I downloaded your release `yi_home_1080p_9FUS_450_0.2.0` and renamed `sys_y203c` and `home_y203c` to `sys_y25` and `home_y25` respectively and flashed the camera successfully with it. So far everything seems to be working. I don't have an obvious shell via UART any longer. YI App now reports firmware as 4.5.0.0C_201910080934, but still wants to update to 4.2.0.0H :). Thanks so far. I still have another camera if it would benefit to extract any flash memory.

roleoroleo, 2019-11-26:
I think you have the same version as https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/23
The platform is always Mstar but I don't know what's the difference.

I am very surprised that the hack works by changing only the file name. I think you were very lucky!

If you want to reach the shell you have to stop the boot before the kernel loads.
Usually pressing ENTER.
It's the u-boot shell and not the linux shell.

rado0x54, 2019-11-26:
Lucky me, ok the <Enter> did the trick. I'm trying to flash "update" to 4.2.0.0H and then extract the memory contents.

roleoroleo, 2019-11-26:
I think it's better if you don't start the upgrade from 4.5.0.0C to 4.2.0.0H.
Online updates are partial and the risk is a mixed 4.2 / 4.5 file system.

rado0x54, 2019-11-26:
Here's the printenv. 
```
MStar # printenv
baudrate=115200
bootcmd=sf probe 0;sf read 0x21000000 0x50000 0x00210000;bootm 0x21000000
bootdelay=0
filesize=0
stderr=serial
stdin=serial
stdout=serial

Environment size: 169/4092 bytes
```
and the sf probe.
```
MStar # sf probe                            
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
```

and sf read
```
MStar # sf read 0x21000000 0x1d0000 0x1e0000
SF: 1966080 bytes @ 0x1d0000 Read: OK
```

roleoroleo, 2019-11-26:
Use this python script:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10#issuecomment-546711510
And dump rootfs and home partitions:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10#issuecomment-546735993

rado0x54, 2019-11-26:
Trying to run your script from the other thread but it exists uncleanly after a while with.

I run:
`sf read 0x21000000 0x1d0000 0x1e0000`
and
`./dump_mem.py --timeout 1 --step 64 /dev/tty.usbserial-A50285BI 115200 0x21000000 0x211e0000`

```
0x2100B6C0  9F F0 1D 30 05 3F A3 47 D7 CC 35 1E DA 18 73 DE |0...G.?..5...s..|
Traceback (most recent call last):
  File "./dump_mem.py", line 263, in <module>
    main()
  File "./dump_mem.py", line 248, in main
    raise e
  File "./dump_mem.py", line 246, in main
    data = dump(serial, opts, log)
  File "./dump_mem.py", line 143, in dump
    line = line.decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfd in position 59: ordinal not in range(128)
```

roleoroleo, 2019-11-26:
I verified the partition list and is the same.
I don't know why this error.

rado0x54, 2019-11-26:
I think it's because I'm running in a MacOS Terminal and it is some strange encoding problem. Let me see if I can figure things out.

rado0x54, 2019-11-27:
Ok, I think it more had to do with either my bad soldering skills or a faulty FTDI232. Seems to be working fine now. You'll find images and logs [on my Google Drive](https://drive.google.com/open?id=1oeXffBvMybrm8spz3_GtERb63A1kaVwN).

Status:
- 2019-11-27T095935_0x21000000_0x211e0000.* uploaded
- 2019-11-27T112601_0x21000000_0x21c30000.* WIP

roleoroleo, 2019-11-27:
Ok, downloaded. The rootfs is ok and it is almost equal to 4.5.0.
Remember that if you want to use the file 2019-11-27T095935_0x21000000_0x211e0000.img to restore the original rootfs, you need to remove the lasts 16 bytes and reverse the endianness.
If you want I can send you back the correct version.
I know that is a boring task, but could you dump home partition also?

rado0x54, 2019-11-28:
Hey @roleoroleo,

Sorry took a while, but I've added the home partition now. Yeah, it would be very nice if you could send me the correct original images back!

roleoroleo, 2019-11-28:
Here you will find your backup:
https://drive.google.com/open?id=1w9UH47PxQhNImHavW5pj9UqE0iA7KfNF
Let me know when you donwloaded it.

roleoroleo, 2019-11-28:
I will work on your cam in the next days.

roleoroleo, 2019-11-29:
Try this:
https://drive.google.com/open?id=1WwBl0n8LZpLjYdB6ZKpmh81_BBngR7LE

rado0x54, 2019-12-01:
Works!
```

                   _     _           _   
               _ _|_|___| |_ ___ ___| |_ 
              | | | |___|   | .'|  _| '_|
              |_  |_|   |_|_|__,|___|_,_|
              |___|
 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.2.2
 -----------------------------------------------------
```

closed 2019-12-01 by rado0x54

roleoroleo, 2019-12-03:
Did you download your original images?

rado0x54, 2019-12-04:
No, did you provide them somewhere for me?

roleoroleo, 2019-12-05:
https://drive.google.com/open?id=1w9UH47PxQhNImHavW5pj9UqE0iA7KfNF
Let me know when you downloaded it.

rado0x54, 2019-12-06:
Done. so I rename these files to home_y25 and sys_y25 respectively and should be able to flash the original firmware back?

roleoroleo, 2019-12-06:
Yes.