roleoroleo/yi-hack-MStar #35: How to restore original firmware and make backup

Vannixxo, 2019-11-26:
Hello, from what I understand there is a difference from Yi HACK V4 because that only enables the firmware load from the sd but this instead is flashing it directly into the cam, is this right?

How can I do a full backup so I can restore the original firmware and remove any trace if needed? Thanks a lot!

roleoroleo, 2019-11-26:
I didn't try yi-hack-v4 but I think it also overwrite the original firmware.
Here yiu can find a method:
https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/issues/10

barneyc, 2019-12-28:
I must admit to being pretty disappointed with the Yi 1080p Home camera.  Doesn't stream at all well over my IoT network - at least not compared to my other cameras.  Although your excellent hack did open it right up for Shinobi, so thanks for that.

Sorry but that thread leaves me none the wiser either.  Your hack included home & sys but I can't see anything _obvious_ to a complete noob such as myself as to how to get a stock home & sys back on the camera.   And without the camera will always end up with the hacked firmware.

Can you shed _more_ light on a solution to get one back to stock?  Cheers

roleoroleo, 2019-12-28:
Sorry but I have the same problems as TheCrypt0.
The stock firmware is not available for download. It's created with a complicated procedure directly from the cam.
I don't know if I can share these firmwares because is Yi material that was given to me in a confidential way by some users.
It's not a technical issue.

NielsFeenstra, 2020-01-05:
Is it possible to buy a new Yi 1080P home (6FUS) camera, download the original firmware from this new device and unbrick my other (6FUS) device?

roleoroleo, 2020-01-05:
Yes, it's possible. See the wiki.
But it's not so simple to buy exactly the same model.
There are many models, identical externally but different internally.

NielsFeenstra, 2020-01-05:
> Yes, it's possible, only if the same model. See the wiki.
> But it's not so simple to buy exactly the same model.
> There are many models, identical externally but different internally.

Thank you for the quick reply.

I’m having the same camera but it’s already flashed with the hack. Can I backup the firmware from thuis device?

roleoroleo, 2020-01-05:
Yes, you can backup the firmware from your hacked device but I think is not useful.
There is no difference between applying a firmware from a backup and one from an image.
If it's the same firmware version.

NielsFeenstra, 2020-01-05:
> Yes, you can backup the firmware from your hacked device but I think is not useful.
> There is no difference between applying a firmware from a backup and one from an image.
> If it's the same firmware version.

I tried to stop the boot sequence on the working device but with no luck. I've tried almost everything. 

So I can't stop the boot at the U-Boot prompt (should be "MStar #") pressing "Enter".

p.s: I'm on a macos

roleoroleo, 2020-01-05:
Are you sure that your TX line is correctly connected?

NielsFeenstra, 2020-01-05:
> Are you sure that your TX line is correctly connected?

Yeah pretty sure. I'm sure the serial adapter RX i working because i've flashed a couple of esp's with it.
I resolder the RX connection and tried it on another YI

roleoroleo, 2020-01-05:
But is connected to the correct pad on your cam?

NielsFeenstra, 2020-01-05:
> But is connected to the correct pad on your cam?

Its the pad from the grey cable. Or is it behind the black and yellow cable.

roleoroleo, 2020-01-05:
It depends on the cam model.

NielsFeenstra, 2020-01-05:
I have the yi Home 1080p 6FUS.

Used the post from Benji

https://github.com/TheCrypt0/yi-hack-v4/issues/49

roleoroleo, 2020-05-24:
Too much time since the last post.
Open a new issue if necessary.

closed 2020-05-24 by roleoroleo