roleoroleo/yi-hack-MStar #36: ffmpeg can't process the RTSP stream

Vannixxo, 2019-11-27:
Hello, I installed the latest version on my 6FUS and everything seems fine. I wanted to use this camera through homebridge and expose the RTSP stream to Apple Home Kit.
The ffmpeg plugin is extracting stills successfully from the dedicated http url, but the video stream doesn't work, is this a know issue?

roleoroleo, 2019-11-27:
Does ffmpeg work properly with the stream?
I'm using h264 stream daily with hass and it works correctly.

Vannixxo, 2019-11-27:
Unfortunately not, the RTSP works flawlessly through VLC, and the homebridge plugin (Camera-ffmpeg) says:

`[11/27/2019, 5:15:01 PM] [Camera-ffmpeg] Start streaming video from camera with 1280x720@299kBit`

I pass the RTSP stream in this way:
tcp -re -i rtsp://user:psw@192.168..........:554/ch0_0.h264

the log is:

`ffmpeg tcp -re -i rtsp://user:psw@192.168..........:554/ch0_0.h264 -map 0:0 -vcodec libx264 -pix_fmt yuv420p -r 30 -f rawvideo -tune zerolatency -b:v 299k -bufsize 299k -maxrate 299k -payload_type 99 -ssrc 5794419 -f rtp -srtp_out_suite AES_CM_128_HMAC_SHA1_80 -srtp_out_params J0rzUWbvTfF5ml6WLngZFqU2+8i6+0NksAlniZmm srtp://192.168.1.101:61337?rtcpport=61337&localrtcpport=61337&pkt_size=1316 -loglevel debug
ffmpeg version 4.1.4 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8.3.0 (Alpine 8.3.0)
  configuration: --prefix=/usr --enable-avresample --enable-avfilter --enable-gnutls --enable-gpl --enable-libass --enable-libmp3lame --enable-libvorbis --enable-libvpx --enable-libxvid --enable-libx264 --enable-libx265 --enable-libtheora --enable-libv4l2 --enable-postproc --enable-pic --enable-pthreads --enable-shared --enable-libxcb --disable-stripping --disable-static --disable-librtmp --enable-vaapi --enable-vdpau --enable-libopus --disable-asm --disable-debug

  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Splitting the commandline.
Reading option 'tcp' ... matched as output url.
Reading option '-re' ... matched as option 're' (read input at native frame rate) with argument '1'.
Reading option '-i' ... matched as input url with argument 'rtsp://user:psw@192.168..........:554/ch0_0.h264'.
Reading option '-map' ... matched as option 'map' (set input stream mapping) with argument '0:0'.
Reading option '-vcodec' ... matched as option 'vcodec' (force video codec ('copy' to copy stream)) with argument 'libx264'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-r' ... matched as option 'r' (set frame rate (Hz value, fraction or abbreviation)) with argument '30'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option '-tune' ...
 matched as AVOption 'tune' with argument 'zerolatency'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '299k'.
Reading option '-bufsize' ...
 matched as AVOption 'bufsize' with argument '299k'.
Reading option '-maxrate' ...
 matched as AVOption 'maxrate' with argument '299k'.
Reading option '-payload_type' ...
 matched as AVOption 'payload_type' with argument '99'.
Reading option '-ssrc' ...
 matched as AVOption 'ssrc' with argument '5794419'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rtp'.
Reading option '-srtp_out_suite' ...
 matched as AVOption 'srtp_out_suite' with argument 'AES_CM_128_HMAC_SHA1_80'.
Reading option '-srtp_out_params' ...
 matched as AVOption 'srtp_out_params' with argument 'J0rzUWbvTfF5ml6WLngZFqU2+8i6+0NksAlniZmm'.
Reading option 'srtp://192.168.1.101:61337?rtcpport=61337&localrtcpport=61337&pkt_size=1316' ... matched as output url.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.

Parsing a group of options: input url rtsp://user:psw@192.168..........:554/ch0_0.h264.
Applying option re (read input at native frame rate) with argument 1.
Successfully parsed a group of options.
Opening an input file: rtsp://user:psw@192.168..........:554/ch0_0.h264.
[tcp @ 0xb69ec770] No default whitelist set
[tcp @ 0xb69ec770] Original list of addresses:
[tcp @ 0xb69ec770] Address 192.168…….. port 554
[tcp @ 0xb69ec770] Interleaved list of addresses:
[tcp @ 0xb69ec770] Address 192.168…….. port 554
[tcp @ 0xb69ec770] Starting connection attempt to 192.168……. port 554

[tcp @ 0xb69ec770] Successfully connected to 192.168……. port 554

[rtsp @ 0xb4edc5d0] SDP:
v=0
[... 344 lines cut ...]
    Last message repeated 1 times
[h264 @ 0xb4eba3c0] nal_unit_type: 6(SEI), nal_ref_idc: 0
[h264 @ 0xb4eba3c0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0xb4eba3c0] unknown SEI type 240

    Last message repeated 21 times
[rtsp @ 0xb4edc5d0] All info found
[rtsp @ 0xb4edc5d0] Setting avg frame rate based on r frame rate

Input #0, rtsp, from 'rtsp://user:psw@192.168..........:554/ch0_0.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : 
ch0_0.h264
  Duration: N/A, start: 0.040000, bitrate: N/A
    Stream #0:0, 66, 1/90000: Video: h264 (Main), 1 reference frame, yuvj420p(pc, progressive, left), 1920x1088, 0/1, 25 fps, 25 tbr, 90k tbn, 180k tbc

Successfully opened the file.
Parsing a group of options: output url tcp.
Successfully parsed a group of options.
Opening an output file: tcp.
[NULL @ 0xb451f030] Unable to find a suitable output format for 'tcp'
tcp: Invalid argument

[11/27/2019, 5:19:11 PM] [Camera-ffmpeg] ERROR: FFmpeg exited with code 1`

roleoroleo, 2019-11-27:
I think that the problem is tcp option.
Maybe you meant "-rtsp_transport tcp"?

Vannixxo, 2019-11-27:
I removed that, so you think it is needed? But anyway it wasn’t working.. I’ll post the log after I removed “tcp” that was a residue of the option you mentioned

roleoroleo, 2019-11-27:
If the wifi connection is not good, it's better to use  "-rtsp_transport tcp" option.

Vannixxo, 2019-11-27:
Okay, I removed it and left the config like this (the standard as readme said):
` "source": "-re -i rtsp://user:psw@192.168.1........:554/ch0_0.h264",
`

Now it doesn't give any exit erroneous exit code, but the stream is very bad and works for 1-2 secs then it freezes on a still image (in HomeKit) and the logs even after quitting the viewer goes on...and on... with [libx264 @ ..............
I included only the last of these messages plus the changing part of the log at the end
What's going on?

This is the log:

[libx264 @ 0xb4df8500] frame=1223 QP=16.07 NAL=2 Slice:P Poc:446 I:3    P:154  SKIP:8003 size=867 bytes

[rtp @ 0xb4523030] Sending NAL 1 of len 195 M=0
[rtp @ 0xb4523030] Sending NAL 1 of len 207 M=0
[rtp @ 0xb4523030] Sending NAL 1 of len 331 M=0
[rtp @ 0xb4523030] Sending NAL 1 of len 121 M=1

[libx264 @ 0xb4df8500] frame=1224 QP=21.97 NAL=2 Slice:P Poc:448 I:314  P:3522 SKIP:4324 size=15908 bytes

[rtp @ 0xb4523030] Sending NAL 1 of len 2905 M=0
[rtp @ 0xb4523030] NAL size 2905 > 1290

[rtp @ 0xb4523030] Sending NAL 1 of len 5003 M=0
[rtp @ 0xb4523030] NAL size 5003 > 1290

[rtp @ 0xb4523030] Sending NAL 1 of len 4731 M=0
[rtp @ 0xb4523030] NAL size 4731 > 1290

[rtp @ 0xb4523030] Sending NAL 1 of len 3256 M=1
[rtp @ 0xb4523030] NAL size 3256 > 1290

frame= 1225 fps=6.0 q=25.0 Lsize=    1140kB time=00:00:42.43 bitrate= 220.1kbits/s dup=1175 drop=0 speed=0.208x    
video:1126kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 1.247702%
Input file #0 (rtsp://user:psw@192.168.1.......:554/ch0_0.h264):
  Input stream #0:0 (video): 116 packets read (772670 bytes); 51 frames decoded
; 
  Total: 116 packets (772670 bytes) demuxed
Output file #0 (srtp://192.168.1.101:64070?rtcpport=64070&localrtcpport=64070&pkt_size=1316):
  Output stream #0:0 (video): 1225 frames encoded; 1225 packets muxed (1153160 bytes); 
  Total: 1225 packets (1153160 bytes) muxed

51 frames successfully decoded, 0 decoding errors

[AVIOContext @ 0xb48a1bc0] Statistics: 0 seeks, 1560 writeouts

[libx264 @ 0xb4df8500] frame I:5     Avg QP:35.08  size: 25381
[libx264 @ 0xb4df8500] frame P:1220  Avg QP:22.90  size:   841

[libx264 @ 0xb4df8500] mb I  I16..4: 33.6% 57.2%  9.2%
[libx264 @ 0xb4df8500] mb P  I16..4:  0.1%  0.1%  0.0%  P16..4:  1.7%  0.2%  0.2%  0.0%  0.0%    skip:97.7%
[libx264 @ 0xb4df8500] 8x8 transform intra:55.6% inter:61.4%
[libx264 @ 0xb4df8500] coded y,uvDC,uvAC intra: 18.4% 28.7% 1.0% inter: 0.4% 0.9% 0.0%
[libx264 @ 0xb4df8500] i16 v,h,dc,p: 31% 35% 13% 21%
[libx264 @ 0xb4df8500] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 27% 16% 50%  1%  1%  1%  1%  1%  1%
[libx264 @ 0xb4df8500] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 35% 33% 15%  3%  2%  3%  4%  2%  4%

[libx264 @ 0xb4df8500] i8c dc,h,v,p: 74% 14% 11%  1%
[libx264 @ 0xb4df8500] Weighted P-Frames: Y:0.0% UV:0.0%
[libx264 @ 0xb4df8500] ref P L0: 90.2%  5.3%  3.3%  1.2%
[libx264 @ 0xb4df8500] kb/s:225.93

Exiting normally, received signal 15.

roleoroleo, 2019-11-27:
Mmmmmh...
Please, try with  "-rtsp_transport tcp" option.

Vannixxo, 2019-11-27:
unfortunately always the same thing... 2 secs of low quality stream, then it freezes on a still image that gradually gets definition (from blocky to perfectly displayed).
I've tried to switch to low res rtsp but nothing changed

camonice, 2019-12-01:
I cannot RTSP stream and got "VLC is unable to open the MRL 'rtsp://CAM_IP:554/ch0_0.h264'. Check the log for details." Any advice please?  Can I disable Yi Could and APP and use RTSP streaming? Thanks

camonice, 2019-12-01:
Manage to stream using ch0_1 channel, guess it is because audio is not supported. Can I stream without Ti cloud/app being enabled?

roleoroleo, 2019-12-01:
RTSP does not depend on the Cloud setting.
ch0_0 and ch0_1 are both without audio.

Vannixxo, 2019-12-01:
> Manage to stream using ch0_1 channel, guess it is because audio is not supported. Can I stream without Ti cloud/app being enabled?

Yes you can, I disabled cloud and both RTSP streams work.

Anyway I didn’t succeed in fixing that ffmpeg issue, I tried every possible settings, even using another homebridge plugin with ffmpeg OMX, someone is guessing it could be a problem with the 555 RTSP implementation (a version had some problems) that is not complying with ffmpeg RTSP “guidelines”

roleoroleo, 2019-12-01:
I tried to compile different version of 555 but withou solving the problem.
I think it's a bug that occurs when there's an unstable network.
I don't know how to solve it, I didn't find any other c lib to stream RTSP.
ffmpeg is too large and uses too much memory and cpu; is not suitable for an embedded platform.

Vannixxo, 2019-12-01:
> I tried to compile different version of 555 but withou solving the problem.
> I think it's a bug that occurs when there's an unstable network.
> I don't know how to solve it, I didn't find any other c [...]

Thanks so much for testing and the efforts! So you experienced the same problem when connecting the RTSP stream to a ffmpeg decoding instance? Can you confirm isn’t my issue? Anyway, my network is really solid and the camera is placed directly over the router... does Yi hack v4 use the same live555 lib? There is a dedicated plugin homebridge-ffmpeg-ui that is based on the same one I use and it used to work for Yi hack v4 users, I tried it but no luck.. same problem

camonice, 2019-12-01:
> RTSP does not depend on the Cloud setting.
> ch0_0 and ch0_1 are both without audio.

ch0_0 seems to be for high resolution, and ch0_1 is for low resolution.

roleoroleo, 2019-12-01:
I have the same issue with ffmpeg.
I don't know if v4 uses the same lib. It's closed source.

nathankellenicki, 2020-01-09:
I have this issue too unfortunately. My setup is a BFUS Yi Home 1080p, updated to 0.2.5 right now. Running homebridge-camera-ffmpeg. The stream lasts a few seconds then fails entirely. Sometimes after a few minutes/hours Homekit refuses to fetch the feed at all.

Vannixxo, 2020-01-09:
> I have this issue too unfortunately. My setup is a BFUS Yi Home 1080p, updated to 0.2.5 right now. Running homebridge-camera-ffmpeg. The stream lasts a few seconds then fails entirely. Sometimes after a few minutes/hours Homekit refuses to fetch the feed at all.
> 
> Does this provide any other kind of stream, such as hls?

I also posted the issue in the ffmpeg hombridge git, and one user last day mentioned he found a "solution", I tested it.. and it works flawlessly! But anyway still ffmpeg can't process the live555 stream this hack is generating as it isn't compliant!

You need to set the video codec-> vcodec to "copy", so homebridge-ffmpeg passes the h264 to HomeKit directly without further processing, here is an example of my config:

`{
                    "name": "YI",
                    "manufacturer": "YI",
                    "serialNumber": "",
                    "model": "YI 1080P Home Camera",
                    "videoConfig": {
                        "source": "-re -i rtsp://192.168.1.3:554/ch0_0.h264",
                        "stillImageSource": "-i http://192.168.1.3:8080/cgi-bin/snapshot.sh",
                        "maxStreams": 3,
                        "maxWidth": 1280,
                        "maxHeight": 720,
                        "maxFPS": 30,
                        "maxBitrate": 300,
                        "vcodec": "copy",
                        "packetSize": 1316,
                        "audio": false,
                        "debug": false
                    }`

nathankellenicki, 2020-01-09:
Interesting. Unfortunately this is my Homebridge output when I try that:

`homebridge         | Filtergraph 'scale=1280:720' was defined for video output stream 0:0 but codec copy was selected.`

Also, how does this work for iPads, which request a 1080p stream?

Vannixxo, 2020-01-09:
> Interesting. Unfortunately this is my Homebridge output when I try that:
> 
> `homebridge | Filtergraph 'scale=1280:720' was defined for video output stream 0:0 but codec copy was selected.`
> 
> Also, how does this work for iPads, which request a 1080p stream?

I think the log is normal.. as you are trying to scale the video but the codec is set to copy, I don’t have an iPad but on my mac it works flawlessly

roleoroleo, 2020-01-09:
Very interesting.
The copy option works correctly for me too.
Also using ffmpeg in command line:
```
fmpeg -rtsp_transport tcp -i rtsp://IP_CAM/ch0_0.h264 -vcodec copy temp.h264
```

bongomanuk, 2020-01-16:
This appears to work for me for livestreaming on lovelace ui on latest hassio. I was getting intermittent lockups before adding '-err_detect ignore_err; flag. Can anyone confirm?

ffmpeg:
camera:
  - platform: ffmpeg
    name: creepy3
    input: -rtsp_transport tcp -err_detect ignore_err -i rtsp://CAM_IP/ch0_0.h264

efkayx, 2020-01-22:
I tried your recommondation to test it but no success. Still getting the error with homebridge-ffmpeg latest version and as well ffmpeg directly from ffmpeg.org latest version and YI cam. Every other application like vlc, video surveillance software etc. is working inside out. 
Not working is the camera stream at all. snapshot is working and also seeable in homekit app.

Below homebridge config:

            "name": "Camera ffmpeg",
            "cameras": [
                {
                    "name": "YI",
                    "videoConfig": {
                        "source": "-rtsp_transport tcp -err_detect ignore_err -re -i rtsp://x:x@192.168.10.55:554/ch0_0.h264",
                        "stillImageSource": "-i http://x:x@192.168.10.55:8080/cgi-bin/snapshot.sh",
                        "maxStreams": 3,
                        "maxWidth": 1280,
                        "maxHeight": 720,
                        "maxFPS": 30,
                        "additionalCommandline": "-protocol_whitelist https,crypto,srtp,rtp,udp",
                        "maxBitrate": 300,
                        "vcodec": "copy",
                        "packetSize": 1316,
                        "audio": false,
                        "debug": true

and below log level with debug activated:

[1/22/2020, 7:20:15 PM] [Camera ffmpeg] Start streaming video from YI with 1280x720@299kBit
ffmpeg -rtsp_transport tcp -err_detect ignore_err -re -i rtsp://x:x@192.168.10.55:554/ch0_0.h264 -map 0:0 -vcodec copy -pix_fmt yuv420p -r 30 -f rawvideo -protocol_whitelist https,crypto,srtp,rtp,udp -b:v 299k -bufsize 299k -maxrate 299k -payload_type 99 -ssrc 5252488 -f rtp -srtp_out_suite AES_CM_128_HMAC_SHA1_80 -srtp_out_params 1lMdrOEuH4K0oJV4igPxVhkQVF6RPXQnKW8CxFH5 srtp://192.168.10.64:61725?rtcpport=61725&localrtcpport=61725&pkt_size=1316 -loglevel debug
ffmpeg version 4.2.1 Copyright (c) 2000-2019 the FFmpeg developers

  built with Apple clang version 11.0.0 (clang-1100.0.33.16)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.2.1-with-options_2 --enable-shared --cc=clang --host-cflags='-I/System/Library/Frameworks/JavaVM.framework/Versions/Current/Headers/ -fno-stack-check' --host-ldflags= --enable-gpl --enable-libaom --enable-libmp3lame --enable-libopus --enable-libsnappy --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --disable-libjack --disable-indev=jack --enable-opencl --enable-videotoolbox --disable-htmlpages --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libfdk-aac --enable-libgsm --enable-libmodplug --enable-libopenh264 --enable-librsvg --enable-librtmp --enable-librubberband --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtesseract --enable-libtwolame --enable-libvidstab --enable-libwavpack --enable-libwebp --enable-libzimg --enable-libzmq --enable-openssl --enable-nonfree --enable-version3 --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --disable-decoder=jpeg2000 --extra-cflags=-I/usr/local/Cellar/openjpeg/2.3.1/include/openjpeg-2.3
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Splitting the commandline.
Reading option '-rtsp_transport' ...
 matched as AVOption 'rtsp_transport' with argument 'tcp'.
Reading option '-err_detect' ...Routing option err_detect to both codec and muxer layer
 matched as AVOption 'err_detect' with argument 'ignore_err'.
Reading option '-re' ... matched as option 're' (read input at native frame rate) with argument '1'.
Reading option '-i' ... matched as input url with argument 'rtsp://x:x@192.168.10.55:554/ch0_0.h264'.
Reading option '-map' ... matched as option 'map' (set input stream mapping) with argument '0:0'.
Reading option '-vcodec' ... matched as option 'vcodec' (force video codec ('copy' to copy stream)) with argument 'copy'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-r' ... matched as option 'r' (set frame rate (Hz value, fraction or abbreviation)) with argument '30'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option '-protocol_whitelist' ... matched as AVOption 'protocol_whitelist' with argument 'https,crypto,srtp,rtp,udp'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '299k'.
Reading option '-bufsize' ...
 matched as AVOption 'bufsize' with argument '299k'.
Reading option '-maxrate' ...
 matched as AVOption 'maxrate' with argument '299k'.
Reading option '-payload_type' ...
 matched as AVOption 'payload_type' with argument '99'.
Reading option '-ssrc' ...
 matched as AVOption 'ssrc' with argument '5252488'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rtp'.
Reading option '-srtp_out_suite' ... matched as AVOption 'srtp_out_suite' with argument 'AES_CM_128_HMAC_SHA1_80'.
Reading option '-srtp_out_params' ...
 matched as AVOption 'srtp_out_params' with argument '1lMdrOEuH4K0oJV4igPxVhkQVF6RPXQnKW8CxFH5'.
Reading option 'srtp://192.168.10.64:61725?rtcpport=61725&localrtcpport=61725&pkt_size=1316' ... matched as output url.
[... cut ...]
[h264 @ 0x7fdaea817000] nal_unit_type: 8(PPS), nal_ref_idc: 3

[h264 @ 0x7fdaea817000] unknown SEI type 240

[h264 @ 0x7fdaea817000] nal_unit_type: 6(SEI), nal_ref_idc: 0
[h264 @ 0x7fdaea817000] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x7fdaea817000] 
unknown SEI type 240
[h264 @ 0x7fdaea817000] Format yuvj420p chosen by get_format().
[h264 @ 0x7fdaea817000] Reinit context to 1920x1088, pix_fmt: yuvj420p

[h264 @ 0x7fdaea817000] Frame num gap 34 32

[1/22/2020, 7:20:15 PM] [Camera ffmpeg] Stopped streaming

roleoroleo, 2020-01-22:
You cant'use vcopy with a 1280x720 res.
Set instead to 1920x1080 @ 20fps.

nathankellenicki, 2020-01-22:
Thanks for the updates. I tried this yesterday and now have it fully working. The feed now rarely breaks.

I'm using homebridge-camera-ui (which I believe is forked from homebridge-camera-ffmpeg), and these are my settings:

```
{
	"name": "Living Room Camera",
	"active": true,
	"videoConfig": {
		"source": "-rtsp_transport tcp -err_detect ignore_err -i rtsp://192.168.50.154/ch0_0.h264",
		"stillImageSource": "-i http://192.168.50.154:8080/cgi-bin/snapshot.sh?res=low&base64=no",
		"maxStreams": 3,
		"additionalCommandline": "-fflags +discardcorrupt",
		"maxWidth": 1280,
		"maxHeight": 720,
		"maxBitrate": 300,
		"maxFPS": 30,
		"vcodec": "copy",
		"packetSize": 1316,
		"audio": false
	}
}
```

efkayx, 2020-01-31:
Hi folks,

thanks very much for your comment based on the issue I'm facing. 

See below homebridge-camera-ffmpeg configuration:

platform": "Camera-ffmpeg",
            "videoProcessor": "/Users/x/Desktop/ffmpeg",
            "cameras": [
                {
                    "name": "YI",
                    "motion": true,
                    "uploader": true,
                    "videoConfig": {
                        "source": "-rtsp_transport tcp -err_detect ignore_err -re -i rtsp://x:x@192.168.10.55:554/ch0_1.h264",
                        "stillImageSource": "-i http://x:x@192.168.10.55:8080/cgi-bin/snapshot.sh",
                        "maxStreams": 3,
                        "additionalCommandline": "-fflags +discardcorrupt",
                        "maxWidth": 1920,
                        "maxHeight": 1080,
                        "maxFPS": 30,
                        "maxBitrate": 1024,
                        "vcodec": "copy",
                        "packetSize": 1316,
                        "audio": false,
                        "debug": true
                    }
                },
Basically the fork of homebridge-ffmpeg -> homebridge-camera-ui  is not working for me due to the latest version of nodejs what I'm using and my homebridge stopped working if I'm gonna restart. So tried again with homebridge-ffmpeg and the magic was to handle the stream via SD stream.
So once I disabled RTSP High Resolution within the webinterface of the YI-Hack and configured the rtsp stream to low resolution (see on top homebridge-ffmpeg = **ch0_1.h264**) and everything is working fine and very stable. Even though over LTE etc.
**RTSP High Resolution | Set low/high resolution stream. (It  takes effect only if the "RTSP" option is enabled).
-- | --**
Only downside it is not 1080p, which is first of all fine but I can't get the high resolution stream running as it stops working after 3 sec approx. and then I need to restart homebridge-ffmpeg again or kill the stream to get it again only working for 3sec. 
Do you have any idea based on the 1080 stream? I also took in consideration from roleoroleo to change the settings to 1920 * 1080 and 20fps and still not working. Only in low resolution stream and i think based on the experience the last couple of days it doesn't make a difference fps and resolution etc.

Another note as I was also playing with motion events etc and now everything is configured and working over home app, I'm wondering how can I get the motion triggering happening in the right way. I was thinking over MQTT and pass the message to a pseudo trigger in home app, which triggers then a movement event via IOS. 
MEaning like somebody is in front of the cam and an movement signal goes straight to the home app to indicate somebody is in front of the cam and create a screenshot immeditaly over the IOS notification screen and using a pseudo movement sensor.
I hope it is clear.

Again thanks for your help guys to get it at least running with SD and very stable :-)

stickpin, 2020-01-31:
Hi All,

I can confirm what @efkayx saying. I am also running low-resolution streams from few cameras for the last few days without any issues.

It seems that the 1080p stream itself not good, I don't know what is the difference between low and high res streams. @roleoroleo any chance you can take a look?

Thanks in advance!

Vannixxo, 2020-01-31:
> Hi folks,
>
> thanks very much for your comment based on the issue I'm facing.
>
> See below homebridge-camera-ffmpeg configuration:
>
> platform": "Camera-ffmpeg",
> "videoProcessor": [...]

What did you manage to get related to motion events in “home” app? (Home kit?) I would like to know

roleoroleo, 2020-02-01:
> @roleoroleo any chance you can take a look?

Yes, I'll take a look but I have already studied the stream a lot without luck.
I don't know if there is a solution.

stickpin, 2020-02-01:
Thanks a lot @roleoroleo!

Btw, for those who interested I am running following config:
`"platform": "Camera-ffmpeg",
            "videoProcessor": "ffmpeg",
            "cameras": [
                {
                    "name": "Camera 1",
                    "active": true,
                    "manufacturer": "YI",
                    "model": "Home Camera 1080p",
                    "serialNumber": "6FUSCAM1",
                    "firmwareRevision": "4.5.0.0C",
                    "videoConfig": {
                        "source": "-rtsp_transport tcp -re -i rtsp://192.168.22.21/ch0_1.h264",
                        "stillImageSource": "-i http://192.168.22.21:8080/cgi-bin/snapshot.sh",
                        "maxStreams": 3,
                        "maxWidth": 1920,
                        "maxHeight": 1080,
                        "maxBitrate": 2000,
                        "minBitrate": 1500,
                        "maxFPS": 30,
                        "vcodec": "copy",
                        "debug": false
                    }
                }`

In my opinion, the following params don't play any rule because **_"vcodec": "copy"_** is in use:
`                        "maxWidth": 1920,
                        "maxHeight": 1080,
                        "maxBitrate": 2000,
                        "minBitrate": 1500,
                        "maxFPS": 30,`

@Vannixxo regarding motion events, you need to represent it somehow, for example as "fake" motion sensors. If you want to do it, you will need to run some MQTT broker. I am running mosquitto together with homebridge and homebridge-mqttthing plugin and everything works very well.

efkayx, 2020-02-01:
Great discussion here folks and glad to see the help and responve from all of you.
First of all kudos @roleoroleo for the hard and excellent work on this YI Hack. This is so awesome.

@Vannixxo exactly this is what I want to realize via MQTT and triggering motion events with a fake motion sensor and the homebridge-mttting plugin. Would you be able to share your configuration with homebridge-mqttthing plugin and mosquitto. I do have mosquitto running on the same instance as the homebridge server and as well activated MQTT on the YI-Cam webinterface.
Thanks in advance @Vannixxo

roleoroleo, 2020-02-01:
I think the h264 stream is correct.
Beacuse if you save the stream and process it offline, there are no problems.
```
h264grabber > temp.h264
```
copy to a desktop system
```
ffmpeg -i temp.h264 temp2.h264
```

Something is wrong in the live555 library or the wireless network is not working properly (driver unstable, far router, too much traffic...).

bongomanuk, 2020-02-01:
720p seems to be better behaved with home assistant for me. I haven't looked too deeply but load on the camera in top seems to be on the high side with 1080p and there is a lot of jitter when pinging the camera. On a strong wifi I get 500ms+ pings. Is this a load issue?

As a side note, I managed to clock the soc to 1ghz from 800mhz but haven't tested to see if that helps the issue. The camera remains stable though. Also, it appears to use alsa for audio, that could be mux'd into the stream fairly easily I guess

roleoroleo, 2020-02-02:
With RTSP client running, my camera is stable with a ping < 10ms
And h264grabber + rRTSPServer is about 14% CPU.
The load is high but not so high to be a problem.

Interesting your work about the overclocking and audio.
Could you explain me some details.

bongomanuk, 2020-02-02:
Clocking is easy, just echo 1000000 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq. I am not sure about alsa being used, but there is a playback and capture device in the device tree if you look around in the FS.

Vannixxo, 2020-02-02:
> Clocking is easy, just echo [1000000](tel:1000000) > /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq. I am not sure about alsa being used, but there is a playback and capture device in the device tree if you look around in the FS.

Audio would be awesome..
@roleoroleo 
I have seen the new toggle to “disable” camera, would it be possible to use it through homebridge to power it off ?

roleoroleo, 2020-02-02:
I don't konw if the toggle enable/disable works as you espect.
There is an API to switch on/off the camera on the fly:
```
http://192.168.1.200:8080/cgi-bin/camera_settings.sh?switch_on=yes
http://192.168.1.200:8080/cgi-bin/camera_settings.sh?switch_on=no
```
and another API to save the setting permanently:
```
cgi-bin/set_configs.sh?conf=camera
```

Vannixxo, 2020-02-02:
> I don't konw if the toggle enable/disable works as you espect.
> There is an API to switch on/off the camera on the fly:
> 
> ```
> http://192.168.1.200:8080/cgi-bin/camera_settings.sh?switch_on=yes
> http://192.168.1.200:8080/cgi-bin/camera_settings.sh?switch_on=no
> ```
> 
> and another API to save the setting permanently:
> 
> ```
> http://192.168.1.200:8080/cgi-bin/set_configs.sh?conf=camera
> ```
> 
> I'm working on a bug fixing about these files.

So are this the same settings as the new option In the user panel? What’s the difference with saving the option permanently? I thought this would turn off or on all the functions of the camera but it is still physically on waiting and accepting the command of turning back on

bongomanuk, 2020-02-02:
If someone has a toolchain set up it would be interesting to see if its possible to capture audio. I am a bit busy to take a look but this example could be used to a quick test. https://gist.github.com/albanpeignier/104902

roleoroleo, 2020-02-03:
> So are this the same settings as the new option In the user panel? What’s the difference with saving the option permanently? I thought this would turn off or on all the functions of the camera but it is still physically on waiting and accepting the command of turning back on

Yes these API are called when you press "Save" on the web page.
If you use only the camera_settings API the camera switches on/off but when you remove power the camera returns to the previous setting.
"Switch on/off" is a fake: the camera is always on and connected to the network as you are saying.

roleoroleo, 2020-02-03:
Please, open new issue or reuse old issues for audio and enable/disable switch.

bongomanuk, 2020-02-04:
Apologies is this has already been mentioned/tried. I noticed the max buffer size for 1080p on rtspserver is set at 30000, I have seen recommendations to up that to 65000 to alight with UDP limits.

roleoroleo, 2020-02-05:
Could you share these recommendations?

efkayx, 2020-02-06:
Hi @stickpin 
regarding motion events, you need to represent it somehow, for example as "fake" motion sensors. If you want to do it, you will need to run some MQTT broker. I am running mosquitto together with homebridge and homebridge-mqttthing plugin and everything works very well.

...exactly this is what I want to realize via MQTT and triggering motion events with a fake motion sensor and the homebridge-mttting plugin. Would you be able to share your configuration with homebridge-mqttthing plugin and mosquitto. I do have mosquitto running on the same instance as the homebridge server and as well activated MQTT on the YI-Cam webinterface.
Thanks in advance @stickpin

Do you have any chance and time to help me out here?

nathankellenicki, 2020-02-06:
I've moved to homebridge-camera-ui instead of homebridge-camera-ffmpeg. It's a fork of it, but it adds an inbuilt motion sensor accessory to the camera so you don't need a fake sensor.

Here is the config for one of my cameras:

```json
	{
		"platform": "CameraUI",
		"videoProcessor": "ffmpeg",
		"cameras": [{
			"name": "Lobby Camera",
			"debug": true,
			"active": true,
			"videoConfig": {
				"debug": true,
				"source": "-rtsp_transport tcp -err_detect ignore_err -i rtsp://192.168.50.69/ch0_0.h264",
				"stillImageSource": "-i http://192.168.50.69:8080/cgi-bin/snapshot.sh?res=low&base64=no",
				"maxStreams": 3,
				"additionalCommandline": "-fflags +discardcorrupt",
				"maxWidth": 1280,
				"maxHeight": 720,
				"maxBitrate": 300,
				"maxFPS": 30,
				"vcodec": "copy",
				"packetSize": 1316,
				"audio": false
			},
			"mqtt": {
				"active": true,
				"host": "192.168.50.47",
				"port": 1883,
				"username": "",
				"password": "",
				"topicPrefix": "agema",
				"topicSuffix": "motion",
				"startMessage": "motion_start",
				"stopMessage": "motion_stop",
				"recordOnMovement": false,
				"recordVideoSize": 30,
				"interval": 120
			}
		}]
	}
```

As mentioned above, this setup has been working perfectly for me for weeks now. :) (On iPhone, iPad Pro, and macOS)

bongomanuk, 2020-02-09:
> Could you share these recommendations?

I saw it mentioned here, it might not be the problem but it could be worth seeing if increasing the buffer helps. http://lists.live555.com/pipermail/live-devel/2013-April/016816.html

roleoroleo, 2020-02-10:
The maxSize value in rRTSPServer.cpp now is 300000.
Probably is too high ;)

alvi76, 2020-02-16:
> > Clocking is easy, just echo 1000000 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq. I am not sure about alsa being used, but there is a playback and capture device in the device tree if you look around in the FS.
> 
> Audio would be awesome..

I agree. I am implementing a baby cam monitor under homeassistant but there is no way to filter any noise with ffmpeg_noise. Since audio is missing, ffmpeg_noise is always "Unavailable".

fabergoncalves, 2020-03-21:
I was testing the RTSP with ffmpeg to see what was the issue. 

On the Pi it hangs after 2 or 3 seconds with the 1080p stream, on Windows using homebridge + ffmpeg works fine with no issues. I got the Yi Home 1080p stream running on Apple Home app for more than 10 minutes without crashing.

It looks like we are depending on an update from the devs from ffmpeg to fix this problem in the Linux distribution.

pfeatherstone, 2020-11-08:
On a related note, i was getting similar behavior when building an RTSP server for the YI using libx265 and using ffmpeg as the RSTP client.

roleoroleo, 2020-12-19:
Sorry for the delayed response.
What rtsp library or server are you using? Or are you writing it from scratch?
I'm looking for an rtsp server (now I'm using live555).

closed 2024-05-11 by github-actions[bot]