roleoroleo/yi-hack-MStar #37: Possibility to Use Built In Speakers?

gfrench96, 2019-12-01:
Hello All, 

First off I would just like to say thank you to the developer(s) for making the firmware function on the new generation of the cameras, and all the work that went into making the hack work with the new release of the firmware. 

Now, I have a project in mind, and I'm not entirely sure how to accomplish it but I know what needs to be done. I have FreePBX setup with the PagingPro module, just as well, I have several of these cameras around my house, my plan is to use the built-in speakers in the camera as part of the paging system. The PagingPro Module allows for multicast transmission of voice data, I need to have a way for the cameras to listen on a specific IP address and port for RTP Multicast streams, and play the result over their internal speakers. Iv'e seen a few promising applications (VLC, OpenRTSP, etc) but im unsure how to install it at the shell of the camera when ssh'd in (yum/apt-get missing is an obvious obstacle, as well as internal space on the embedded memory of the camera, when trying to wget a VLC RPM i run out of space almost immediately). 

Any creative or constructive advice would be appreciated.

roleoroleo, 2019-12-01:
Before choosing the program to receive the stream we need to understand how to access the speaker and how to reproduce sounds.

gfrench96, 2019-12-01:
> Before choosing the program to receive the stream we need to understand how to access the speaker and how to reproduce sounds.

Hey RoleoRoleo, thanks for your reply, and thank you for the great release, I agree that we need to understand how they work, do you know of any applications I can use to probe around the system and play with the speakers? Maybe if I can learn how to interact with them or get them to produce sound, then we can go from there, of course I’m happy to report my findings back. Let me know. Thanks.

roleoroleo, 2019-12-01:
Some hint:
https://github.com/fritz-smh/yi-hack/issues/58

gfrench96, 2019-12-04:
> Some hint:
> [fritz-smh/yi-hack#58](https://github.com/fritz-smh/yi-hack/issues/58)

Thank you for pointing me in that direction, while the capability of being able to play pre recorded messages is interesting, I'm trying to find some way to spawn a RTSP listener on the device, and have it always listening for the intercom signal to pass live messages over the speaker.

roleoroleo, 2020-05-24:
The audio device is busy (used by rmm) and we cannot use it.

closed 2020-05-24 by roleoroleo