roleoroleo/yi-hack-MStar #38: IR Leds seems pulsing

Extreme79, 2019-12-01:
Hi.
I flashed today the firmware over a 1080p BFus dome and over two 1080p home Bfus.
All seems to be ok except 1 thing.
All the three cameras seems to have pulsing ir led: I noticed that in night environment: the image suffers of continuos (and regular) pulsing on light level.
On the dome I verified that, watching the camera with another camera, The ir leds go up and down on intensity in a regular way.
Unfortunately I've not tested in low loght the cameras before the hack flash (unpacked, updated to the latest official and hacked :) ).. so I'm not 100% sure that this is caused by the hack, but is possible.
[2019-12-02T00-02-40.zip](https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/files/3908837/2019-12-02T00-02-40.zip)

I attached a clip to try to show the issue. On the background it's me :P and on the top left you see the ir light source from Yi Dome. You can clearly see the ir light drop at sec. 5 to turn back normal after few seconds.
The same happen with the Home cameras.

@roleoroleo Can you check this?

Vannixxo, 2019-12-03:
I noticed the same exact thing on my 6FUS, don't know how it was with original firmware.. but my other US27 YI 70p home camera doesn't pulsate

Extreme79, 2019-12-03:
It's really annoying, but I suspect that issue is caused by the original code and not by the hack.. anyway a check on it could be useful.
I tried also to get in touch with Yi not talking (of course) about the hack.. but not reply since 2 days.

Extreme79, 2019-12-11:
@roleoroleo I got a brand new BFUS home camera. I'll test the device as stock to check the presence of the issue. I'll update the issue after testing the stock camera

Extreme79, 2019-12-12:
Brand new camera show the same issue, also with stock firmware.
ir light pulsing with 4.5fw, pulsing too with 4.6.

So this is the confirm that it's not an hack issue.

closed 2019-12-12 by Extreme79