roleoroleo/yi-hack-MStar #40: Where are the firmware?

ivityc, 2019-12-03:
I am looking for the firmware for Yi 1080p Home BFUS but I cannot find any link to it.
Thanks to let me know.

Extreme79, 2019-12-03:
I don't want to say something wrong.. but the BFUS is a Yi **Dome** model.. and **not the Home**, that I know is a 6FUS. wait also other replies, just to be sure.

Anyway, check the label on the camera/box.

ivityc, 2019-12-03:
hi, the camera I have report a BFUS and it is a new YI 1080P Home. No dome.
But this is not the problem, I really do not find the links to the firmware files from the main page.

Extreme79, 2019-12-04:
Probably you are searching for this. Check it. 
https://github.com/roleoroleo/yi-hack-I think nderstood yoih6FUS_4.5.0/releases

roleoroleo, 2019-12-04:
"Get the correct firmware files for your camera from the releases section."

ivityc, 2019-12-04:
Hi again,

I know which file I need, according to table below from this link https://github.com/roleoroleo/yi-hack-6FUS_4.5.0 I need to grab the 3rd one ( Yi 1080p Home BFUS | 4.5.0* | y203c )

Camera | Firmware | File prefix | Remarks
-- | -- | -- | --
Yi 1080p Home 6FUS | 4.5.0* | y203c | -
Yi 1080p Home 9FUS | 4.5.0* | y203c | -
Yi 1080p Home BFUS | 4.5.0* | y203c | -

but I do not find a hyperlink to it and if I download the full git .zip there are no firmware in there.
If I unzip I can find a folder named y203c but it is empty. 
So shall I compile it myself via the scripts in the zip or does exist a precompiled firmware ?
Thanks

mgiako, 2019-12-04:
You can find the files in releases: https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases

ivityc, 2019-12-04:
Ok Thanks, that's was the link I was looking for. I see it a standard link for git hub but I am new to it so was not immediately capturing my attention...
The issue can be closed .
Cheers

closed 2019-12-04 by ivityc