roleoroleo/yi-hack-MStar #41: Support for 9CUS

borodiliz, 2019-12-04:
Hi,

Thanks for your work with this hack!

I've just received two Yi Outdoor Camera 1080p 9CUS Firmware: 3.0.0.0D_201809111054
It's this model supported? If not , I will try to dump logs / rootfs / home

Thanks and regards,
Borja

roleoroleo, 2019-12-05:
I need a bootlog to know if it's a MStar and if I can support this model.

JHenzi, 2019-12-30:
Did you ever get the dump, didn't look like it. Just picked this up on sale, I can try v4 or see about providing them if you suggest this over v4.

roleoroleo, 2019-12-30:
The hardware is different.
If it loads v4, doesn't load mine.
I didn't receive any dump.

closed 2020-01-09 by roleoroleo

Dreamoffice, 2020-04-18:
Hi 
i also got a Yi Outdoor Camera 1080p 9CUS, is there a chance to get the hack running?

UlfgarIlFabbro, 2021-02-06:
La 

> 
> 
> Did you ever get the dump, didn't look like it. Just picked this up on sale, I can try v4 or see about providing them if you suggest this over v4.

The v4 version loads and boots quietly but many features are missing such as onvif and ftp upload. Can your version work?

roleoroleo, 2021-02-06:
If your cam works with v4 it can't work with MStar or Allwinner.