roleoroleo/yi-hack-MStar #44: ipc_cmd

minimalnetwork, 2019-12-06:
Congratulations and thank you for your work! I'm investigating about ipc_cmd binary. I want to disable IR lens filter (black and white filter), because even if you deactivate the LED, the camera activates the infrared lens, and iamage becomes black and white in dark scenes. I saw that sending ipc messages, you can deactivate status LED, IR led and modify other settings. There is an extended index of messages to modify more settings apart from those defined in ipc_cmd.h? Where did you get them from?

Thank you!

roleoroleo, 2019-12-06:
I wrote an ipc "sniffer", so when I send a message using the app, I can see the hex code.
The code is similar to ipc_cmd.

Extreme79, 2019-12-10:
[semi-ot] This could be a way to activate motion detection from - example - a domotic controller?

roleoroleo, 2019-12-10:
I don't understand.
You can send a "motion detection start" and a "motion detection stop".
But what's the purpose?

Extreme79, 2019-12-11:
Simple example: instead h24 active motion detection (with unwanted notifications) I can program my domotic controller to activate motion detection on every cam on burglar alarm arming and deactivate it on alarm disarm.
Or similar usage, of course.

roleoroleo, 2019-12-11:
I do the same using home assistant.
The cam sends the events but the automation starts only if the switch "alarm on" is activated.

Extreme79, 2019-12-11:
[OT] my plan is to use my domotic controller (actually zipabox, but I'm learning hass to migrate all) to activate motion detection on the cam, to let the family members, for the moment, to receive notifications through the app without the need of manually activate the feature manually by the app.
Think you that will be possibile to control PTZ through ipcmd (and then by url)?

roleoroleo, 2019-12-11:
I don't know if it's possible to disable motion detection events on the camera.
I will investigate it.

About PTZ we could work together if you want.
Because I don't have a dome camera.
If I send you the ipc sniffer you could verify if it's possible to read ptz commands.

minimalnetwork, 2019-12-11:
I have written a sniffer and I have captured PTZ messages. This night when I have access to my personal PC, I can send you these messages.

Extreme79, 2019-12-11:
I'm available too, if you want to test with different models (my dome is a BFUS). 
@roleoroleo If you send me the ipc sniffer I can test it this evening.

roleoroleo, 2019-12-11:
If you disable motion detection from app the ipc messages continue to pass through the system.
Eventually you need to disable the mqtt publish.
Here the ipc_sniff program:
https://drive.google.com/open?id=1FwNf45Vo9A93s6rjzwY4TtQvV7DZF3f3

@minimalnetwork 
Did you try to resend the captured messages? The cam moves?

minimalnetwork, 2019-12-12:
> If you disable motion detection from app the ipc messages continue to pass through the system.
> Eventually you need to disable the mqtt publish.
> Here the ipc_sniff program:
> https://drive.google.com/open?id=1FwNf45Vo9A93s6rjzwY4TtQvV7DZF3f3
> 
> @minimalnetwork
> Did you try to resend the captured messages? Does the cam move?

Hello!
YES, cam it moves. This is are the captured messages:
RIGHT
hex mode
010000000800000006400640180000000400000000000000
string mode
**\x01\x00\x00\x00\x08\x00\x00\x00\x06\x40\x06\x40\x18\x00\x00\x00\x04\x00\x00\x00\x00\x00\x00\x00**
---------------------
LEFT
hex mode
010000000800000006400640180000000300000000000000
string mode
**\x01\x00\x00\x00\x08\x00\x00\x00\x06\x40\x06\x40\x18\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00**
-------------------
DOWN
hex mode
010000000800000006400640180000000200000000000000
string mode
**\x01\x00\x00\x00\x08\x00\x00\x00\x06\x40\x06\x40\x18\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00**
--------------------
UP
hex mode
010000000800000006400640180000000100000000000000
sting mode
**\x01\x00\x00\x00\x08\x00\x00\x00\x06\x40\x06\x40\x18\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00**
--------------------

**#### NOTE: remember to include STOP message after to send UP, DOWN, RIGHT or LEFT message, because camera doesn't stop. Sleeping about 0.5 or 1s between messages is OK. ####**

STOP
hex mode
01000000080000000740010000000000
string mode
**\x01\x00\x00\x00\x08\x00\x00\x00\x07\x40\x01\x00\x00\x00\x00\x00**

There was a good idea to add these instructions to ipc_cmd and include a PTZ control to web interface or ONVIF.

Hope this helps!

roleoroleo, 2019-12-12:
Thank you.
I will work on these commands and I will send you an alpha hack to test, if you want.

roleoroleo, 2019-12-12:
What suffix is your model?
y25? h201c? ...

minimalnetwork, 2019-12-12:
Sorry, it is true... it is a 6FUS cam.

roleoroleo, 2019-12-12:
Sorry, I don't understand.
y203c is a home camera or I am wrong...

minimalnetwork, 2019-12-12:
Sorry, yes, you are right. My cam is a Dome and I was flashed a h201c suffix firmware.

roleoroleo, 2019-12-13:
Do you want to try?
https://drive.google.com/open?id=1A6EmJ2SeBpBGWLjkEKjf4VuOrh7UqcrV

minimalnetwork, 2019-12-14:
It works! Good work! Maybe sleep time after stop message, should be less. Now movements are too long. You must divide /3 sleep time. SVGs arrows are not shown in Safari. I don't know why. I attached a capture.
[image]

roleoroleo, 2019-12-14:
Good news! I will correct the sleep time.
For the svg images, I'm not an expert in web dev. Can anyone help me to improve maintenance and ptz pages?

minimalnetwork, 2019-12-17:
I'm very bad programming in C and webdev, but if you send me maintenance and ptz files, I can do something. I think that maintenance page is OK, what do you want to change in these page?
OFFTOPIC: I want to play a 720p directly from the camera but I see that streams are extracted from /proc/star/OMX/VMFE0 and VMFE1. I suppose that these streams are 1920x1080 and 640x360 repectively and they are extracted from memory directly, I'm wrong? Do you know if there is a way to grab or transmit a mid resolution RTSP?

roleoroleo, 2019-12-17:
> I'm very bad programming in C and webdev, but if you send me maintenance and ptz files, I can do something.
I updated the sources.

> I think that maintenance page is OK, what do you want to change in these page?
Someone reported that reboot doesn't work in chrome.

> OFFTOPIC: I want to play a 720p directly from the camera but I see that streams are extracted from /proc/star/OMX/VMFE0 and VMFE1. I suppose that these streams are 1920x1080 and 640x360 repectively [...]
You are right, There isn't a mid resolution stream.
I think you could reconfigure the stream using setprop command but I never tried.
See /home/app/init.sh.
```
 setprop mi.vi.src 2
        setprop mi.vi.img.sub 0
        setprop mi.sys.shrink_mem 1
        setprop mstar.omx.gop.disable 1
        setprop mi.osd.gop.use 0
        setprop mi.venc.bufcnt 1
        setprop mi.venc.sub.bufcnt 1
        setprop mi.vi.bufcnt 3
        setprop mi.vi.sub.bufcnt 3
        setprop mi.vi.sub.width 640
        setprop mi.vi.sub.height 360
        setprop mi.venc.stop.flush 1
        setprop mi.video.height.force.aligned32 1
        setprop omx.mainstream.min.resolution.width 1920
        setprop omx.mainstream.min.resolution.height 1088
        setprop mstar.omx.avqe.aecgain 2
        setprop mstar.omx.avqe.aecmode 15
#       setprop mstar.omx.avqe.dbg.enable 1
#       export AVQE_DBG_PATH=/tmp/sd/
        setprop mi.vi.yuvmon.pat /home/ms
```

roleoroleo, 2019-12-19:
Please, try the last version:
https://drive.google.com/open?id=1051syKu6lQxqZ5bdBKu5Qg4MxtD3IKIW
I added the onvif ptz support but I can't try it.

minimalnetwork, 2019-12-19:
Oops! When I enter in PTZ page, web says that "PTZ functions are not available for this camera model." Maybe in ptz.js there are an error. `if (key == "model_suffix" && data[key] == "h201c")` seems to be wrong. If I have a while, I will try to compile httpd to see that contains key and data[key] variables.

roleoroleo, 2019-12-20:
Which model suffix is reported in the Status page?
You could verify variable using browser debugger.

roleoroleo, 2019-12-20:
Found the problem.
Overwrite this js:
https://drive.google.com/open?id=1HfH_1GRG7YIe8GEZi81Rl5wcXKC93yHA

minimalnetwork, 2019-12-20:
Same problem. The issue is that 'suffix' variable is empty. Result of cgi-bin/status.json is:
`{
"hostname":"yi-hack-001",
"fw_version":"0.2.3",
"home_version":"4.6.0.0A_201908271549",
"model_suffix":"",
"serial_number":"6FUSXXXXXXXXXXX",
"local_time":"Sat Dec  7 19:36:15 UTC 2019",
"uptime":"72340.24",
"load_avg":"2.49 2.67 2.53",
"total_memory":"60856",
"free_memory":"14904",
"local_ip":"192.168.89.13",
"netmask":"255.255.255.0",
"gateway":"192.168.89.1",
"mac_addr":"50:13:95:xx:xx:xx",
"wlan_essid":"Wifixxxxx" 
}`
model_suffix variable is taken from `cat /home/yi-hack/model_suffix` and model_suffix file doesn't exist. I have searched inside file system and I have not found nothing to retrieve model suffix. Do you know where is located @roleoroleo ?

roleoroleo, 2019-12-20:
I will verify.
If you want to try manually touch a file in /home/yi-hack with content h201c

minimalnetwork, 2019-12-20:
Sorry, I was try to overwrite all.js to a previous version firmware (first 0.2.3 version). Now with a current 0.2.3 version, model_suffix exists, but the problem exists ("PTZ functions are not available for this camera model."). I have updated `updatePage` (ptz.js file) function and now It works:

```
function updatePage() {
        $.ajax({
            type: "GET",
            url: 'cgi-bin/status.json',
            dataType: "json",
            success: function(data) {
                for (let key in data) {
                    if (key == "model_suffix") {
			if (data[key] == "h201c") {
                            $('#ptz_title').hide();
                            $('#ptz_main').show();
			} else {
                        $('#ptz_title').show();
                        $('#ptz_main').hide();
                       }
		    }
                }
            },
            error: function(response) {
                console.log('error', response);
            }
        });
    }
    return {
        init: init
    };
```
Your function had got entire data array without stopping when finding the key = "model_suffix"

minimalnetwork, 2019-12-20:
I apologize, my browser and temporal files was confused me @roleoroleo. Your all.js.gz file was OK and It works. I have seen that your corrected function in all.js.gz file is practically identical to mine.

roleoroleo, 2019-12-20:
Let me know if onvif ptz works.

minimalnetwork, 2019-12-20:
It works in all directions! Good job!

roleoroleo, 2019-12-20:
Both relative and continuous movements?

minimalnetwork, 2019-12-20:
Yes. I'm not very introduced about ONVIF, and I used ONVIF Device Manager for windows. In continuous mode, cam was moving until button direction was released. In relative mode, cam moves about a 0.4 secs for every button pressing. PAN/TILT values compressed between 0/100 is not affecting the movements. I suppose that is a normal behavior in this camera. If there is another better program to test it, let me know please.

roleoroleo, 2019-12-21:
Ok. I think it's enough, thank you.
The important thing is that the continuous mode works unti button is pressed.

closed 2019-12-27 by roleoroleo