roleoroleo/yi-hack-MStar #45: 1080 dome - h201c - no work

bepece1, 2019-12-08:
I can't hack my Yi Cam (1080 dome [model 6FUSY1E0VHAC54190422]) with firmware (4.6.0.0A_201908271549)
I copied the both files (home_h201c and sys_h201c) in root sd card format FAT32 and reboot cam but yi-hack-MStar no work so as the original firmware continues in cam.

I've tried format sd card from cam and other products but the same result

What can I do???

Please help

bepece1, 2019-12-13:
> I can't hack my Yi Cam (1080 dome [model 6FUSY1E0VHAC54190422]) with firmware (4.6.0.0A_201908271549)
> I copied the both files (home_h201c and sys_h201c) in root sd card format FAT32 and reboot cam [...]

Is it possible this hack needs a third file called "rootfs_h201c" in adiction to "home_h201c" and "sys_h201c"???
Please help!!!! @roleoroleo @skylarhays

roleoroleo, 2019-12-13:
rootfs and sys are the same partition.
You don't need it.
Please try with another sd. It must work.

jasonchen31, 2020-01-01:
I have similar problem with my "Yi 1080p Dome". And I tried also download the images from yi-hack-v4. It works. I noticed the file size of the rootfs is quite different in between. And I checked the root partition size of "Yi 1080p Dome" it is only about 900KB (/dev/root):
/dev/root            959420208 854483622 104936585  89% /
tmpfs                11528033431   2998188 11525035243   0% /dev
/dev/mtdblock5       9738115106 6149283892 3588831214  63% /home
tmpfs                12280578656 1208269824 11072308832  10% /tmp
However the rootfs file of yi-hack-MStar is something like 1.7MB. I think that's the reason we cannot download the image. 

[Suggestion]
Would it be possible to move some large libraries from the rootfs to SD card (e.g. "<sd>/yi-hack-MStar" folder) and create a link to these external files? (Maybe this is how yi-hack-v4 solve this issue).

roleoroleo, 2020-01-02:
If you are using v4 version you can't use my version.
The hardware platform is different.

jasonchen31, 2020-01-02:
I see. Thanks for your efforts and reply!

closed 2020-02-22 by roleoroleo