roleoroleo/yi-hack-MStar #47: Onvif Media Profiles: only one seems available

Extreme79, 2019-12-11:
I added the camera to Onvifer APP.
When the app scan the xml onvif envelope, an error on media profile is shown and next only one media profile seems to be available (the hd one).
Is that expected or is an issue?
Could be a malformed xml, but this app can't export the onvifer request to check it.

roleoroleo, 2019-12-11:
Did you try with ONVIF Device Manager?

Extreme79, 2019-12-11:
I'll check also with odm this evening, and now I'll check the code you suggested.

anyway, I mean that I see only one media profile *exploring the onvif response* and not watching the camera. Obviously the video that I see follow the rtsp stream that I choose to see (but the app, following the onvif response, show only one media profile).

Extreme79, 2019-12-11:
Checked with odm. Odm too is able see only one profile, the hd one. That screen is related to the dome, but on the home is exactly the same.
[image]
this is a portion of the onvifer's log, about media profiles retrival:
`2019-12-11 22:20:48.330 2/20 
Sending command: GetProfiles Authentication time:Sun Dec 11 22:20:48 GMT+01:00 119 
http://192.168.1.155 
2019-12-11 22:20:48.330 1/20 
Entering SOAP call loop. 
2019-12-11 22:20:48.330 0/20 
Updating media profiles... 
2019-12-11 22:20:48.531 0/20 
Response: <?xml version="1.0" encoding="UTF-8"?> 
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl"><SOAP-ENV:Header><wsa5:Action SOAP-ENV:mustUnderstand="true">http://www.onvif.org/ver10/media/wsdl/GetProfiles</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><trt:GetProfilesResponse><trt:Profiles token="Profile_0"><tt:Name>Profile_0</tt:Name><tt:VideoSourceConfiguration token="Profile_0"><tt:Name></tt:Name><tt:UseCount>0</tt:UseCount><tt:SourceToken>Profile_0</tt:SourceToken><tt:Bounds x="0" y="0" width="1920" height="1080"></tt:Bounds></tt:VideoSourceConfiguration><tt:VideoEncoderConfiguration token="Profile_0"><tt:Name>Profile_0</tt:Name><tt:UseCount>0</tt:UseCount><tt:Encoding>H264</tt:Encoding><tt:Resolution><tt:Width>1920</tt:Width><tt:Height>1080</tt:Height></tt:Resolution><tt:Quality>0</tt:Quality><tt:RateControl><tt:FrameRateLimit>0</tt:FrameRateLimit><tt:EncodingInterval>0</tt:EncodingInterval><tt:BitrateLimit>0</tt:BitrateLimit></tt:RateControl><tt:Multicast><tt:Address><tt:Type>IPv4</tt:Type></tt:Address><tt:Port>0</tt:Port><tt:TTL>0</tt:TTL><tt:AutoStart>false</tt:AutoStart></tt:Multicast><tt:SessionTimeout>PT0S</tt:SessionTimeout></tt:VideoEncoderConfiguration></trt:Profiles></trt:GetProfilesResponse></SOAP-ENV:Body></SOAP-ENV:Envelope> 

2019-12-11 22:20:48.632 1/20 
Retrieval of media profiles succeeded. `

roleoroleo, 2019-12-11:
There is only one profile:
`
if [[ $(get_config RTSP) == "yes" ]] ; then
    if [[ $(get_config RTSP_HIGH) == "yes" ]] ; then
        h264grabber -r high | RRTSP_RES=0 RRTSP_PORT=$RTSP_PORT RRTSP_USER=$USERNAME RRTSP_PWD=$PASSWORD rRTSPServer &
        ONVIF_PROFILE_0="--name Profile_0 --width 1920 --height 1080 --url rtsp://$LOGIN_USERPWD%s$D_RTSP_PORT/ch0_0.h264 --snapurl http://$LOGIN_USERPWD%s$D_HTTPD_PORT/cgi-bin/snapshot.sh?res=high --type H264"
    else
        h264grabber -r low | RRTSP_RES=1 RRTSP_PORT=$RTSP_PORT RRTSP_USER=$USERNAME RRTSP_PWD=$PASSWORD rRTSPServer &
        ONVIF_PROFILE_1="--name Profile_1 --width 640 --height 360 --url rtsp://$LOGIN_USERPWD%s$D_RTSP_PORT/ch0_1.h264 --snapurl http://$LOGIN_USERPWD%s$D_HTTPD_PORT/cgi-bin/snapshot.sh?res=low --type H264"
    fi
fi
`
Profile_0 (1920x1080) if you select "RTSP High Resolution" or Profile_1 (640x360) if you don't select "RTSP High Resolution".

Extreme79, 2019-12-11:
I read this piace of code.. And I fallen in error reading this
"onvif://www.onvif.org/Profile/S **$ONVIF_PROFILE_0** **$ONVIF_PROFILE_1**"

I considered all the profiles variables to be valorized, so I was expecting two profiles. Watching better the above code, only one variable has a value, setted within the if.
it's not a problem for me, but someone with NVR could be found useful the double profile choice via onvif.

anyway, this clarified me all.

roleoroleo, 2019-12-12:
I think the cam doesn't have enough cpu to handle 2 stream.

Extreme79, 2019-12-12:
Probably this is true. I've not enough skills about onvif and video streams, so I don't know how this is generally handled by ip cams. In other ipcams I own (low cost china,anyway) the onvif expose two stream (low res and hi res) and user can choose what is needed (also different streams at the same time).. and I doubt that this cams come with powerful cpu :P .
Anyway not a real issue for me. If needed the lowres url can be called. 
The "issue" could exist only for people that need to configure NVR that admit exclusively onvif cams.

roleoroleo, 2019-12-12:
The problem is not the onvif daemon but the h264grabber program that extracts frames from memory and generates the rtsp stream.
The program was created after reverse engineering and is not optimized (this is not done directly by the manufacturer).
We could probably try running it twice, both for high resolution and low resolution, using 2 different tcp ports to publish it.
I don't know if the CPU can do it and I don't know if it's really necessary.

closed 2019-12-27 by roleoroleo