roleoroleo/yi-hack-MStar #48: ssh/tenet details

truxntrax, 2019-12-12:
Hi,

Really liking the firmware, but I cannot get the mqtt notion detection to work.  So, have enabled ssh, and was going to check via top, that mqtt is running.  However, what are the login details please for me to check this?

Many thanks

John

roleoroleo, 2019-12-12:
root with blank password

truxntrax, 2019-12-12:
Thanks, I can see mqttv4 is running.  However, I am "listening" using mqqt tools in home assistant, and I can't see the camera sending any messages to indicate motion_start/ motion_stop...
I have double check settings: mqtt server, user, password, port etc...  I have enabled and disabled mqtt, but can't "hear" any messages.  Is there anything else I need to set?

Many thanks for quick reply!

John

truxntrax, 2019-12-13:
Doh!  I had disabled motion detection in the yi app interface.  All seems to be working now. 

Many thanks

John

roleoroleo, 2019-12-13:
Interesting.
What flag did you disable in the app?

truxntrax, 2019-12-13:
There is a motion on off toggle switch option in the yi app. I had set this to off, as am using zoneminder for the motion detection.

roleoroleo, 2019-12-13:
Could you make a screenshot, please?

truxntrax, 2019-12-13:
Sure.
[image]

closed 2019-12-27 by roleoroleo