roleoroleo/yi-hack-MStar #50: Contributing to 9FUS

NickAnderegg, 2019-12-14:
Hi @roleoroleo,

I couldn't get in touch with you via email, so I thought I'd try this. I'd like to assist with developing yi-hack-MStar. I have a Yi 1080p Home v3, which is the 9FUS with the 4.2.0 firmware. I see that it is listed in the README as "Work in progress," so I want to avoid duplicating work. Do you have a fork or branch for this firmware?

[image]

roleoroleo, 2019-12-15:
See
https://github.com/roleoroleo/yi-hack-MStar/issues/34

roleoroleo, 2020-01-05:
Try the last version.

VikIborra, 2020-03-10:
I have 9FUS cameras but are v1, not v3. Maybe it's the problem.

kubawolanin, 2020-03-15:
Hi @roleoroleo 
First, thank you for your work on this. 
I use both yi-hack-v4 from @TheCrypt0 on Yi Outdoor and Yi Dome 720p and now flashed my *Yi Home 9FUS* with your hack.
I'm impressed how many useful features you've included there!
I really wish I could use PTZ on my TheCrypt0-s based Dome (`1.9.1.0J_201809191135`).

Just wanted to confirm that I successfuly flashed `y203c` into `9FUSY21CQSUEPO191010`. 
I tried `y25` with no luck. Simply copied both unpacked files into 32GB microsd card, didn't rename any files.
Maybe that'll help someone.
[image]

Thanks again!

roleoroleo, 2020-03-15:
Thank you for your report.
About your Dome, I think it's possible to add ptz to the cam.
Probably there are the same ipc commands.
But I can't do it without the hardware.

closed 2020-04-06 by roleoroleo