roleoroleo/yi-hack-MStar #51: Some new alarm features

oppinus, 2019-12-17:
Hi,
in the yi-hack-v4, there is some reference to theese future features: 
- Alarm functionality via Telegram (@frekel's PR #177 in yi-hack-v3)  or
- Auto upload of the recorded footage to the cloud.

Do you think to work in the future on some of theese features?
For my opinion, if the camera will be capable to send an alarm via email, telgram, etc (with jpeg or mp4) without YI cloud, all will be fantastic.

roleoroleo, 2019-12-17:
In general I think it is better to implement these functions externally (i.e. domotic system).
The camera sends alarm events via mqtt so this is already possible now.
I think I will not work on this but someone could do a PR.

oppinus, 2019-12-18:
You're right, but many people don't have a home domotic system.
For them the only alternative would be a nvr and use onvif on the cameras.
But if the camera had been autonomous, at least for alarms sent via email (with video and / or jpeg attached), it would be complete.
Mqtt is an alternative having a home domotic system (I'm using openhab), but then you have to download the video from the camera and I don't think there is even the name of the video files in the mqtt messages.
Anyway I'll do some experiments and in case I'll let you know.

roleoroleo, 2019-12-18:
For hass there is a specific "driver" to download files via ftp.
If you need it, we could modify the MQTT message and add other details.

oppinus, 2019-12-18:
The maximum would be to have in the MQTT message, besides the event (start / stop), the name of the folder and the name of the file, so as to be able to download the video on a home automation system via http / ftp.
If it was in json format, for example:
start:
{"Time": "2019-12-18T17: 00: 53", "Event": "MOTION_START", "Dir": "", "File": ""}
stop:
{"Time": "2019-12-18T17: 00: 53", "Event": "MOTION_STOP", "Dir": "2019Y12M18D07H", "File", "45M00S60.mp4"}
it would be perfect!

roleoroleo, 2019-12-22:
It's not easy.
When an event is detected the system sends an "MOTION ON" message and when the movement ends the "MOTION OFF" message is sent.
Several minutes can pass between events and therefore many mp4 files (one per minute) can be generated.
Normally, with a short event, there are two mp4 files.

roleoroleo, 2020-07-16:
Solved in 0.2.8.

closed 2020-07-16 by roleoroleo