roleoroleo/yi-hack-MStar #52: proxychain seems not working

x571, 2019-12-22:
/home/yi-hack/bin # proxychains4 telnet www.google.com
[proxychains] config file found: /home/yi-hack/etc/proxychains.conf
[proxychains] preloading /home/yi-hack/lib/libproxychains4.so
proxychains can't load process....: No such file or directory

x571, 2019-12-22:
forget about my msg. proxychains works but some how the camera could not be paired to app after wifi successfully connected. can someone help check the log? Thanks
[log.txt](https://github.com/roleoroleo/yi-hack-MStar/files/3992060/log.txt)

x571, 2019-12-22:
btw, my rtsp server runs well and after mapping port, I could access the camera anywhere (well without binding to app)

roleoroleo, 2019-12-22:
Is it cloud enabled?

x571, 2019-12-22:
> Is it cloud enabled?

could is enabled.

linsong837, 2020-02-14:
Any update on this issue? I seem gave the same problem.

roleoroleo, 2020-02-14:
Enable the cloud from yi-hack settings and try to reset the cam to factory default using the button.

linsong837, 2020-02-16:
Tried again with cloud enabled and factory default reset. Still didn't work. Any advise?

roleoroleo, 2020-02-17:
Check if p2p_tnp process is running and check if internet connection is working.

linsong837, 2020-02-19:
How to check what process is running?  The internet is working through. I can ping google in SSH.

roleoroleo, 2020-02-19:
'ps' command

linsong837, 2020-02-20:
The p2p_tnp process is running and Internet is working.
----------------------------------------------------
/home/yi-hack # ps
PID   USER     TIME  COMMAND
    1 root      0:02 /init
    2 root      0:00 [kthreadd]
    3 root      5:04 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root     17:16 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:06 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:03 [kworker/0:1]
   17 root      0:00 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:20 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   76 root      0:00 [mmcqd/0]
   87 root      0:00 [cryptodev_queue]
   98 root      0:00 [spi0]
  126 root      0:00 [kworker/0:2]
  130 root      0:00 [cfg80211]
  140 root      3:10 ./log_server
  141 root     17:04 ./dispatch
  150 root      1d08 ./rmm
  200 root      0:00 [kworker/u2:3]
  245 root      2:23 ./mp4record
  248 root      0:19 ./oss
  249 root      0:06 ./watch_process
  262 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  267 root      0:00 telnetd
  287 root      0:00 pure-ftpd (SERVER)
  309 root      0:00 dropbear -R
  341 root      0:00 ntpd -p pool.ntp.org
  345 root      0:00 h264grabber -r high
  346 root     56:08 rRTSPServer
  352 root      0:00 onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi H
  392 root      5:26 [RTW_CMD_THREAD]
  417 root      0:12 ./arp_test
 1196 root      0:00 dropbear -R
 1227 root      0:00 -sh
 1241 root      0:00 ps
 1528 root      2:54 /home/app/recbackup
 2168 root      3:40 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf
 2209 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script
 2310 root      0:01 ./cloud
 3458 root      0:00 ./p2p_tnp

linsong837, 2020-02-26:
Any comment on this?
Thanks.

linsong837, 2020-02-27:
What app do you guys use to bind the camera?
I tried Yi home and Yi IoT, none of them worked.

x571, 2020-02-27:
last time I tried a lot but no work for me. I would give my Yi camera for free if anyone feel interested. bay area pick it up.

roleoroleo, 2020-02-27:
Try to disable all feature except ssh and httpd:

- Enable cloud
- No RTSP
- No ONVIF
- No FTP
- No NTPD
- No MQTT

In this manner, the cam runs in a configuration equivalent to the original fw and must work.

linsong837, 2020-03-06:
Tried, still not working.
I saw some login failure in log, but not sure if it caused the failure of binding.

][2/26/3:1:7:181]: p2p_tnp.c(tnp_proc-6406) -------------- NetInfo: -------------------
[
][2/26/3:1:7:181]: p2p_tnp.c(tnp_proc-6407) Internet Reachable     : YES
[
][2/26/3:1:7:181]: p2p_tnp.c(tnp_proc-6408) P2P Server IP resolved : YES
[
][2/26/3:1:7:181]: p2p_tnp.c(tnp_proc-6409) P2P Server Hello Ack   : YES
[
][2/26/3:1:7:181]: p2p_tnp.c(tnp_proc-6410) Local NAT Type         :[
][2/26/3:1:7:182]: p2p_tnp.c(tnp_proc-6415)  Unknow
[
][2/26/3:1:7:182]: p2p_tnp.c(tnp_proc-6428) My Wan IP : x.x.x.x
[
][2/26/3:1:7:182]: p2p_tnp.c(tnp_proc-6429) My Lan IP : 192.168.x.x
[
][2/26/3:1:7:182]: p2p_tnp.c(tnp_proc-6431) InitStr(MJFBIBLBICELLEHOOKFGPPECGLCJFKHEGKEALBJNIJBJCOAGPIIIBDMDPBDEAELAOGLCCBPBMKCNAHDPAKBM)
[
][2/26/3:1:7:182]: p2p_tnp.c(tnp_proc-6432) did(TNPCHND-892728-WTXNX)
[
][2/26/3:1:7:182]: p2p_tnp.c(p2p_set_tnp_init_status-2112) p2p_set_tnp_init_status 4 send_msg ok!
[
][2/26/3:1:9:215]: cloud.c(yi_sync_time-5407) 
[
][2/26/3:1:12:282]: cloud.c(yi_sync_time-5407) 
[
][2/26/3:1:13:708]: p2p_tnp.c(tnp_proc-6563) PPPP_Listen_With_Key failed, ret = -3008
[
][2/26/3:1:15:352]: cloud.c(yi_sync_time-5407) 
[
][2/26/3:1:17:111]: p2p_tnp.c(state_statistics-6247) check_login fail 0
[
][2/26/3:1:18:444]: cloud.c(yi_sync_time-5407) 
[
][2/26/3:1:21:447]: cloud.c(yi_sync_time-5453) yi_sync_time fail!
[
][2/26/3:1:24:451]: cloud.c(yi_sync_time-5400) cmd = /home/app/cloudAPI -c 136 -url http://api.xiaoyi.com/v2/ipc/sync_time 
[
][2/26/3:1:24:514]: cloud.c(yi_sync_time-5407) 
[
][2/26/3:1:27:112]: p2p_tnp.c(state_statistics-6247) check_login fail 0
[
][2/26/3:1:32:581]: cloud.c(yi_sync_time-5407)

fernando155, 2020-05-06:
good afternoon.

I got exactly the same problem. The camera is working, but is not possible to connect with yi app. I tryed to resert everything to default, change the ip address, even using the phone network, and still not pairing with the camera. 

Anyone knows how to get the original software from another camera? I bought two and i will se if I can bring it back

roleoroleo, 2020-05-06:
Did you try to change app?

fernando155, 2020-05-07:
yes, i do. I tryed with me phone and with my computer a few times and always fails. 

As I said, is any way to get the original software?

roleoroleo, 2020-05-07:
What app are you using?

fernando155, 2020-05-07:
yi home

roleoroleo, 2020-05-07:
> Anyone knows how to get the original software from another camera? I bought two and i will se if I can bring it back

See the wiki.
But I think the original fw will not solve your problem.

fernando155, 2020-05-07:
> > Anyone knows how to get the original software from another camera? I bought two and i will se if I can bring it back
> 
> See the wiki.
> But I think the original fw will not solve your problem.

and do you know anyway to keep the wifi and password? I want to use it as motion sensor, keep the screenshoots and save videos, I don´t mind if it doesn´t work with the app, but right now is nothing that i cant do with it.

roleoroleo, 2020-05-07:
In any case, if "Disable cloud" is not enabled, the original features are running.
That's why you should be able to connect with the app.
Try to bring the cam closer to your access point.

fernando155, 2020-05-07:
> In any case, if "Disable cloud" is not enabled, the original features are running.
> That's why you should be able to connect with the app.
> Try to bring the cam closer to your access point.

the cam is just at 20 cm from the router. And I´m agree with you, It doesn´t make sense at all.

I´m just realice that the Hostname, I put YiCon, and I´m not able to modify it for some reason. I change it a few times but as soon as I reboot the camera, it comes with the same. Can be a possibility that because it calls different, yi app is not able to identify it?

roleoroleo, 2020-05-07:
No, the hostname is not a problem.

fernando155, 2020-05-09:
ok so... new software 0.3.2 installed and now... WORKING AGAIN!!!!! probrably that camera doesn´t like 0.3.1. Lets see how it works because last time the first 2h was fine

roleoroleo, 2020-05-09:
Good news.

closed 2020-05-24 by roleoroleo

ohlong, 2020-08-09:
Unfortunately, the same issue came up here, running 0.3.5.
The YI home app is stuck in "Retrieving Pairing Status" page.

Motion detection is not working and no recorded videos are stored anymore.
Tried to deactivate all features except ssh and httpd, still no luck.

Also I have problems connecting with ssh and ftp as well (before deactivating it), neither with blank password nor with a user-defined one.

Strange facts: 
-I noticed that the date and time is not correct, maybe this could be the origin of the problem?
-I cannot set a particular hostname and by default it is blank (used to be yi-hack).

roleoroleo, 2020-08-10:
Try to clean the sd card and to apply again the hack.

ohlong, 2020-08-10:
Thank you for your answer,

I formatted the SD card (8GB) and copied the two files again, home_h201c and sys_h201c.
The camera boots, the yellow light stays on for a few moments, and then starts turning on and off while speaking 'waiting to connect'. This process continues forever, and the fw flashing does not seem to be performed.

It is not necessary to use the Yi App actually, however the problem currently is that I do not have access to the ssh/ftp, and, more importantly, when I turn off and on the camera, it does not store the wifi password and I have to perform the pairing process every time; It manages to connect after all, however the pairing process fails and I am prompted to reset the camera.

ohlong, 2020-08-10:
Also, still no motion events are recorded, neither locally, nor on my "cloud" ftp.

roleoroleo, 2020-08-11:
If the connection phase is not completed the hack cannot work.
Try to pair again the cam with the app and then apply the hack.

ohlong, 2020-08-11:
I am afraid I cannot pair the cam anymore, I have tried to quite a few times.
The process goes on as expected, however the last step, "Retrieving Pairing Status" page in the Yi App cannot be completed.
Just before, the camera scans the QR code from the phone successfully, thus managing to connect to my home network, but the pairing process times out and I cannot use the camera from the app.
The result is a new page in the Yi App reading "Failed to load. Please reset the camera." and after some minutes the camera speaks that the pairing process has failed.

roleoroleo, 2020-08-11:
Did you try to reset the cam with the button?

ohlong, 2020-08-11:
Yes, actually I have not noticed any difference between pressing the reset button and pulling the cable; the said process is just repeated.

roleoroleo, 2020-08-12:
Do you have a backup of the original fw?

ohlong, 2020-08-12:
No, I have noted down the version (4.6.0.0A_201908271549), but I was not aware that I was able to keep a backup of the original fw.
Might you know if there is a downloadable version online?

roleoroleo, 2020-08-13:
Online there are only partial update.
Useless...
I will check If I saved it.