roleoroleo/yi-hack-MStar #53: difference between DOME and cloud DOME

grigoroff, 2019-12-23:
Hi,
i have YI-CLOUD DOME 1080P 6FUS with 0.2.0 (https://github.com/TheCrypt0/yi-hack-v4/releases) flashed firmware and i try to "upgrade" with yours.
I download h201c files and drop to clean FAT32 SDcard. But nothing changed and no yellow blink led on boot... I try with 4 different SDcard but nothing changed.

my questions are : 

1- Did the YI DOME 1080P 6FUS firmware can flash YI-CLOUD DOME 1080P 6FUS ?
2- I have ssh access can I flash manually and reboot ?

thanks for your work and answers.

roleoroleo, 2019-12-23:
If you are using v4 version  you can't use my version.
The hardware platform is different.

grigoroff, 2019-12-23:
thanks for your reply.
just to be sure i understood well : 
I use V4 but it's not the problem. The problem is DOME and CLOUD-DOME are different hardware.

that it ?

roleoroleo, 2019-12-23:
Yes.
My version works only on MStar platform (MStar ARM cpu).

grigoroff, 2019-12-23:
:-(
ok thanks a lot 

ps : my yi home 6FUS working like a charm with your and last release.
thanks again

closed 2019-12-23 by grigoroff