roleoroleo/yi-hack-MStar #54: ssh key

ranrinc, 2019-12-23:
Is there a possibility to turn of password and used SSH keys instead. So its more secure? If yes, where is the setting to not allowing plain text password?

Thanks for the great work

roleoroleo, 2019-12-24:
You can follow a "normal" dropbear guide.
Generate a key on your client device:
`ssh-keygen -b 4096`

Login to your camera and create the file /home/yi-hack/.ssh/authorized_keys
`touch /home/yi-hack/.ssh/authorized_keys`

Copy your key form ~/.ssh/id_rsa.pub to /home/yi-hack/.ssh/authorized_keys using vi (copy and paste).

Correct the file permissions:
`chmod 0600 /home/yi-hack/.ssh/authorized_keys`

Restart the cam:
```
sync
reboot -f
```

If you want to disable the password login, edit /home/yi-hack/script/system.sh and add the option "-s" to dropbear.

roleoroleo, 2019-12-24:
Please note that you need to repeat this configuration if you update a new release.

ranrinc, 2019-12-28:
@roleoroleo Thanks... got it working..

roleoroleo, 2020-01-09:
See wiki.

closed 2020-01-09 by roleoroleo