roleoroleo/yi-hack-MStar #55: BFUS - RTSP and Synology problem

cfus98, 2019-12-24:
I have two Yi cameras. The yi_home_1080p is working fine with Synology Surveillance station and the yi_dome (BFUSY...) stopped working after flashing newest yi-hack firmware. 

I can see the RTSP feed in VLC but the feed fails with Surveillance Station. I have tried rtsp://IP-CAM/ch0_0.h264 and rtsp://IP-CAM:554/ch0_0.h264 (user defined setup)

Camera FW: 4.6.0.0A_201908271549
Hack version: h201c

Any suggestions would be welcome or is this a bug that needs fixing?

Thanks for making this.

roleoroleo, 2019-12-24:
With which version did it work correctly?
With which version did it stop working?

cfus98, 2019-12-24:
It was working with the 2.2 version (Nov 29). Then for some reason the camera stopped working two days ago and I downloaded 2.3 (Dec 22) and reflashed. The firmware seems to be working everywhere but Synology Surveillance Station (version 8.2.6-6009). I cannot see a recent update in the Synology logs.

roleoroleo, 2019-12-24:
Please, try again with 0.2.2.

DaveThe, 2020-01-05:
I have the same problem with 0.2.4 version.
Synology does not see any onvif camera.
Home assistant with motion eye see the camera but show a grey screen.

have you any suggestions?

roleoroleo, 2020-01-05:
Probably is the same issue described here: https://github.com/roleoroleo/yi-hack-MStar/issues/36
And I can't solve it at the moment.

roleoroleo, 2020-01-07:
> 
> 
> I have the same problem with 0.2.4 version.
> Synology does not see any onvif camera.
> Home assistant with motion eye see the camera but show a grey screen.
> 
> have you any suggestions?

I'm trying with motioneye (standalone, not hassio add-on) and I have no problems.
I added as a RTSP cam with rtsp://IP-CAMERA/ch0_0.h264 and it works fine.

cfus98, 2020-01-08:
Hi roleoroleo, I updated to the new firmware 0.2.4 and now the rtsp feed works perfectly with my synology nas. Thanks for fixing it.

roleoroleo, 2020-01-08:
I didn't fix it intentionally.
Please verify if it works on the next 0.2.5.

nathankellenicki, 2020-01-09:
I posted in #36, however unfortunately 0.2.5 didn't fix the issue for me (homebridge-camera-ffmpeg).

ady-crow, 2020-11-23:
Hi roleoroleo. did i see you say that ONVIF is NOT working on Synology at the minute?
Will save me time if you confirm

thanks for the hard work

roleoroleo, 2020-11-23:
Yes, at the moment it's not working.
Synology implementation is not standard imho.

ady-crow, 2020-11-23:
Thanks

Ady

roleoroleo, 2021-02-11:
Released a new version: https://github.com/roleoroleo/yi-hack-MStar/commit/b644f1427e6e190e4be22bb1895be4e372f0edd7

closed 2021-02-11 by roleoroleo