roleoroleo/yi-hack-MStar #56: yi-hack-MStar 0.2.3 + BFUS cam - Low security ONVIF

chuk4ik, 2019-12-26:
Used ONVIF Device Manager 2.2.250. Connect from another location to an external ip camera. The program at the link http: // wan_ip_cam / onvif / device_service - receives full information on a stream of the form rtsp: // login: pass@IP-CAM/ch0_0.h264 without requesting a login / password.

Those. in fact, the program highlighted the login and password for access via http rtsp and ONVIF

From the settings there program is only adding a URI of the form: http: // ip_add_cam / onvif / device_service

ONVIF Device Manager 2.2.250
https://sourceforge.net/projects/onvifdm/

Sorry for the bad English.

roleoroleo, 2019-12-27:
Yes i know. At the moment it's a "feature".
But I can remove it, if necessary to publish the service.

chuk4ik, 2019-12-28:
> 
> 
> Yes i know. At the moment it's a "feature".
> But I can remove it, if necessary to publish the service.

And can this be done on / off "features"?
Perhaps it will be convenient for someone. I just turned off ONFIV for now. It may not look special to me (to curious hackers), but I’m afraid to shine logins / passwords.

roleoroleo, 2019-12-28:
I will remove login info in the next release.

roleoroleo, 2020-01-05:
Please try the last release.

closed 2020-02-07 by roleoroleo