roleoroleo/yi-hack-MStar #60: Upgrade form yi-hack-v4 doesn't seem to work

clustor, 2019-12-31:
Just trying to play with Yi 1080p Home with the hacks. I started with yi-hack-v4 as that's the first hit from Google. I am able to SSH just fine. 

I am trying to upgrade from latest yi-hack-v4 to y203c_0.2.3 but the camera doesn't flash the new firmware. 

1. I tried formatting the SD card and copying both home and sys files but no effect
2. I tried rename the sys file as rootfs but no effect

I double checked the MD5SUMs to make sure they match on the /tmp/sd partition as well. 

Hope I am not missing anything. Thanks for your work on this !!

roleoroleo, 2019-12-31:
If you are using v4 version you can't use my version.
The hardware platform is different.

clustor, 2020-01-01:
Interesting. How can I find if the hardware is based on MStar or Hi3518e? Just curious, if I can merge and build your code for Hi3518e with appropriate build chain.

roleoroleo, 2020-01-01:
`/ # cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 12.04
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : MStar Infinity3 (Flattened Device Tree)
Revision        : 0000
Serial          : 0000000000000000
`

closed 2020-02-22 by roleoroleo