roleoroleo/yi-hack-MStar #61: 9FUSY25xxx need firmware

egs2003, 2020-01-01:
What firmware do I need for this camera?
Attached I have a picture of the QRCode. The camera from the inside PCB side.  I have marked the UART pins. I have pulled and attached a DUMP of the original firmware (4.2.0.0H_201909041620). The dump I have made according to the description "https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware". The addresses of the partitions were the same.
if I use the script with the end address minus 1, I get an error message because of the multiple of 64

used commands
sf read 0x21000000 0x1d0000 0x1e0000
py script.py --timeout 1 --step 64 com5 115200 0x21000000 0x211e0000
sf read 0x21000000 0x3b0000 0xc30000
py script.py --timeout 1 --step 64 com5 115200 0x21000000 0x21C30000

Also the BOOT log. There is possibly a WLAN password in the dumps, but only a test WLAN. 
Hope I can contribute to the development and get a working software in return.

[image]
[image]
[image]

[putty.log](https://github.com/roleoroleo/yi-hack-MStar/files/4013933/putty.log)
[dumps.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4013934/dumps.zip)

roleoroleo, 2020-01-01:
It seems that your dump has a problem with a file.
In any case I have compared the content with this https://github.com/roleoroleo/yi-hack-MStar/issues/34 and it is the same.
So you can try the version y25_0.2.3.tgz

PS I updated the wiki.

egs2003, 2020-01-01:
What do you mean with "It seems that your dump has a problem with a file."
Can i use my dump to get back to the original firmware if necessary?

roleoroleo, 2020-01-01:
There is a file (oss binary in /home/app) with a lot of "0" in the middle.
Probably an error during the dump.
At the moment I think you cannot use it as a restore, but I can "repair" it if necessary.
Try to repeat the dump, only the home partition.

egs2003, 2020-01-01:
uff, this took me several hours

roleoroleo, 2020-01-01:
Yes, I know.
If you don't want, it is not necessary.

egs2003, 2020-01-02:
y25_0.2.3.tgz is working for me. thx
here is the new dump, the bytes are already swapped
[dump.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4016025/dump.zip)

roleoroleo, 2020-01-02:
I'm sorry, I was wrong: the corrupt oss file was the other, not yours.
I confirm that you can use your dump as a restore of the original fw.
Thanks for your support in correcting the wiki.

closed 2020-02-22 by roleoroleo