roleoroleo/yi-hack-MStar #62: Can't flash camera

SamPhoenix91, 2020-01-01:
I am trying to flash the camera with the firmware, but cannot get it to download.

It runs Yi, but when I boot with the SD card in nothing happens.

This is the camera: 
https://www.amazon.co.uk/gp/product/B07P8ZY7PB/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

arhunt35, 2020-02-06:
I had this. The way I got it to work was to format sd card in the app, pop into computer then copy across the files. I think the cam didn't like the way the card was formatted

supermanscottbristol, 2020-02-19:
Any joy?  I'm suffering the same thing.
SD Card works fine in the camera, formatted from camera, firmware added but it doesnt flash on bootup.

roleoroleo, 2020-02-19:
There are three possibilities: there are problems with the SD, you are using the wrong file (y203c, h201c or y25) or your cam is not supported.

supermanscottbristol, 2020-02-19:
Thanks @roleoroleo  - Just got there.  I reformatted, made a 16gb partition on the card and for whatever reason that worked !?

Thank you for the hard work @roleoroleo

roleoroleo, 2020-05-24:
Too much time since the last post.
Open a new issue if necessary.

closed 2020-05-24 by roleoroleo