roleoroleo/yi-hack-MStar #63: New feature : preset for PTZ

bart11064, 2020-01-02:
Hi !
I'd like to thank @roleoroleo for his great work, in particular adding PTZ support with the last release.

In addition to this PTZ support, i'd like to propose an enhancement : using the preset fonctionnality. With this fonctionnality, the position of the camera is stored and you can get back to the exact same position without having to use left/right/up/down features.

I made my own tests, using ipc_sniff @roleoroleo released in another thread.
Below is the result, hope it's enough to make it work.
What are your thoughts ?

```
=====
Add 1st preset
=====
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 00 40 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE

=====
Add 2nd preset
=====
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 00 40 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE

=====
Goto 1st preset
=====
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 02 40 01 00 04 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE

=====
Goto 2nd preset
=====
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 02 40 01 00 04 00 00 00 01 00 00 00
GOT UNRECOGNIZED MESSAGE

=====
Delete 1st preset
=====
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 01 40 01 00 04 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE

=====
Delete 2nd preset
=====
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 01 40 01 00 04 00 00 00 01 00 00 00
GOT UNRECOGNIZED MESSAGE
```
_**Preset creation message is unique
Presets id increment from 00 to 07
Maximum = 8 presets**_

roleoroleo, 2020-01-02:
Are you thinking of a web based support or an onvif support?

bart11064, 2020-01-02:
What kind of question ? Both of them of course !! 🤣

Seriously, it's for an automation through domotic, so i'll be happy with either onvif or web (which i can call through an HTTP request)

roleoroleo, 2020-01-02:
As you can imagine, the effort is quite different...
Do you want to save these presets from outside or just recall them?

Is there a way to get the number of presets already saved?
Maybe another ipc command?

bart11064, 2020-01-02:
Recall only is great. I can set my presets through the official app once for all, and recall them later.

I don't think there is a way to get existing presets (no ipc command, and nothing sniffed using official app).
Maybe the official app send the save command and links the preset id to my account...?

What about showing all the 8 presets by default : if recalling one not saved it won't do anything ?

Another direction I should look into ?

roleoroleo, 2020-01-02:
I will try to work on this feature.

roleoroleo, 2020-01-03:
Please try this beta:
https://drive.google.com/open?id=1_3JK-0XNa6zIbC2Khp_ysVYHFxDWCtsQ

bart11064, 2020-01-03:
Thanks roleoroleo
When using web interface this isn't working.
However, when I use `ipc_cmd -p x`, this is working as expected.

bart11064, 2020-01-03:
> 
> 
> Thanks roleoroleo
> When using web interface this isn't working.
> However, when I use `ipc_cmd -p x` through SSH, this is working as expected.

Sorry, had to clear my browser cache.
This is working also through web interface. This is perfect, thanks ! 👏

roleoroleo, 2020-01-03:
Did you try onvif also?

bart11064, 2020-01-03:
Did not see you made it through ONVIF.
Well, I tried with ONVIF Device Manager, and nothing happens. PTZ move is OK, but can't get goto preset function.

roleoroleo, 2020-01-03:
I forgot a line.
Change
```
onvif_srvd --pid_file /var/run/onvif_srvd.pid --model "Yi Hack" --manufa
cturer "Yi" --ifs wlan0 --port $ONVIF_PORT --scope onvif://www.onvif.org/Profile
/S $ONVIF_PROFILE_0 $ONVIF_PROFILE_1 $ONVIF_USERPWD --ptz --move_left "/home/yi-
hack/bin/ipc_cmd -m left" --move_right "/home/yi-hack/bin/ipc_cmd -m right" --mo
ve_up "/home/yi-hack/bin/ipc_cmd -m up" --move_down "/home/yi-hack/bin/ipc_cmd -
m down" --move_stop "/home/yi-hack/bin/ipc_cmd -m stop"
```
to
```
onvif_srvd --pid_file /var/run/onvif_srvd.pid --model "Yi Hack" --manufa
cturer "Yi" --ifs wlan0 --port $ONVIF_PORT --scope onvif://www.onvif.org/Profile
/S $ONVIF_PROFILE_0 $ONVIF_PROFILE_1 $ONVIF_USERPWD --ptz --move_left "/home/yi-
hack/bin/ipc_cmd -m left" --move_right "/home/yi-hack/bin/ipc_cmd -m right" --mo
ve_up "/home/yi-hack/bin/ipc_cmd -m up" --move_down "/home/yi-hack/bin/ipc_cmd -
m down" --move_stop "/home/yi-hack/bin/ipc_cmd -m stop" --move_preset "/home/yi-
hack/bin/ipc_cmd -p"
```
in system.sh

bart11064, 2020-01-03:
There it is ! ONVIF is working too !

Just for your knowledge, it seems that ONVIF Device Manager doesn't get the list of existing presets either.
It lists all the 8 presets and the set preset function can erase a previous one.
[image]

In addition, for information, it seems that the camera get back automatically to preset id 1 (and not id 0) after reboot. So ID1 should be the Home position.

roleoroleo, 2020-01-03:
The list of presets that I send with onvif doesn't contain any pan/tilt/zoom information.
Because I don't know how to get it from the system.
Therefore a preset contains only name and token (both equal to the preset number [0..7]).
The support is not fully compliant but it works.

If the home is preset 1 I will add to onvif data.

Thank you.

closed 2020-01-14 by roleoroleo