roleoroleo/yi-hack-MStar #64: ONVIF SOAP error

bart11064, 2020-01-04:
Hi

When trying to connect to the camera from my automation server (Jeedom) through ONVIF, I get the following error.

Request : 
```
<s:Envelope
    xmlns:s="http://www.w3.org/2003/05/soap-envelope">
    <s:Header>
        <Security s:mustUnderstand="1"
            xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd">
            <UsernameToken>
                <Username>%%USERNAME%%</Username>
                <Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">%%PASSWORD%%</Password>
                <Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">%%NONCE%%</Nonce>
                <Created
                    xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">%%CREATED%%
                </Created>
            </UsernameToken>
        </Security>
    </s:Header>
    <s:Body
        xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
        xmlns:xsd="http://www.w3.org/2001/XMLSchema">
        <GetCapabilities
            xmlns="http://www.onvif.org/ver10/device/wsdl">
            <Category>All</Category>
        </GetCapabilities>
    </s:Body>
</s:Envelope>
```

Response : 
```
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope
    xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope"
    xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding"
    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xmlns:xsd="http://www.w3.org/2001/XMLSchema"
    xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex"
    xmlns:wsa5="http://www.w3.org/2005/08/addressing"
    xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2"
    xmlns:xmime="http://tempuri.org/xmime.xsd"
    xmlns:xop="http://www.w3.org/2004/08/xop/include"
    xmlns:tt="http://www.onvif.org/ver10/schema"
    xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2"
    xmlns:wstop="http://docs.oasis-open.org/wsn/t-1"
    xmlns:tds="http://www.onvif.org/ver10/device/wsdl"
    xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl"
    xmlns:trt="http://www.onvif.org/ver10/media/wsdl">
    <SOAP-ENV:Body>
        <SOAP-ENV:Fault>
            <SOAP-ENV:Code>
                <SOAP-ENV:Value>SOAP-ENV:MustUnderstand</SOAP-ENV:Value>
            </SOAP-ENV:Code>
            <SOAP-ENV:Reason>
                <SOAP-ENV:Text xml:lang="en">The data in element 'Security' must be understood but cannot be processed</SOAP-ENV:Text>
            </SOAP-ENV:Reason>
        </SOAP-ENV:Fault>
    </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```

After a quick search on the web, I found this link : https://stackoverflow.com/questions/44084976/gsoap-fault-soap-envmustunderstandno-subcode
Do you think this could be applicable on this hack ? Is this related to #56 ?

Thanks for your help.

roleoroleo, 2020-01-04:
How can I replicate this error?
Did you enable security?

roleoroleo, 2020-01-04:
I'm realizing now that I compiled the daemon without security.
I will try to rebuild it but I think openssl is needed.

roleoroleo, 2020-01-05:
If you want to try...
[onvif_srvd.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4023426/onvif_srvd.tar.gz)

bart11064, 2020-01-07:
Sorry for the delay.
Everything works great, thank you!
Now, trying to figure out why GetMediaUri respond "400 Bad Request",even if I set correct inputs... But that's another story 😁

closed 2020-01-14 by roleoroleo