roleoroleo/yi-hack-MStar #65: wpa may crashed, now reset

ic0xgkk, 2020-01-05:
Hello,

Everytime I scan a wifi QRcode, then the camera will reboot. How to dael with it?

All log file are here: 

https://drive.google.com/drive/folders/1vBlisCxPbXK8HAaSw92DseCBPlYYkueL?usp=sharing

log file:

```
[
][1/1/0:3:58:325]: rmm.c(msg_proc-4523) pid[188][
][1/1/0:3:58:403]: dispatch.c(main-5303) open_ptz fail[
][1/1/0:3:58:476]: dispatch.c(get_config-1223) got sn(***********************)[
][1/1/0:3:58:476]: dispatch.c(get_config-1224) got pwd()[
][1/1/0:3:58:476]: dispatch.c(get_config-1225) got ssid()[
][1/1/0:3:58:476]: dispatch.c(get_config-1226) got tnp_init_string()[
][1/1/0:3:58:476]: dispatch.c(get_config-1234) reset because no wifi config[
][1/1/0:3:58:476]: dispatch.c(send_saveconfig_msg-426) send_saveconfig_msg msg snd success[
][1/1/0:3:58:476]: dispatch.c(get_config-1281) get_config save conf
[
][1/1/0:3:58:477]: dispatch.c(choose_server-650) in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, log_server = http://log.xiaoyi.com, sname = familymonitor-y25, dlproto = micn
[
][1/1/0:3:58:477]: dispatch.c(choose_server-650) in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, log_server = http://log.xiaoyi.com, sname = familymonitor-y25, dlproto = micn
[
][1/1/0:3:58:477]: dispatch.c(main-5385) init ok, cost time(2666 ms)[
][1/1/0:3:58:477]: dispatch.c(save_config-1024) got pwd()()[
][1/1/0:3:58:712]: dispatch.c(p_worker-3204) save_config[
][1/19/6:41:36:97]: dispatch.c(p_worker-3292) DISPATCH_SET_DEFAULT_TIME 1453185696[
][1/19/6:41:36:151]: rmm.c(init_font-972) font:/home/base/iso8859-1-24x48.bin
[
][1/19/6:41:36:203]: rmm.c(init_font-984) filesize=36864
[
][1/19/6:41:36:204]: rmm.c(init_font-1004) init font /home/base/iso8859-1-24x48.bin ptr size(4)[
][1/19/6:41:36:204]: rmm.c(init_font-972) font:/home/base/iso8859-1-8x16.bin
[
][1/19/6:41:36:204]: rmm.c(init_font-984) filesize=4096
[
][1/19/6:41:36:204]: rmm.c(init_font-1004) init font /home/base/iso8859-1-8x16.bin ptr size(4)[
][1/19/6:41:36:691]: rmm.c(rmm_init_isp-1480) pSensorID =8757 S C 2 2
[
][1/19/6:41:39:401]: cloud.c(sys_init-5517) open share mem ok
[
][1/19/6:41:39:584]: oss.c(main-2957) oss stream = ./oss_lapse, venc chn = 0x2000[
][1/19/6:41:39:684]: oss.c(main-2957) oss stream = ./oss_fast, venc chn = 0x1000[
][1/19/6:41:39:802]: oss.c(main-2976) fshare_open ok[
][1/19/6:41:39:802]: oss.c(main-2957) oss stream = ./oss, venc chn = 0x400[
][1/19/6:41:39:894]: oss.c(main-2976) fshare_open ok[
][1/19/6:41:40:15]: oss.c(main-2976) fshare_open ok[
][1/19/6:41:40:164]: p2p_tnp.c(main-6653) fshare_open ok[
][1/19/6:41:40:257]: p2p_tnp.c(p2p_set_tnp_init_status-2112) p2p_set_tnp_init_status 1 send_msg ok!
[
][1/19/6:41:40:257]: dispatch.c(do_mq_process-5058) invalid msg 0x1002
[
][1/19/6:41:40:358]: rmm.c(send_wait_msg-619) msg snd success[
][1/19/6:41:40:464]: watch_process.c(get_watch_info-116) check_interval=10
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=dispatch;cmd=cd /home/app;./dispatch &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=cloud;cmd=cd /home/app;./cloud &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=rmm;cmd=reboot;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=mp4record;cmd=cd /home/app;./mp4record &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=arp_test;cmd=cd /home/app;./arp_test &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=oss;cmd=cd /home/app;./oss &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=oss_fast;cmd=cd /home/app;./oss_fast &;
[
][1/19/6:41:40:554]: watch_process.c(get_watch_info-134) process=oss_lapse;cmd=cd /home/app;./oss_lapse &;
[
][1/19/6:41:40:671]: rmm.c(zbar_proc-4407) decoded QR-Code symbol "***********************"
[
][1/19/6:41:40:758]: rmm.c(trans_info-441) trans_json info=***********************
[
][1/19/6:41:40:758]: rmm.c(trans_info-513) trans_json result=***********************
[
][1/19/6:41:40:758]: rmm.c(trans_info-441) trans_json info=***********************
[
][1/19/6:41:40:758]: rmm.c(trans_info-513) trans_json result=***********************
[
][1/19/6:41:40:758]: rmm.c(trans_info-441) trans_json info=***********************
[
][1/19/6:41:40:759]: rmm.c(trans_info-513) trans_json result=***********************
[
][1/19/6:41:40:759]: rmm.c(zbar_proc-4430) rmm got /tmp/got_wpa ssid(***********************) pwd(***********************) bind(***********************)
[
][1/19/6:41:40:759]: dispatch.c(do_mq_process-5058) invalid msg 0x1005
[
][1/19/6:41:40:825]: rmm.c(send_scanok_msg-646) msg snd success[
][1/19/6:41:40:876]: dispatch.c(choose_server-650) in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, log_server = http://log.xiaoyi.com, sname = familymonitor-y25, dlproto = micn
[
][1/19/6:41:40:930]: rmm.c(send_wifi_conf_msg-676) msg snd success[
][1/19/6:41:41:11]: dispatch.c(do_mq_process-3413) got pwd(***********************)[
][1/19/6:41:41:64]: dispatch.c(send_connectting_msg-138) msg snd success[
][1/19/6:41:41:65]: dispatch.c(do_mq_process-5058) invalid msg 0x1004
[
][1/19/6:41:41:361]: dispatch.c(do_monitor_wifi-1668) wifi disconnected, now reconnect wifi
[
][1/19/6:41:41:841]: rmm.c(msg_proc-4563) got RMM_SPEAK_WAIT[
][1/19/6:41:44:714]: rmm.c(msg_proc-4648) got RMM_SPEAK_SCAN_OK[
][1/19/6:41:46:676]: rmm.c(msg_proc-4583) got RMM_SPEAK_CONNECTTING[
][1/19/6:41:50:428]: dispatch.c(do_monitor_wifi-1610) wpa may crashed, now reset
[
][1/19/6:41:50:547]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:41:50:632]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:41:50:632]: watch_process.c(check_watch_info-171) arp_test crashed![
][1/19/6:41:50:632]: dispatch.c(do_monitor_wifi-1615) no usb dev

```

roleoroleo, 2020-01-05:
Which model?
Do you have a log where I can see the u-boot output?

ic0xgkk, 2020-01-05:
Model: 5FCNY0EGJL

Your y25 firmware is the only one that I can boot in.  

Console log with u-boot output:

```
▒▒▒▒▒▒▒▒
IPL gd156225
D-01.
HW Reset
64MB
BIST0_0001-OK
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x00041314

-----------------------U-Boot 2015.01 (Feb 01 2018 - 10:42:26)-----------------------

Version: I3g901be38
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 one.bin
    one.bin Not exist(fatsize err)

+++++++++++++++++++   check one_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 one_y23
    one_y23 Not exist(fatsize err)

+++++++++++++++++++   check uboot_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 uboot_y23
    uboot_y23 Not exist(fatsize err)

+++++++++++++++++++   check kernel_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 kernel_y23
    kernel_y23 Not exist(fatsize err)

+++++++++++++++++++   check sys_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 sys_y23
    sys_y23 exist, file_size(1733872)
------>fatload mmc 0 0x21000000 sys_y23
reading sys_y23
1733872 bytes read in 99 ms (16.7 MiB/s)
------>crc32 0x21000000 0x1a74f0 0x23b3bb24
[... 352 lines cut ...]
MSYS: DMEM kept entry found: name=S2:VENCDMP0, phys=0x236F0000, length=0x00056400
MSYS: DMEM request: [S2:VENCDMP1]:0x00056400
MSYS: DMEM kept entry found: name=S2:VENCDMP1, phys=0x23750000, length=0x00056400
MSB2@v1.1-01:r&d analysis.
MSYS: DMEM request: [VSPL-I0P3B0]:0x0005A000
MSYS: DMEM kept entry found: name=VSPL-I0P3B0, phys=0x237B0000, length=0x0005A000
MSYS: DMEM request: [VSPL-I0P3B1]:0x0005A000
MSYS: DMEM kept entry found: name=VSPL-I0P3B1, phys=0x23810000, length=0x0005A000
MSYS: DMEM request: [VENC-50]:0x00025800
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
MSYS: DMEM kept entry found: name=VENC-50, phys=0x23870000, length=0x00025800
MSYS: DMEM request: [S3:VENCDMOUT]:0x00005600
MSYS: DMEM kept entry found: name=S3:VENCDMOUT, phys=0x22F98000, length=0x00005600
MSYS: DMEM request: [S3:VENCDMP0]:0x00056400
MSYS: DMEM kept entry found: name=S3:VENCDMP0, phys=0x238A0000, length=0x00056400
MSYS: DMEM request: [S3:VENCDMP1]:0x00056400
MSYS: DMEM kept entry found: name=S3:VENCDMP1, phys=0x23900000, length=0x00056400
MSYS: DMEM request: [VENC-51]:0x00025800
MSYS: DMEM kept entry found: name=VENC-51, phys=0x23960000, length=0x00025800

[PID_LIST] pid_list_init ok, [ ver=Apr 28 2018, 17:10:59 ]
random: dropbear urandom read with 29 bits of entropy available

```

roleoroleo, 2020-01-05:
As you can see in your log, the file accepted is y23.
This model is not yet supported from my hack.

ic0xgkk, 2020-01-05:
Unfortunately, my camera have been daed and no firmware provided by official. I have rename the _y25 to _y23, and boot it in successfully. I want to have a try with y25 firmware to rescue my camera haha. 

As you can see, the main board show a name **y25_mb_mt1.1**.

roleoroleo, 2020-01-05:
If you remember exactly the firmware version (i.e. 4.5.0.0C_201910080934) I could try to create a frankenstein fw.

ic0xgkk, 2020-01-06:
Thank you. I save the log when my camera worked, the firmware version is **2.1.0.0E_201809191630**

roleoroleo, 2020-01-06:
Ok, try this fw:
[home_y23.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4025521/home_y23.zip)

Unzip it before copy on sd.
It's a mixed firmware y25/y23.

ic0xgkk, 2020-01-06:
Some thing went wrong, it show following message after scan the QRcode and reboot.

```
watchdog watchdog0: watchdog did not stop! 
reboot: Restarting system
```

Full console log:

```

-----------------------U-Boot 2015.01 (Feb 01 2018 - 10:42:26)-----------------------

Version: I3g901be38
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x0B05, 0xC8, 0x40, 0x18)
SF: Detected nor0 with total size 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.

+++++++++++++++++++   check one.bin    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 one.bin
    one.bin Not exist(fatsize err)

+++++++++++++++++++   check one_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 one_y23
    one_y23 Not exist(fatsize err)

+++++++++++++++++++   check uboot_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 uboot_y23
    uboot_y23 Not exist(fatsize err)

+++++++++++++++++++   check kernel_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 kernel_y23
    kernel_y23 Not exist(fatsize err)

+++++++++++++++++++   check sys_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 sys_y23
    sys_y23 Not exist(fatsize err)

+++++++++++++++++++   check home_y23    +++++++++++++++++++
------>setenv filesize 0
------>fatsize mmc 0 home_y23
    home_y23 exist, file_size(6820164)
------>fatload mmc 0 0x21000000 home_y23
reading home_y23
6820164 bytes read in 349 ms (18.6 MiB/s)
------>crc32 0x21000000 0x681144 0x23b3bb24
CRC32 value from addr 0x21000000 -> 0x21681143 is 0xeac33c19
    crc same, skip ......
[NetUpgrade] ts_1st=0x9ea
No ethernet found.
[... 278 lines cut ...]
MSYS: DMEM request: [VENC-48]:0x00025800
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
mrqc_set_rqcf - skip set RQCT_CFG_SEQ
MSYS: DMEM request: [S1:VENCDMOUT]:0x00005600
MSYS: DMEM request: [S1:VENCDMP0]:0x00056400
MSYS: DMEM request: [S1:VENCDMP1]:0x00056400
MSYS: DMEM request: [VSPL-I0P3B0]:0x00016800
MSYS: DMEM request: [VSPL-I0P3B1]:0x00016800
MSYS: DMEM request: [VSPL-I0P3B2]:0x00016800
watchdog watchdog0: watchdog did not stop!
reboot: Restarting system
1▒▒▒▒▒▒▒▒
IPL gd156225
D-01.
SW Reset
64MB
BIST0_0001-OK
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x00041314

```

roleoroleo, 2020-01-06:
No luck...

ic0xgkk, 2020-01-07:
emmmm, a sad story....

Thank you

roleoroleo, 2020-05-24:
Added y23 support.
Wait for the next release.

closed 2020-05-24 by roleoroleo