roleoroleo/yi-hack-MStar #66: Wifi configuration lost every power cycle

sfs325, 2020-01-08:
I have a Yi 1080p Home 6FUS camera. I flashed the firmware and it works perfectly. The problem I have now is that everytime I power cycle the camera it loses the wifi configuration and I need to cofigure it again (I start listeing waiting to connect and so on).

Best regards,
Sergio

Rocket200, 2020-01-08:
i have same Problem with a 4FUS 3-4 weeks before

I can not finished the wifi connect process with the Original APP (disabled cloud)
but I can now go back to the hack (webif) and I enabled there the cloud Service again and reboot.
and again I connect the cam with the Original App and after a reboot I go back to the hack (webif) and disabled the cloud again.

this work for me 

I think the Original APP need to save the WIFI SSID and Password somewhere on the CAM

sfs325, 2020-01-08:
Thank you very much for your comment.

I didn't really understand your procedure. What I understood is, I should configure it with the cloud enabled and then I can disable it? (I have it disabled now.)

Rocket200, 2020-01-08:
Yes Same me bevor 
Enable the cloud again 
Connect with the APP 
I Reboot sometime 
at Last disable the cloud again

mklan, 2020-03-12:
Sadly this does not work for me. It still loses wifi on each power cut. Anyone with the same problem?

alexkutsan, 2020-04-11:
I have the save issue. i was looking into log. txt and find a lot of resolving hostname error. How to fix dns ? 
```
/tmp # cat log.txt | grep resol
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
curl_easy_perform() failed: {"curl_code":6} Couldn't resolve host name
][1/19/6:48:0:806]: p2p_tnp.c(tnp_proc-6408) P2P Server IP resolved : NO
][1/19/6:53:20:807]: p2p_tnp.c(tnp_proc-6408) P2P Server IP resolved : NO

```

divemasterjm, 2020-09-03:
same issue here, every reboot i have to show qr code for wifi connection

executer-uno, 2020-09-09:
Same problem, https://github.com/shadow-1/yi-hack-v3/issues/274
Does someone know where WIFI SSID and Password could be stored in camera? Then It will be easy to connects by WinSCP and edit this file manually.

roleoroleo, 2020-09-13:
They are stored in a raw partition.
You can't edit it with winscp.

divemasterjm, 2020-09-27:
> They are stored in a raw partition.
> You can't edit it with winscp.

roleoroleo any progress? is very anoying having cameras without connection after rebooting or power/router failure

roleoroleo, 2020-09-27:
No, I tried different configurations but I can't replicate this problem.

krue81, 2020-10-04:
Since last night, I've been having the same issue with a Yi Dome 1080p. I rebooted the camera and began to hear the "Waiting to connect" voice... I tried to access the camera via the Yi Home App to no avail... deleted the camera from the app and resetted it to add it again... The camera reads the QR Code successfully and connects my Wifi network, but the pairing process stops at retrieving pairing status (after 2 or 3 minutes) and shows a "Failed to load, please reset the camera" message. 

At this point, I can access the camera via its ip:8080, but as soon as I reset it, I lose wifi connection and the camera starts again the pairing process (Waiting to connect...).

roleoroleo, 2020-11-21:
If anyone still has this problem I made a wiki page which might be useful:

https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials

divemasterjm, 2020-11-23:
i've tried it but still the same

ethempekin, 2020-11-26:
I have this problem with h201c. Local time is also not updating. 
My fw 0.3.5 I wonder if I should update it to 0.3.8 can solve the problem.

roleoroleo, 2020-11-26:
I found an error in the script sdwifi and I updated it in the wiki.
Please try again with the new version.

divemasterjm, 2020-11-26:
doesn't work (((

ethempekin, 2020-11-26:
My problem resolved when upgrading to 0.3.8.

roleoroleo, 2020-11-27:
@divemasterjm 
Could you try again?
I improved the script more.

divemasterjm, 2020-11-27:
i tried your script, and every time i reboot camera still waiting for wifi

divemasterjm, 2020-11-27:
ive inserted sd card
reboot camera
keep asking waiting to connect

roleoroleo, 2020-11-27:
Ok, I will check again.

executer-uno, 2020-11-29:
Hi, thanks for your effort!
I updated my Yi 1080p Home 4FCN with actual FW https://github.com/roleoroleo/yi-hack-MStar/releases/tag/0.3.8
All behaves the same. Scan QR -> WiFi connected and web interface ok. Reboot -> Waiting to connect...

After I try a trick https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials but looks like nothing happened.
After reboot I check with WinSCP if somewhere configure_wifi.cfg stored on camera, but no files found in root and subfolders. May I help somehow more? To check it with WinSCP or Telnet etc...?

roleoroleo, 2020-11-29:
Check if your settings are correctly stored in flash:
`hexdump -C /dev/mtd/mtd5`

executer-uno, 2020-11-29:
In /tmp/sd/log/log_wifi_connected.tar.gz i found some kind of log data. here it is attached (with cleared passwords)
[wifi_connected.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5612368/wifi_connected.zip)

executer-uno, 2020-11-29:
> Check if your settings are correctly stored in flash:
> `hexdump -C /dev/mtd/mtd5`

[image]
Yes. SSID and Password are stored in mtd5. But I can check it only if I already show QR to camera an it connects to WiFi. I can not check if file is there while it is telling "waiting to connect" after reboot.

roleoroleo, 2020-11-30:
This is not correct.
[image]
This word should be "00".
Did you use the last sdwifi.tar.gz?

executer-uno, 2020-11-30:
Yep. The latest (today rechecked from https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials)
Still no luck, but I done some investigations, could be make some new idea:
1) I reboot camera with FAT32 SD card with the yi-hack folder on it. It still boots same way. LED...click...wellcome...waiting to connect... I show it QR with wifi config - it becomes online. BUT. In Startup.sh there is "/tmp/sd/yi-hack/configure_wifi.sh" - looks like SD card script call, but there is no yi-hack folder in /tmp/sd/ when I check it in WinSCP.
[image]

2) I check if SD card content appears somewhere else in hierarchy - no. But i found /etc/init.d/yi-hack file with some script callup. So i try to append yi-hack/configure_wifi.sh there, to be sure script will be called. SD content copyed to /etc/init.d/yihack folder:
[image]
But it makes no any change after reboot either. I even made chmode +x for configure_wifi.sh, So it should be executable.

3) SD card formatted in Win10 as FAT32. Its just 1GB size, clear and only yi-hack folder contained. I also try with folder contents copyed to SD root. It not helped. Should I try with some proper SD card format tool?

roleoroleo, 2020-12-01:
Probably when the cam is not configured, the script is not executed.

executer-uno, 2020-12-01:
[PuTTY log cold boot wo MMC.log](https://github.com/roleoroleo/yi-hack-MStar/files/5624563/PuTTY.log.cold.boot.wo.MMC.log)
[putty_log - show QR and connect to WiFi.log](https://github.com/roleoroleo/yi-hack-MStar/files/5624564/putty_log.-.show.QR.and.connect.to.WiFi.log)
[PuTTY log cold boot with MMC.log](https://github.com/roleoroleo/yi-hack-MStar/files/5624565/PuTTY.log.cold.boot.with.MMC.log)

I managed to get a log from UART. But can not catch 
"Stop the boot at the U-Boot prompt (should be "MStar #") pressing "Enter"."
to send any commands. 
In logs I see nothing interesting. Almost no difference between boot with SD and without SD, so, looks like sripts not starts.

Will keep trying to catch "MStar#" prompt on boot up.

executer-uno, 2020-12-01:
Ok, just putty restart was required. Any helpful things I can do in MSTAR# prompt?
Here is 'help' response:
```
MStar # help
?       - alias for 'help'
base    - print or set address offset
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
cmp     - memory compare
cp      - memory copy
crc32   - checksum calculation
dbg     - set debug message level. Default level is INFO
dcache  - enable or disable data cache
debug   - Disable uart rx via PAD_DDCA to use debug tool
dhcp    - boot image via network using DHCP/TFTP protocol
dstar   - script via SD/MMC
eeprom  - EEPROM sub-system
env     - environment handling commands
estar   - script via network
estart  - EMAC start
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatread - FAT fatread with FSTART
fatsize - determine a file's size
go      - start application at address 'addr'
gpio    - Config gpio port
help    - print command description/usage
i2c     - I2C sub-system
icache  - enable or disable instruction cache
initDbgLevel- Initial varaible 'dbgLevel'
loop    - infinite loop on address range
macaddr - setup EMAC MAC addr
md      - memory display
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mssdmmc - Mstar SD/MMC IP Verification System
mstar   - script via TFTP
mw      - memory write (fill)
mxp     - MXP function for Mstar MXP partition
net_upgrade- do net update from the specified file that is in tftpserver

nm      - memory modify (constant address)
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
reset   - Perform RESET of the CPU
riu     - riu  - riu command

run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
sdupgrade- do SD card auto upgrade - one.bin

setenv  - set environment variables
sf      - SPI flash sub-system
sfbin   - for uploading sf image to a server(via network using TFTP protocol)
srcfg   - sensor pin and mclk configuration.
tftpboot- boot image via network using TFTP protocol
version - print monitor, compiler and linker version
```

executer-uno, 2020-12-01:
Hmm. It works!
sdwifi.tar.gz really helps!

It could be a improperly formatted SD card by Win10. It works after I formatted it to FAT32 with RuFus. It was one of the steps. And I doubt about Win10 issue, because I can not reproduce it with FAT16 full format -> FAT32 quick format - it still works.

Also I digging around in uboot MStar# prompt. Making:
```
mmc rescan
mmc part
fatls mmc 0
fatls mmc 0 /yi-hack
```
I seen that files are visible and available. Then I perform reset command - camera reboots twice, as described in sdwifi.tar.gz wiki and it becomes available in my home wifi network. Nice. I have a log output available, but there are even no script echo output, so I cant see anything interesting that can points the initial problem.
PS: Surely I tried to reboot camera before, without supply cut, from web interface.

PPS: hexdump -C /dev/mtd/mtd5 now shows zeroes before SSID, as @roleoroleo sad. Great, but not clear ))

All UART console actions log:
[putty_log - script finally started.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5625142/putty_log.-.script.finally.started.zip)

roleoroleo, 2020-12-01:
Great!
Maybe @divemasterjm could try another SD card or another format system?

PS I found that it's possible to load a custom wpa_supplicant.conf from SD.
Unfortunately is not permanent, if you reboot without SD the cam starts with the old credentials.

Create wpa_supplicant.conf in the folder Factory of the SD card.
```
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="YOUR_SSID"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="YOUR_PASSWORD"
}
```

divemasterjm, 2020-12-01:
i formated with rufus and put also factory folder and it worked!!!
thanks both of you!!!

roleoroleo, 2020-12-19:
Thank you for your support.
I close this issue.

closed 2020-12-19 by roleoroleo