roleoroleo/yi-hack-MStar #67: Support for Hi3518e cameras, like Yi Outdoor?

grungeguitarist, 2020-01-08:
Hi,

I really like this Yi hack. It is much better than the yi-hack-v4 by TheCrypt0s. I would like to have all the new features (snapshot, ONVIF, better RTSP) for Hi3518e Yi cameras, like the Yi Outdoor. Unfortunately, it only works with MStar chipset models.

Is it planned to support Hi3518e Yi cameras in the near future? This would be really great!

roleoroleo, 2020-01-08:
It's not planned.
And there are some features that cannot be supported on Hi3518e.
For example RTSP and image snapshot, because the kernel driver is different.
Other features probably would be portable: ONVIF and PTZ.
But it's not planned.

closed 2020-01-14 by roleoroleo

odelma, 2020-10-16:
Now that Yi Outdoor support has been added to yi-hack-v4 , could it be reconsidered for MStart? I really appreaciate all the features on this hack like snapshot and ONVIF, which are missing from the hack-v4. Or is there some findamental difference on that hack which allows it to be used on Yi Outdoor?

roleoroleo, 2020-10-18:
The difference is the platform.
The hardware is different and this hack can't work on the hi3528e cameras.
Only high level software could work: for example onvif yes and snapshot no.

odelma, 2020-10-20:
Ok, thanks for confirming.