roleoroleo/yi-hack-MStar #71: PTZ and Home Assistant

ZioTitanok, 2020-01-17:
I'm trying to import my Yi Dome 1080p in Home Assistant.
It was super easy to import the stream, using the ONVIF component.
On the other hand, the Camera doesn't reply on PTZ command request via Home Assistant, while it does via a generic Windows software and, of course, via web interface.

Any suggestion?
Thank you.

roleoroleo, 2020-01-17:
Unfortunately I don't have a dome cam so I can run only few tests.

Regarding your issue I think there is a problem with ha, I tested it now and doesn't work:
https://github.com/home-assistant/home-assistant/issues/27744
https://github.com/home-assistant/home-assistant/pull/30152

But there is a workaround, if you want.
When I tested the software I did not configure the ptz inside onvif but as a set of switches.
configuration.yaml:
```
  - platform: template
    switches:
      switch_cam_left:
        value_template: "off"
        turn_on:
          service: rest_command.ptz_cam_left
        turn_off:

rest_command :
  ptz_cam_left:
    url: http://192.168.0.200:8080/cgi-bin/webui?command=ptzl
```
And in lovelace:
```
    cards:
      - camera_image: camera.cam
        entities:
          - entity: switch.switch_cam_left
            icon: 'mdi:arrow-left-bold-circle-outline'
        title: MyCam
        type: picture-glance
```

ZioTitanok, 2020-01-18:
Nice workaround. Would be perfect till HA team find a fix.

I'm for sure setting something in the wrong way (of course I changed the IP). I get a 404 Error from HA: "Error. Url: http://192.168.1.2:8080/cgi-bin/webui?command=ptzl. Status code 404."

Just to be sure: are the other commands ptzl, ptzr, ptzu and ptzd (left, right, up, down)?

roleoroleo, 2020-01-18:
Maybe I don't remember exactly.
Go to the web page with the PTZ control and try if it works.
Then try the command with a browser.

ZioTitanok, 2020-01-18:
Found:

rest_command:
  ptz_cam_left:
    url: http://CAMERA_IP:8080/cgi-bin/ptz.sh?dir=LEFT 

And of course, RIGHT, UP and DOWN.
Hope this can help.

roleoroleo, 2020-01-19:
Correct!
There is another possibile option to increase or decrease the size of movement: TIME=x
x is the time between the move command and the stop command.
Default is 0.3 s.
For example:
```
http://CAMERA_IP:8080/cgi-bin/ptz.sh?dir=LEFT&time=0.8
```

closed 2020-01-28 by roleoroleo