roleoroleo/yi-hack-MStar #72: Help needed to identify cam

frankcom, 2020-01-17:
Hello, I highly appreciate your work on yi-Hack, but now I need some help :)
I have bought a "Mibao" cam on Amazon a few months ago and I am currently using the recommended YI IoT app to make it working.
I highly expect that the device is being a "YI Technology" cam, but I do not know how I can identify, which type of cam.
There is only the text "Smart Camera" on the label - no exact description.
Any idea how I could verify it? 
Many thanks

roleoroleo, 2020-01-17:
The model number and the labels are a good start.
But I suggest you to open it and take some pictures.

frankcom, 2020-01-17:
Thanks for your quick reply. I hope attached material helps.
[image]
[image]
[image]
[image]

roleoroleo, 2020-01-19:
In the pictures I can't see anything that makes us understand architecture.
But it doesn't look like anything known.
I think that my firmware will not work on this.

frankcom, 2020-01-19:
Thanks, I hoped for positive response as the app which is working is from yi technology :)

closed 2020-01-28 by roleoroleo