roleoroleo/yi-hack-MStar #77: Feature Request: Google drive uploader

sorokin-e, 2020-01-24:
Hello,

Auto upload of the recorded footage to the cloud (eg. Google Drive, Dropbox, etc.).
I think it's useful and demanded feature.  

Or at least working 'curl' package - this allows realizing necessary feature:
[https://github.com/porunov/xiaomi_gdrive](https://github.com/porunov/xiaomi_gdrive)

roleoroleo, 2020-01-27:
If I cross compile all necessary softwares (curl, libcurl, libz) , are you able to test the scripts?

sorokin-e, 2020-01-27:
Yep, sure. I'm ready to help.

roleoroleo, 2020-01-27:
Here it is:
[gdrive.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4117684/gdrive.tar.gz)

skylarhays, 2020-01-28:
Wow!!! Great work guys!

sorokin-e, 2020-01-29:
Hi,
It seems we need not only working 'curl'
From GDriveConf file I've found out about :
```
rm /bin/egrep
rm /bin/expr
rm /bin/stat
rm /bin/wc

ln -s $YI_HACK_PREFIX/yi-hack-v4/bin/busybox /bin/egrep
ln -s $YI_HACK_PREFIX/yi-hack-v4/bin/busybox /bin/expr
ln -s $YI_HACK_PREFIX/yi-hack-v4/bin/busybox /bin/stat
ln -s $YI_HACK_PREFIX/yi-hack-v4/bin/busybox /bin/wc
```
But linkage doesn't work for me:
```
cp /tmp/sd/gdrive/yi-hack-v4/bin/busybox /home/yi-hack/bin/
ln -s /home/yi-hack/bin/busybox /bin/wc
# wc
sh: wc: not found
```
Built-in busybox doesn't support these commands:
`# busybox --list |grep -w -E "wc|stat|egrep|expr"`

I think we need to rebuild `busybox ` with these commands.

Take care,

roleoroleo, 2020-01-29:
I could add commands to the existing busybox but there is very little space available on the / root file system.
I think it would be better to find alternative commands.
Egrep can probably be replaced with "grep -E".
expr could be replaced with double parentheses and the dollar - e.g. $(($x+1))
wc could be replaced by "grep -c ^ filename"
...

sorokin-e, 2020-02-07:
I've tried to adapt gdrive scripts, but currently faced with curl issues:
```
/tmp/sd # sh gdrive/GDriveConf
Configuration is started. Wait for several seconds.
Enter your Client ID:
1062585917265-48r2oq*****************
Enter your Client secret:
bRMPLiuOa-********************

Put this link into your browser:
https://accounts.google.com/o/oauth2/v2/auth?scope=https://www.googleapis.com/auth/drive&state=state&redirect_uri=urn:ietf:wg:oauth:2.0:oob&response_type=code&client_id=1062585917265-48*************************5.apps.googleusercontent.com&access_type=offline

Copy and paste your code here:
4/wQG-xGl-hTmauebDwUJN7***************************
curl: (1) Protocol "https" not supported or disabled in libcurl
```
I think curl is compiled without SSL support.

roleoroleo, 2020-02-07:
Ok, I will rebuild curl with ssl.

roleoroleo, 2020-02-07:
Here it is:
[gdrive.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4172192/gdrive.tar.gz)

buzlightyear, 2020-02-08:
lack of libcrypto.so.1.1 and libssl.so.1.1

buzlightyear, 2020-02-08:
For libssl I tryed with simply copy/paste libssl.so.1.0.0 in libssl...1.1, but for libcrypto it gives error `curl: /tmp/sd/yi-hack/lib/libcrypto.so.1.1: no version information available (required by /tmp/sd/yi-hack/lib/libcurl.so.4)`

roleoroleo, 2020-02-08:
Sorry, wrong archive.
Try this:
[gdrive.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4174002/gdrive.tar.gz)

buzlightyear, 2020-02-09:
Gives me error while looking into gdrive folder. Later I will check what's went wrong, of the other guy can confirm if also him had the same error could be a good thing :)

sorokin-e, 2020-02-10:
Need help with this function:
```
get_next_dir(){
	newDir=""
	for n in $(ls -1 ${record_dir} | grep -iv tmp); do
        if [ "$(expr ${current_dir} \< ${n})" = "1" ]; then
            newDir=$(echo ${n})
            break
        fi
    done
    echo ${newDir}
}
```
To avoid using "expr" I'm using double parentheses and the dollar like this:
```
	if [ "$((${current_dir} \< ${n}))" = "1" ]; then
or
        if [ "$($((${current_dir} \< ${n})))" = "1" ]; then
```
But anyway get errors:
```
/tmp/sd/gdrive/GDriveUploader: line 93: arithmetic syntax error
```
Completely messed with these parentheses :(

roleoroleo, 2020-02-11:
Try this:
```
get_next_dir() {
    newDir=""
    for n in $(ls -1 ${record_dir} | grep -iv tmp); do
        if [ "$current_dir" \< "$n" ]; then
            newDir=$(echo ${n})
            break
        fi
    done
    echo ${newDir}
}
```

sorokin-e, 2020-02-11:
It works somehow!
Very rough draft, but files uploads: https://github.com/sorokin-e/xiaomi_gdrive
Welcome to test ^_^

naeloob, 2020-02-29:
I would use rclone for that.

Version Linux ARM 32Bits
Latest right now : https://github.com/rclone/rclone/releases/download/v1.51.0/rclone-v1.51.0-linux-arm.zip

Rclone needs the certificates to validate https conections and it look for them on /etc/ssl/certs/ca-certificates.crt

So, we download the certificates to the sd :
 wget -O /tmp/sd/ca-certificates.crt https://raw.githubusercontent.com/bagder/ca-bundle/master/ca-bundle.crt

Create the path :
 mkdir /etc/ssl
 mkdir /etc/ssl/certs

And create a link : 
 ln /tmp/sd/ca-certificates.crt /etc/ssl/certs/ca-certificates.crt -s

This way you can use rclone with every remote it supports.
Disclaimer : there's very low memory free, it can limite the use of rclone. For example, there's no free ram to use the crypt remote, to encrypt the files.

closed 2020-05-24 by roleoroleo

majkers, 2020-05-27:
When I enter code from browser I get this error:

`/tmp/sd/gdrive/GDriveConf: line 62: curl: not found
grep: /tmp/sd/gdrive/conf/response: No such file or directory
grep: /tmp/sd/gdrive/conf/response: No such file or directory
Your code was expired or it is not correct. Please refresh a link in your browser and enter a new valid code:
`

What might be wrong?

michael11123, 2020-05-28:
[image]
i have same problem

roleoroleo, 2020-05-28:
Did you copy curl on /home/yi-hack/bin?

michael11123, 2020-05-28:
can you guide me more Detail?
i have try copy curl from gdrive/yi-hack-v4/bin  to  /home/yi-hack/bin. it did'n work.
[image]

michael11123, 2020-05-29:
> Did you copy curl on /home/yi-hack/bin?
> Or better, add the directory to the PATH variable.

pls help me.

michael11123, 2020-05-29:
[image]

roleoroleo, 2020-07-07:
@michael11123 
Sorry for the late reply, I missed this issue beacuse it's closed.
Do you still need help?

Dmitrory, 2020-08-20:
I had the same errors. 
https://github.com/roleoroleo/yi-hack-MStar/issues/77#issuecomment-635377892

I added curl to PATH and got rid of the first one, but what about the error
[image]

Can you help me?

roleoroleo, 2020-08-27:
https://github.com/roleoroleo/yi-hack-MStar.gdrive

Dmitrory, 2020-09-01:
> https://github.com/roleoroleo/yi-hack-MStar.gdrive

Bus error
grep: /tmp/sd/yi-hack/gdrive/conf/response: No such file or directory
Your code was expired or it is not correct. Please refresh a link in your browser and enter a new valid code:
grep: /tmp/sd/yi-hack/gdrive/conf/response: No such file or directory

roleoroleo, 2020-09-01:
Which command are you running?

Dmitrory, 2020-09-01:
> 
> 
> Which command are you running?

`sh /tmp/sd/yi-hack/gdrive/GDriveConf`
Such a problem after entering the code that was received after authorization about the link.

roleoroleo, 2020-09-01:
Check if the file is in linux format: lf and not crlf.

hetzbh, 2020-10-16:
Sorry for commenting, but do you plan to add it to an upcoming version? (like 3.1)?

roleoroleo, 2020-10-17:
No sorry, no place in the flash.

YVINEE, 2024-03-10:
to solve curl "No such file"
In GDriveConf
before the line :
chmod +x ${gdrive_bin_dir}curl
add :
echo "Fix curl does not work on Yi Dome Guard"
cd ${gdrive_bin_dir}
rm curl
wget https://github.com/moparisthebest/static-curl/releases/download/v8.5.0/curl-armv7
mv curl-armv7 curl
echo "Fix curl done"

Notice I added also :
&prompt=consent
before :
&access_type=offline
to be sure to a have the refresh_token