roleoroleo/yi-hack-MStar #79: Possibility to turn sleep on/off?

poiedk, 2020-01-25:
Is there a way to turn sleep on/off?

roleoroleo, 2020-01-25:
What is sleep?

poiedk, 2020-01-25:
I meant standby or anyway to turn on/off camera like in the official application.
Thank you for your response!

roleoroleo, 2020-01-27:
I don't know exactly what happen when you "switch off" the cam.
All the services are on and the cam is cloud connected.
The only difference is that you can't see the video on the app.
However, I think that is possible to add a switch in "Camera settings" page.

poiedk, 2020-01-27:
Also the camera goes into hiding position!

roleoroleo, 2020-01-29:
Try the last version.

poiedk, 2020-01-29:
Nice, working like a charm.
One thing that I noticed is when i do
`curl http://CAMERA_IP:8080/cgi-bin/camera_settings.sh?switch_on=no`
the camera closes but motor won't move to hide it!
Other than that you are doing awesome work!
Thanks again!

roleoroleo, 2020-01-29:
Try to go to preset 1 before (or after) switch off the cam.

poiedk, 2020-01-29:
You mean `http://CAMERA_IP:8080/cgi-bin/preset.sh` ?

roleoroleo, 2020-01-29:
Yes.

poiedk, 2020-01-29:
Nothing happens! Tried before and after.

roleoroleo, 2020-01-29:
http://CAMERA_IP:8080/cgi-bin/preset.sh?num=1

poiedk, 2020-01-29:
> http://CAMERA_IP:8080/cgi-bin/preset.sh?num=1

Already tried this, nothing happens!

roleoroleo, 2020-01-29:
Ok.

nicoolaro, 2020-01-31:
In my case on/off doesn't turn on the camera.
Do i need to turn it on from the app first?

poiedk, 2020-02-01:
Do you click save configuration after switch on/off?

nicoolaro, 2020-02-01:
> Do you click save configuration after switch on/off?

that works but the lens doesn't move in front , stays like its off except i can see black picture

poiedk, 2020-02-01:
These settings won't synchronize with official Yi app, try to move the camera on direction you want from PTZ section and check again to turn on/off!

nicoolaro, 2020-02-01:
> These settings won't synchronize with official Yi app, try to move the camera on direction you want from PTZ section and check again to turn on/off!

thank you, but this is not good then, if you move camera with PTZ and then try to use Yi App it really messes up the Yi App default position, now when i turn camera ON in app, lens is completely out of direction/

Thanks anyway.

roleoroleo, 2020-02-01:
To better understand I would need a dump of ipc messages.
Could someone sniff ipc packets when enabling/disabling the camera?
Using this binary:
[ipc_sniff.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4142864/ipc_sniff.gz)

poiedk, 2020-02-01:
> To better understand I would need a dump of ipc messages.
> Could someone sniff ipc packets when enabling/disabling the camera?
> Using this binary:
> [ipc_sniff.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4142864/ipc_sniff.gz)

If any guide on how to use ipc binary!

roleoroleo, 2020-02-01:
Gunzip the file.
Copy the file via ftp on /tmp
Connect with putty.
Run ipc_sniff.
Switch on/off the cam using the official app.
You will see a lot of numbers on putty.
Copy and paste them here.

nicoolaro, 2020-02-01:
> Gunzip the file.
> Copy the file via ftp on /tmp
> Connect with putty.
> Run ipc_sniff.
> Switch on/off the cam using the official app.
> You will see a lot of numbers on putty.
> Copy and paste them here.

what are credentials for ftp and ssh, cant find it

poiedk, 2020-02-01:
> 
> 
> Gunzip the file.
> Copy the file via ftp on /tmp
> Connect with putty.
> Run ipc_sniff.
> Switch on/off the cam using the official app.
> You will see a lot of numbers on putty.
> Copy and paste them here.

/tmp # sh ipc_sniff
ipc_sniff: line 1: syntax error: unexpected word (expecting ")")

poiedk, 2020-02-01:
> 
> 
> > Gunzip the file.
> > Copy the file via ftp on /tmp
> > Connect with putty.
> > Run ipc_sniff.
> > Switch on/off the cam using the official app.
> > You will see a lot of numbers on putty.
> > Copy and paste them here.
> 
> what are credentials for ftp and ssh, cant find it

**root** for password and leave blank for password

nicoolaro, 2020-02-01:
> > > Gunzip the file.
> > > Copy the file via ftp on /tmp
> > > Connect with putty.
> > > Run ipc_sniff.
> > > Switch on/off the cam using the official app.
> > > You will see a lot of numbers on putty.
> > > Copy and paste them here.
> > 
> > 
> > what are credentials for ftp and ssh, cant find it
> 
> **root** for username and leave blank for password

Thanks, can you share command how to run this file from putty, also this file goes to root/tmp or create a /tmp in yi-hack folder

poiedk, 2020-02-01:
The file goes to root/tmp,
what i did was `cd tmp` and then `sh ipc_sniff` but i get an error!

roleoroleo, 2020-02-01:
Copy the file in /tmp
Run it without sh:
```
cd /tmp
./ipc_sniff
```

nicoolaro, 2020-02-01:
> Copy the file in /tmp
> Run it without sh:
> 
> ```
> cd /tmp
> ./ipc_sniff
> ```

/tmp # ./ipc_sniff
-sh: ./ipc_sniff: Permission denied

i changed root password via telnet if that makes any difference

nicoolaro, 2020-02-01:
> Copy the file in /tmp
> Run it without sh:
> 
> ```
> cd /tmp
> ./ipc_sniff
> ```

/tmp # ./ipc_sniff
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 01 00 00 00 0e 40 0e 40 00 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 e1 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 e2 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 e1 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 01 00 00 00 0e 40 0e 40 00 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 8f 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 e3 00 01 00 04 00 00 00 01 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 e4 00 01 00 04 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
02 00 00 00 08 00 00 00 74 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 7c 00 7c 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 31. Status: Success!
Parsing message.
02 00 00 00 04 00 00 00 07 10 01 00 0f 00 00 00 2f 74 6d 70 2f 6d 6f 74 69 6f 6e                                                                                         2e 6a 70 67
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 01 00 00 00 0e 40 0e 40 00 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
04 00 00 00 10 00 00 00 13 60 13 60 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 158. Status: Success!
Parsing message.
01 00 00 00 04 00 00 00 21 40 01 00 8e 00 00 00 7b 22 63 6f 64 65 22 3a 30 2c 22                                                                                         74 79 70 65 22 3a 34 2c 22 61 6c 61 72 6d 5f 74 79 70 65 22 3a 31 2c 22 64 69 6                                                                                        4 22 3a 22 42 46 55 53 59 31 45 4f 45 51 41 45 51 51 31 39 30 39 30 32 22 2c 22                                                                                         64 65 76 69 63 65 5f 76 65 72 73 69 6f 6e 22 3a 22 34 2e 36 2e 30 2e 30 41 5f 32                                                                                         30 31 39 30 38 32 37 31 35 34 39 22 2c 22 74 69 6d 65 22 3a 22 31 35 38 30 35 3                                                                                        8 38 39 35 35 37 31 33 22 2c 22 6d 6f 64 65 22 3a 22 48 32 30 31 43 22 7d
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
02 00 00 00 08 00 00 00 75 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 01 00 00 00 0e 40 0e 40 00 00 00 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE

roleoroleo, 2020-02-06:
The dump you posted doesn't contain anything strange.
I don't know why the cam doesn't go to the default position but without a cam I can't do anything else.
Sorry.

poiedk, 2020-02-06:
> 
> 
> Nice, working like a charm.
> One thing that I noticed is when i do
> `curl http://CAMERA_IP:8080/cgi-bin/camera_settings.sh?switch_on=no`
> the camera closes but motor won't move to hide it!
> Other than that you are doing awesome work!
> Thanks again!

@roleoroleo i found that if i send this command 6 times only then the motor hides the camera lens!

roleoroleo, 2020-02-06:
I don't understand.
Could you explain better?

poiedk, 2020-02-06:
If I send this command `curl http://CAMERA_IP:8080/cgi-bin/camera_settings.sh?switch_on=no` the camera turns off but the lens won't move to preset position, if i keep sending this command for another 5 times then the camera lens goes to preset position!

roleoroleo, 2020-02-06:
Instead of curl command, try to use ipc_cmd with putty.
```
ipc_cmd -t off
```

poiedk, 2020-02-06:
> Instead of curl command, try to use ipc_cmd with putty.
> 
> ```
> ipc_cmd -t off
> ```

Great, that worked.
Thank you!

roleoroleo, 2020-02-06:
Ok, I will solve the problem in the next release.

roleoroleo, 2020-02-22:
Please try release 0.2.7

nicoolaro, 2020-03-03:
> Please try release 0.2.7

Well now camera doesn't go into "back " position when switched off. It just go up and turns off. Anyways i'm not sure if this can be further improved, i can live with this. thanks

roleoroleo, 2020-03-03:
I will try to improve it.
If you run the command manually, does it work?
`ipc_cmd -t off`

nicoolaro, 2020-03-04:
> I will try to improve it.
> If you run the command manually, does it work?
> `ipc_cmd -t off`

nope, same thing, it switches off but doesn't go 180 degrees back

miclass, 2020-04-12:
> 
> 
> > I will try to improve it.
> > If you run the command manually, does it work?
> > `ipc_cmd -t off`
> 
> nope, same thing, it switches off but doesn't go 180 degrees back

What do you mean with 180 degrees back?
Is there a functionality to keep the rotate the sensor and keep it far from the light?
It would be very usefull I would need to put the cam in front of direct sun and the direct beams can harm the sensor.

Thanks.

nicoolaro, 2020-04-12:
No. 180 is off mode.

closed 2020-05-07 by roleoroleo