roleoroleo/yi-hack-MStar #80: RTSP broken on Y25 0.2.5 / 9FUS

rado0x54, 2020-01-27:
Just FYI that RTSP is broken in release 0.2.5 for Y25 / 9FUS. It seems like there is no `rRTSPServer` binary present in the build.

With 0.2.4 everything is well. `/home/yi-hack/bin/rRTSPServer` is there and working. 

Please feel free to close, just wanted to point that out.

roleoroleo, 2020-01-28:
The same problem posted here, I think.
https://github.com/roleoroleo/yi-hack-MStar/issues/23#issuecomment-572564880
But I verified the archive and the files are correct.

roleoroleo, 2020-03-17:
I fixed this issue in the last release.
Could you check it?

rado0x54, 2020-05-21:
I don’t have my hardware accessible right now. Will close for now and reopen if the issue still comes up with future versions.

closed 2020-05-21 by rado0x54