roleoroleo/yi-hack-MStar #82: Record rotation

johnnymnemonic84, 2020-02-03:
Hi, is it possible to schedule a rotation of the records?
Something like remove after x days or/and after x size

I'm trying to enable crontab to do it but don't know how to "translate" it to web management

roleoroleo, 2020-02-03:
You could add a new setting (i.e. a text input with a number of days) in "Configuration" page similar to HTTPD port setting (configurations.html).
Then you could change the js file to add the new setting (configuration.js).

For the static components you need to:
- Change /home/yi-hack/script/system.sh to add the cron management.
For example:
```
if [[ $(get_config CLEAN_PERIOD) != "0" ]] ; then
    mkdir -p /var/spool/cron/crontabs/
    echo "  0  *  *  *  *  /home/yi-hack/bin/script_to_clean_records.sh" >/var/spool/cron/crontabs/root
    /usr/sbin/crond -c /var/spool/cron/crontabs/
fi
```
- Add the new settings to /home/yi-hack/etc/system.conf (CLEAN_PERIOD=0)

johnnymnemonic84, 2020-02-04:
Ok thanks, if someone need the script I wrote this with a fixed percentage of size

`#!/bin/sh
cd /tmp/sd/record
while [ `df -h /tmp/sd/ | grep mmc | awk '{print $5}' | tr -d '%'` -gt 90 ]
do
  rm -rf `ls -lt | tail -n1 | awk '{print $9}'`
done`

roleoroleo, 2020-02-04:
Do you think you will create a complete PR?
Otherwise I can work on it to integrate your script.

roleoroleo, 2020-02-22:
Please try release 0.2.7

johnnymnemonic84, 2020-02-23:
I have installed it today, thanks

roleoroleo, 2020-02-24:
I found an error.
Please rename
/home/yi-hack/script/clean_record.sh
to
/home/yi-hack/script/clean_records.sh

johnnymnemonic84, 2020-02-24:
Done thank you

closed 2020-05-24 by roleoroleo