roleoroleo/yi-hack-MStar #83: Variable MAC address / static IP

arhunt35, 2020-02-06:
Both of my cameras produce different MAC addresses when I turn off / on. They cycle through three different ones. This has made assigning a single static IP on my router impossible. Is it possible to either permanently fix the MAC address or set a static IP on the camera?

roleoroleo, 2020-02-06:
You could try to edit /home/app/init.sh and add these line before `ifconfig wlan0 up`:
```
ifconfig wlan0 hw ether xx:xx:xx:xx:xx:xx
```

arhunt35, 2020-02-06:
Thanks for the super quick reply (and for the project as well - it's excellent).  I'll give it a go.

arhunt35, 2020-02-06:
This worked for me.  Thanks again.

closed 2020-02-06 by arhunt35

roleoroleo, 2020-02-06:
If you upgrade the fw, you need to repeat the operation.

arhunt35, 2020-02-06:
Thanks.  I figured that would be the case.  Is there an easy way to add this to the web interface? If you can point me in the right direction I'll give it a go.

roleoroleo, 2020-02-07:
It's a long chain but is possible.
You have to modify:

- the .html page where you want to put the box;
- the .js file with the same name of the page;
- the cgi (shell script) called by js ajax function;
- the .conf file where you want to store the configuration parameter;
- the init.sh file that read .conf file at boot and set the address.